introlab/rtabmap_ros #628: run  rtabmapviz  happen crashed

auoCleanRobot, 2021-08-13:
OS and ROS version : Manjaro 21.1.0 and ros-noetic
After I build rtabmap_ros successfully and open  "roslaunch rtabmap_ros demo_robot_mapping.launch localization:=true"
with rviz also successfuly.
But I use rtabmapviz happen crash. The following is from crash's message when open rtabmapviz.
please help me . thanks

>  gdb -ex run --arg bash rosrun rtabmap_ros rtabmapviz
GNU gdb (GDB) 10.2
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from bash...
(No debugging symbols found in bash)
Starting program: /usr/bin/bash rosrun rtabmap_ros rtabmapviz
[Detaching after fork from child process 67951]
[Detaching after fork from child process 67953]
[Detaching after fork from child process 67955]
[Detaching after fork from child process 67956]
[Detaching after fork from child process 67957]
[Detaching after fork from child process 67958]
process 67947 is executing new program: /home/robot/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7fffc89ef640 (LWP 67962)]
[New Thread 0x7fffc81ee640 (LWP 67963)]
[New Thread 0x7fffc79ed640 (LWP 67964)]
[New Thread 0x7fffc71ec640 (LWP 67965)]
[New Thread 0x7fffc69eb640 (LWP 67966)]
[New Thread 0x7fffc61ea640 (LWP 67967)]
[New Thread 0x7fffc59e9640 (LWP 67968)]
[New Thread 0x7fffc51e8640 (LWP 67969)]
[New Thread 0x7fffc49e7640 (LWP 67970)]
[New Thread 0x7fffc41e6640 (LWP 67971)]
[New Thread 0x7fffc39e5640 (LWP 67972)]
[New Thread 0x7fffc31e4640 (LWP 67973)]
[New Thread 0x7fffc29e3640 (LWP 67974)]
[New Thread 0x7fffc21e2640 (LWP 67975)]
[New Thread 0x7fffc19e1640 (LWP 67976)]
[New Thread 0x7fffc11e0640 (LWP 67977)]
[ INFO] [1627810966.415120310]: Starting node...
[New Thread 0x7fffc03ce640 (LWP 67978)]
[New Thread 0x7fffbdd4d640 (LWP 67979)]
[New Thread 0x7fffbd444640 (LWP 67981)]
[New Thread 0x7fffbcc43640 (LWP 67982)]
[New Thread 0x7fffa7fff640 (LWP 67983)]
[New Thread 0x7fffaffff640 (LWP 67984)]
[New Thread 0x7fffaf7fe640 (LWP 67985)]
[ INFO] [1627810966.736577327]: rtabmapviz: Using configuration from "/home/robot/.ros/rtabmapGUI.ini"
--Type <RET> for more, q to quit, c to continue without paging--

Thread 1 "rtabmapviz" received signal SIGSEGV, Segmentation fault.
0x00007ffff2272164 in inflate_fast () from /usr/lib/libpcl_surface.so.1.9
(gdb) bt
#0  0x00007ffff2272164 in inflate_fast () at /usr/lib/libpcl_surface.so.1.9
#1  0x00007fffe47218d0 in inflate () at /usr/lib/libPocoFoundation.so.80
#2  0x00007fffe1f48f0d in uncompress2 () at /usr/lib/libz.so.1
#3  0x00007fffe1f49004 in uncompress () at /usr/lib/libz.so.1
#4  0x00007fffe8f42eb8 in  () at /usr/lib/libQt5Core.so.5
#5  0x00007fffe8f46b49 in QResource::uncompressedData() const () at /usr/lib/libQt5Core.so.5
#6  0x00007fffe8f46c5a in  () at /usr/lib/libQt5Core.so.5
#7  0x00007fffe8f46d45 in  () at /usr/lib/libQt5Core.so.5
#8  0x00007fffe8f2b20b in QFile::open(QFlags<QIODevice::OpenModeFlag>) () at /usr/lib/libQt5Core.so.5
#9  0x00007fffbf20b52a in  () at /usr/lib/libKF5ConfigCore.so.5
#10 0x00007fffbf1f2569 in  () at /usr/lib/libKF5ConfigCore.so.5
#11 0x00007fffbf2107da in KSharedConfig::KSharedConfig(QString const&, QFlags<KConfig::OpenFlag>, QStandardPaths::StandardLocation) () at /usr/lib/libKF5ConfigCore.so.5
#12 0x00007fffbf2109fd in KSharedConfig::openConfig(QString const&, QFlags<KConfig::OpenFlag>, QStandardPaths::StandardLocation) () at /usr/lib/libKF5ConfigCore.so.5
#13 0x00007fffbf4b4d49 in KIconTheme::KIconTheme(QString const&, QString const&, QString const&) ()
    at /usr/lib/libKF5IconThemes.so.5
#14 0x00007fffbf4a54f1 in  () at /usr/lib/libKF5IconThemes.so.5
#15 0x00007fffbf4a59b7 in  () at /usr/lib/libKF5IconThemes.so.5
#16 0x00007fffbf4ab24a in  () at /usr/lib/libKF5IconThemes.so.5
#17 0x00007fffbf4abaf3 in KIconLoader::global() () at /usr/lib/libKF5IconThemes.so.5
#18 0x00007fffbd45f9c6 in KStyle::pixelMetric(QStyle::PixelMetric, QStyleOption const*, QWidget const*) const ()
    at /usr/lib/libKF5Style.so.5
#19 0x00007fffe9bd66ef in QAbstractButton::iconSize() const () at /usr/lib/libQt5Widgets.so.5
#20 0x00007fffe9c7e42e in QPushButton::initStyleOption(QStyleOptionButton*) const () at /usr/lib/libQt5Widgets.so.5
#21 0x00007fffe9c7f0c8 in  () at /usr/lib/libQt5Widgets.so.5
#22 0x00007fffe9c7f1e6 in QPushButton::QPushButton(QString const&, QWidget*) () at /usr/lib/libQt5Widgets.so.5
--Type <RET> for more, q to quit, c to continue without paging--
#23 0x00007fffe9d30144 in  () at /usr/lib/libQt5Widgets.so.5
#24 0x00007fffe9d301c7 in  () at /usr/lib/libQt5Widgets.so.5
#25 0x00007fffe9d30398 in  () at /usr/lib/libQt5Widgets.so.5
#26 0x00007fffe9d307ce in QProgressDialog::QProgressDialog(QWidget*, QFlags<Qt::WindowType>) () at /usr/lib/libQt5Widgets.so.5
#27 0x00007fffe30d05be in rtabmap::PreferencesDialog::PreferencesDialog(QWidget*) () at /usr/local/lib/librtabmap_gui.so.0.20
#28 0x00005555555cb7f3 in PreferencesDialogROS::PreferencesDialogROS(QString const&) ()
#29 0x00005555555a25d5 in rtabmap_ros::GuiWrapper::GuiWrapper(int&, char**) ()
#30 0x000055555558fbd3 in main ()
(gdb) quit
A debugging session is active.

        Inferior 1 [process 67947] will be killed.

Quit anyway? (y or n) y
 ~

matlabbe, 2021-09-03:
Humm, maybe OS related. Did you try rtabmap binaries (`ros-noetic-rtabmap-ros`) in case it is a compilation problem?

Note however that rtabmapviz is not mandatory for the slam system to work, its purpose is exactly the same than rviz with some custom visualizations. You can use rviz for visualization instead.

auoCleanRobot, 2021-09-05:
rtabmap itself can run well without error. It's rtabmapviz that gets into trouble.

matlabbe, 2021-09-05:
Compatibility between Qt, VTK and PCL is often fragile. I try generally to make sure rtabmap is working with the combination of their binaries versions on vanilla Ubuntu LTS distros (18.04, 20.04...).

closed 2021-10-10 by auoCleanRobot