introlab/rtabmap_ros #629: RGBD Odom Orientation and Position

jacklu333333, 2021-08-13:
Hi, I am using a dual-camera setup for rtabmap with `Kinetic` and `Astra`, following the `demo_two_kinects.launch`.

I want to take advantage of the RGB information to obtain better accuracy,  with visual information. Therefore, I hybrid the `wheel_odom` with `rgbd_odom` provided by rtabmap with `robot_localization`. 
However, in my setup, when I inspect the data of `rgbd_odom`, I found that the orientation always points toward the same direction. 

1. I am wondering how the `position` and `orientation` are calculated by rtabmap?

2. Does it require other information such as IMU so that it can calculate the `position` and `orientation`? Or the RGBD information is enough?

Note: I set up my camera with Kinetic heading toward the front and Astra heading toward the rear. I am not sure it is the reason causing the problem or not.

Thank you in advance.

Best Regards,
Jack Lu

matlabbe, 2021-09-03:
What does look like the odometry message? It should compute orientation, IMU is optional.

jacklu333333, 2021-09-06:
Thank you. You replied to me [here](http://official-rtab-map-forum.206.s1.nabble.com/Question-about-IMU-information-usage-td8469.html#a8521).

Best Regards,
Jack Lu

closed 2021-09-06 by jacklu333333