introlab/rtabmap_ros #633: Build rtabmap docker image on Xavier.

Mikor-mkr, 2021-08-24:
Hello again,

this time I have an issue trying to build this docker file:

```
FROM introlab3it/rtabmap_ros:noetic

# nvidia-container-runtime
ENV NVIDIA_VISIBLE_DEVICES \
    ${NVIDIA_VISIBLE_DEVICES:-all}
ENV NVIDIA_DRIVER_CAPABILITIES \
    ${NVIDIA_DRIVER_CAPABILITIES:+$NVIDIA_DRIVER_CAPABILITIES,}graphics
```

But when I try to run it as is I get the following error:
`WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
`
And after some google searches I came up at this [post](https://stackoverflow.com/questions/66662820/m1-docker-preview-and-keycloak-images-platform-linux-amd64-does-not-match-th)

Then I tried to run the image by adding `--platform=linux/arm64/v8` which led to this error:
```
Unable to find image 'rtabmap_ros3d:latest' locally
docker: Error response from daemon: pull access denied for rtabmap_ros3d, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
```
Although, the image rtabmap_ros3d exists locally.
```
$ docker image list
REPOSITORY                TAG         IMAGE ID       CREATED          SIZE
rtabmap_ros3d             latest      6aa14820f27b   51 minutes ago   3.16GB
introlab3it/rtabmap_ros   noetic      599db7007cd2   5 weeks ago      3.16GB
nvidia/cuda               11.0-base   2ec708416bb8   12 months ago    122MB
```

Any ideas? Thanks in advance.

matlabbe, 2021-08-24:
Hi, not sure if it related to rtabmap code, or just a M1 docker issue. Is another image working? Can you `docker -it --rm ...` into `introlab3it/rtabmap_ros  /noetic`  or `nvidia/cuda:11.0-base` images?

Mikor-mkr, 2021-08-24:
I will get back to you on this tomorrow as I have just left the office.