introlab/rtabmap_ros #638: Several unhandled exception cause process crash

easylyou, 2021-09-07:
Hi, I'm doing some fuzzing test and found several unhandled exceptions which cause process crash.
I manually reviewed the code and found the exceptions are mainly from two aspects.
- throwing by other codes or library.
- assert fail.

For the two types, the main common cause is the malformed message from camera or laser. In fact sensor could be wrong sometimes, maybe it is better to do checks and ignore malformed message instead of making process crash?

Here are the logs I collected. Because the process terminated by exception, I could not collect the back-traces. It seems hard to locate the error place.
```
terminate called after throwing an instance of 'cv_bridge::Exception'
  what():  Image is wrongly formed: step < width * byte_depth * num_channels  or  1280 != 334692672 * 4 * 1
```
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot store a negative time point in rclcpp::Time
```
```
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) ../modules/calib3d/src/calibration.cpp:1211: error: (-215:Assertion failed) fabs(sc) > DBL_EPSILON in function 'cvFindExtrinsicCameraParams2'
```
```
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2021-08-26 18:51:08.530) MsgConversion.cpp:1419::convertRGBDMsgs() Condition (imageWidth % depthWidth == 0 && imageHeight % depthHeight == 0 && imageWidth/depthWidth == imageHeight/depthHeight) not met! [rgb=320x240 depth=256x240]
```
```
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2021-08-24 21:22:51.521) LaserScan.cpp:301::init() Condition (rangeMax>rangeMin) not met!
```
```
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2021-08-25 09:34:32.189) LaserScan.cpp:302::init() Condition ((angleIncrement>0 && angleMax>angleMin) || (angleIncrement<0 && angleMax<angleMin)) not met!
```

matlabbe, 2021-09-09:
Underlying libraries like OpenCV use a lot of asserts to make sure data sent to their function are valid. This is responsibility of the user of the library to provide right data. We use similar approach for rtabmap library, to avoid to deal of failure cases silently, we show where it failed and the user should make sure to provide the right data. Personally, I prefer an assert than a function returning a default value for which my code could use wrongly afterwards.

However, the assert approach doesn't show the full backtrace up to the assert, which may be difficult to know from which part of the code it originated, particularly when we don't know the library.

With some experience, here is what is seems the causes of the asserts you shown:

```
terminate called after throwing an instance of 'cv_bridge::Exception'
  what():  Image is wrongly formed: step < width * byte_depth * num_channels  or  1280 != 334692672 * 4 * 1
```
It is from `cv_bridge`, which is independent of rtabmap, there is something wrong with the input `sensor_msgs/Image` data sent to rtabmap node. You can `rostopic echo` your image topics and see why the meta data is not correct, there is a problem with the publishing node.

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot store a negative time point in rclcpp::Time
```
Are stamps in topic messages are correct?

```
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) ../modules/calib3d/src/calibration.cpp:1211: error: (-215:Assertion failed) fabs(sc) > DBL_EPSILON in function 'cvFindExtrinsicCameraParams2'
```
This one is tricky, maybe related to this bug https://github.com/introlab/rtabmap/issues/503. It may happen inside `cv::solvePnPRansac()` function, is the camera_info topic valid? Ideally, a rosbag reproducing 100% of time this issue could help to debug.

```
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2021-08-26 18:51:08.530) MsgConversion.cpp:1419::convertRGBDMsgs() Condition (imageWidth % depthWidth == 0 && imageHeight % depthHeight == 0 && imageWidth/depthWidth == imageHeight/depthHeight) not met! [rgb=320x240 depth=256x240]
```
I assume it is form the following assert on `ros2` branch:
https://github.com/introlab/rtabmap_ros/blob/36be6925f0b85eff9f26c97e30d28b7c77bc3a6f/src/MsgConversion.cpp#L1416-L1419
This issue has been resolved on ros1 `master` branch, we don't need anymore that rgb size should be exactly divided by depth size.

```
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2021-08-24 21:22:51.521) LaserScan.cpp:301::init() Condition (rangeMax>rangeMin) not met!
```
There is a problem with the input `sensor_msgs/LaserScan` topic, the range max is not higher than range min?!

```
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2021-08-25 09:34:32.189) LaserScan.cpp:302::init() Condition ((angleIncrement>0 && angleMax>angleMin) || (angleIncrement<0 && angleMax<angleMin)) not met!
```
Same as the previous asset, the `sensor_msgs/LaserScan` topic doesn't contain valid parameters.

easylyou, 2021-09-10:
Thanks very much for your replay！
I understand asserts are widely used in libraries, which is a simple and general way to check input data. The error handler of asserts is simply to abort the node. In our long-running experiments, although the probability is low, it is possible a strange message send by publisher occurs and then all tasks are broken... It is of course something wrong in publisher and should be fixed by the publisher side. In fact coding a bug-free node or debugging a bug is hard. So I believe making publisher acting properly all the time is not a easy thing, especially the publisher may be unreliable or untrusted.
On the other hand, the error handler seems not complex on the received side.  I think printing a WARN log and only dropping it is good enough.

matlabbe, 2021-09-10:
In some places like this one:
https://github.com/introlab/rtabmap_ros/blob/b2ac4b6ea77817ad8940a9c34a69dd299b2d5a61/src/CoreWrapper.cpp#L1241-L1242
we return with an error without closing the node. Other example here in a function:
https://github.com/introlab/rtabmap_ros/blob/b2ac4b6ea77817ad8940a9c34a69dd299b2d5a61/src/MsgConversion.cpp#L1766-L1780
but for some reasons we didn't return false here for the following condition:
https://github.com/introlab/rtabmap_ros/blob/b2ac4b6ea77817ad8940a9c34a69dd299b2d5a61/src/MsgConversion.cpp#L1756-L1758

The assert above could be fixed to return false instead with an error msg. Someone would have to read all the code to find lines like that that could be changed for a returning false with error. I guess it is more easy for functions returning already a boolean telling that the call failed or not. For functions not returning a boolean, it may be trickier to change to an error msg. All functions would need to be modified to return a boolean, or all functions would need to manage throw/catch exceptions. 

The problem with exceptions is that it may corrupt some states or data in rtabmap node, which may cause strange problems even more difficult to debug afterwards. For example, if you have a process modifying the memory or database with multiple steps, then if one step fails, you would have to catch that exception and deconstruct the previous steps before rethrowing again the exception. This would have be done for every step. Some steps may be impossible to recover without some backups before doing the process. It seems to me a big refactoring work to do (and may not be efficient in terms of computation time and memory to keep in cache a "Undo" stack). Fixing a bad publisher seems to be the right "patch" to do :)

easylyou, 2021-09-13:
Thanks for your reply.  Your reply is very detailed!
"The problem with exceptions is that it may corrupt some states or data in rtabmap node". I very much agree that. I underestimated the difficulty. 

I think the "backup" solution is actually not needed to address some of them.  For example, we can check `step < width * byte_depth * num_channels` or `rangeMax>rangeMin` early.  Once a laser message received, do a check first and drop it if check fails. Then we do not need `assert` in the later code. Whatmore, `assert` is also wasting cycles if it is repeated.

"all functions would need to manage throw/catch exceptions." I understand it and agree that, "all functions .. manage throw/catch exceptions" is not a good idea. But I think a exception from a third-party library need to be caught. For example, `cv::Exception` from `libopencv` is better to catch I think.

Anyway, there could be lots of modifications and necessary tests for rtabmap node to address them. And some of them still need publisher to fix. 

Thanks again for your time! I think I can close the issue.

matlabbe, 2021-09-13:
I agree that some asserts like checking input topic parameters could be converted to a "return error" instead. If someone has some time to change those kind of asserts, I'll be happy to review the pull request. I have currently limited time and I think for such refactoring, we would need unit tests (which is a task that I push always behind higher priority issues, and that we are still in transition between ros1 and ros2). 

cheers,
Mathieu

closed 2021-09-13 by matlabbe