introlab/rtabmap_ros #640: rtabmap with livox avia

aditdoshi333, 2021-09-14:
Hello,

Just curious to know that is there any example of running rtabmap with livox avia. If not, Can you guide me a bit how to make it work?

matlabbe, 2021-09-15:
For rtabmap node, `subscribe_scan_cloud` can be set to true, then remap `scan_cloud` topic to livox's PointCloud2 topic.

For `icp_odometry`, there is currently no working example with livox butterfly pattern, so external odometry is required (like using [LOAM-Livox](https://github.com/hku-mars/loam_livox) as input odometry to rtabmap node). [We recently integrated F-LOAM](http://official-rtab-map-forum.206.s1.nabble.com/icp-odometry-with-LOAM-crash-tp8261p8563.html), which can be used by `icp_odometry`, maybe it could work.

alienshengzi, 2022-05-03:
I have the same problem to solve

kosmonauta144, 2023-11-06:
@matlabbe  what about livox lidars with regular pattern like Mid360  for mapping and icp odometry?