introlab/rtabmap_ros #642: ICP Odometry F2M - No keyframes are added / local map is not updated when inliers drop below threshold

aseligmann, 2021-09-17:
Hello Mathieu,

I am using RTAB-Map for SLAM based only on LiDAR inputs (sensor data aggregated from 3 ring-like LiDARs), and I'm using the `icp_odometry` node for odometry.

Specifically, I am running `icp_odometry` with `Odom/Strategy = 0`, i.e. in F2M mode. 
However, it seems that the local map is never updated, even with `Odom/ScanKeyFrameThr = 0`.

When I'm moving the LiDARs around, the inlier-ratio drops until the odometry is lost:
```raw
[ INFO] [1631882186.131840614, 10.177000000] [/rtabmap/icp_odometry]: Odom: ratio=0.286955, std dev=0.079810m|0.025238rad, update time=0.136829s
[ INFO] [1631882186.430054016, 10.304000000] [/rtabmap/icp_odometry]: Odom: ratio=0.249799, std dev=0.066048m|0.020886rad, update time=0.144275s
[ INFO] [1631882186.683516523, 10.402000000] [/rtabmap/icp_odometry]: Odom: ratio=0.234624, std dev=0.068627m|0.021702rad, update time=0.146289s
[ INFO] [1631882186.958847015, 10.512000000] [/rtabmap/icp_odometry]: Odom: ratio=0.233638, std dev=0.064841m|0.020504rad, update time=0.176061s
[ INFO] [1631882187.404416159, 10.713000000] [/rtabmap/icp_odometry]: Odom: ratio=0.219604, std dev=0.065338m|0.020662rad, update time=0.156655s
[ INFO] [1631882187.719456939, 10.806000000] [/rtabmap/icp_odometry]: Odom: ratio=0.212001, std dev=0.065574m|0.020736rad, update time=0.206968s
```

The SLAM node itself works fine until the odometry is lost. 

How can I ensure that the local ICP odometry map is updated more frequently, to avoid having the odometry get lost?

---

Here are the launch parameters that I'm using for `icp_odometry`:
```xml
<node pkg="rtabmap_ros" type="icp_odometry" name="icp_odometry" output="screen">

  <remap from="scan_cloud" to="$(arg clouds_filtered_topic)"/>
  <param name="frame_id"        type="string" value="$(arg base_frame_id)"/>  
  <param name="odom_frame_id"   type="string" value="$(arg odom_frame_id)"/>
  <param name="map_frame_id"    type="string" value="$(arg map_frame_id)"/>
  <param name="expected_update_rate" type="double" value="$(arg expected_update_rate)"/>

  <remap if="$(arg use_imu)" from="imu" to="$(arg imu_topic)"/>
  <param if="$(arg use_imu)" name="guess_frame_id"   type="string" value="$(arg base_frame_id)_stabilized"/>
  <param if="$(arg use_imu)" name="wait_imu_to_init" type="bool" value="true"/>

  <!-- ICP parameters -->
  <param name="Icp/PointToPlane"        type="string" value="true"/>
  <param name="Icp/Iterations"          type="string" value="40"/>
  <param name="Icp/VoxelSize"           type="string" value="$(arg voxel_size)"/>
  <param name="Icp/DownsamplingStep"    type="string" value="1"/> <!-- cannot be increased with ring-like lidar -->
  <param name="Icp/Epsilon"             type="string" value="0.001"/>
  <param name="Icp/PointToPlaneK"       type="string" value="20"/>
  <param name="Icp/PointToPlaneRadius"  type="string" value="0"/>
  <param name="Icp/PointToPlaneLowComplexityStrategy"       type="string" value="1"/>
  <param name="Icp/PointToPlaneMinComplexity"               type="string" value="0.0025"/>
  <param name="Icp/MaxTranslation"      type="string" value="3"/> 
  <param name="Icp/MaxCorrespondenceDistance" type="string" value="$(arg corr_dist)"/>
  <param name="Icp/PM"                  type="string" value="true"/> <!-- libpointmatcher -->
  <param name="Icp/PMOutlierRatio"      type="string" value="0.7"/>
  <param name="Icp/CorrespondenceRatio" type="string" value="0.01"/>  
  <param name="Icp/RangeMax"            type="string" value="20"/>

  <!-- Odom parameters -->       
  <param name="Odom/KeyFrameThr"           type="string" value="0.7"/>
  <param name="Odom/VisKeyFrameThr"        type="string" value="0.7"/>
  <!-- <param name="Odom/ScanKeyFrameThr"       type="string" value="0.7"/> -->
  <param name="Odom/ScanKeyFrameThr"       type="string" value="0"/>
  <param name="Odom/Strategy"              type="string" value="0"/>
  <param name="OdomF2M/BundleAdjustment"   type="string" value="0"/>
  <param name="OdomF2M/ScanSubtractRadius" type="string" value="0.2"/>
  <param name="OdomF2M/MaxSize"            type="string" value="15000"/>    
  <param name="OdomF2M/ScanMaxSize"        type="string" value="15000"/>

</node>
```

matlabbe, 2021-09-17:
Can you record and share a small bag with your lidar data?

Note that icp_odometry works better when rebuilding rtabmap from source with libpointmatcher support.

aseligmann, 2021-09-20:
I've compiled both rtabmap (0.20.7-melodic) and rtabmap_ros (0.20.7-melodic) from source, built with libpointmatcher (also built from source).

Here's a minimum working example demonstrating the issue:
https://www.dropbox.com/s/op9hityayepz4th/github_issue_642.zip?dl=1
The above .zip file contains a rosbag, the rtabmap.db, the console log, a launch file and a screenshot of how the data should look in RViz.
(Unfortunately I can't share everything about the use case, but the launch file should be able to demonstrate the issue.)

The launch file contains two commented out nodes, an aggregator and an assembler, which were used to generate the `/slam/clouds_combined` and `/slam/clouds_assembled` clouds in the rosbag. Between combining and assembling, I am performing some prefiltering, which I unfortunately cannot show the parameters for, but the `/slam/clouds_filtered` shows the resulting cloud which is fed to the assembler.

Thank you for taking the time to assist in troubleshooting!

aseligmann, 2021-09-23:
Bringing everything up to the 0.20.13 version solved the issue.

closed 2021-09-23 by aseligmann