introlab/rtabmap_ros #645: Odometry jump back and forth when use one camera for tracking and another camera for mapping

szx0112, 2021-09-20:
Hi @matlabbe 

I have two d435i cameras mounted on a rig to scan the internal of a pipe. I use one (front viewed) camera for tracking (generate poses and initial map). The visual odometry computed with this front camera is then utilized as the external odometry (like a simulated wheel odometry) of the second (side viewed) camera to further update the existing map (at a higher density, resolution. VO from the side camera alone easily got lost during the mapping due to the limited FOV). 

Both cameras have the static_transform to the base_link. Like base_link -> camera_5_base (i.e., front camera) -> camera_5_frames,  base_link -> camera_1_base (i.e., side_camera) -> camera_1_frames.

I got expected results (i.e., smoothed odometry) to generate the initial map using only front camera. I record the computed odometry and use it as the external odometry (i.e., /simulated_wheel_odom) during the mapping of the side camera. However, I see the odometry jump back to forth in Rtabmap GUI. This happen when I either use the VO from the front camera as the only odometry input or fuse the front and side VO using EKF. 

Below is a snapshot from the RTABMAP GUI when VO of the front camera is used as the only input odometry. Purple dots are the pose computed during the initial map (loaded using Edit -> Download all clouds), yellow dots are poses computed during the mapping of the side camera. I initially guess the jumps are caused by the wrong loop closures, however I notice it still exist even I disable it (Kp/MaxFeatures = -1). I actually got less jumps when fusing the VO from the front and side cameras. But I cannot get smoothed result in both cases.

[image]

Below is the rqt_graph 

[image]

Below is the Snapshot of /simulated_wheel_odom topic. As far as I known, /rtabmap/odom publish the topic with child_frame_id as base_link.  Thus, I think it can be directly imported during the side camera mapping because the two cameras are mounted on a rig (shared the same origin which is the base_link in my case)
```
---
header: 
  seq: 688
  stamp: 
    secs: 1631139670
    nsecs: 140150603
  frame_id: "odom"
child_frame_id: "base_link"
pose: 
  pose: 
    position: 
      x: -0.00829805247486
      y: -0.0502983480692
      z: 1.43164122105
    orientation: 
      x: 0.0105424086884
      y: 0.0116537047125
      z: -0.0488143452171
      w: 0.998684238629
  covariance: [2.227460612275536e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 2.227460612275536e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 2.227460612275536e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 2.227460612275536e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 2.227460612275536e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 2.227460612275536e-05]
twist: 
  twist: 
    linear: 
      x: 0.008366166614
      y: -0.00809245370328
      z: -0.00143815914635
    angular: 
      x: -0.0207436159253
      y: -0.015122840181
      z: -0.0196839887649
  covariance: [1.113730306137768e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.113730306137768e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.113730306137768e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.113730306137768e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.113730306137768e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.113730306137768e-05]

```
Some logs during the mapping.
[ INFO] [1632167363.062659328, 1631139672.346298875]: rtabmap (2558): Rate=0.00s, Limit=0.000s, RTAB-Map=0.1024s, Maps update=0.0063s pub=0.0000s (local map=635, WM=609)
[ INFO] [1632167363.275097123, 1631139672.356883139]: Odom: quality=100, std dev=0.005529m|0.005529rad, update time=0.059045s
[ WARN] (2021-09-20 14:49:23.805) OccupancyGrid.cpp:357::createLocalMap() Cannot create local map, scan is empty (node=2559).
[ INFO] [1632167363.818168623, 1631139672.384086917]: rtabmap (2559): Rate=0.00s, Limit=0.000s, RTAB-Map=0.0991s, Maps update=0.0048s pub=0.0000s (local map=636, WM=610)
[ WARN] (2021-09-20 14:49:23.909) OccupancyGrid.cpp:357::createLocalMap() Cannot create local map, scan is empty (node=2560).
[ INFO] [1632167363.924319355, 1631139672.389624898]: rtabmap (2560): Rate=0.00s, Limit=0.000s, RTAB-Map=0.0959s, Maps update=0.0064s pub=0.0000s (local map=636, WM=610)
[ INFO] [1632167364.135974361, 1631139672.400197616]: Odom: quality=114, std dev=0.005346m|0.005346rad, update time=0.061316s
[ INFO] [1632167364.199941832, 1631139672.403217114]: Odom: quality=91, std dev=0.005387m|0.005387rad, update time=0.062267s

I can't figure out the problem. Any comment is really appreciated. Thanks.

szx0112, 2021-09-23:
Hi @matlabbe Thanks for your reply. You are right. I believe I have three nodes published duplicated tf tree and I think they are: (1) rosbag recorded tf tree as `map -> odom -> base_link -> camera_i_base -> camera_i_frames -> ...`; (2) the newly computed rtabmap and rgbd_odometry node: tf tree same as the rosbag recorded; (3) the efk_localization node. It seems that the node publishs `odom -> base_link` tf tree.

To correct the tf tree. I delete the `/tf` topic in rosbag, and set `publish_tf = false` in ekf_localization. And now I believe I have the unique tf tree published only by rtabmap.

Next, I try to correct the frame_id as suggested. Because the  the `odom_frame_id` of the rosbag played odometry topic (`/simulated_wheel_odom`) is `odom`. Thus, I remap the odom topic published from the `rgbd_odometry` node as `odom_combined`. I also set the` odom_frame_id` as `odom_combined`. I set the guess_frame_id as `odom` as suggested. Finally, I remap the input odom topic and `odom_frame_id` of rtabmap node as `odom_combined`. I expect the tf tree looks like: `map -> odom_combined -> odom -> base_link -> ..`.

However, I found I either got tf tree where `map` is the parent link of both `odom` and `odom_combined` (in parallel), or I got `map->odom_combined -> X (broken link) odom -> base_link -> ... (two disconnected tf trees )`. After the modification, I also found the /rtabmap/rtabmap node  not subscribe to any odometry topic anymore (only subsecribe the odom_info or odom_combined_info and rgbd/image topic). 

I feel confused on what is the correct way to modify the odometry topic and the `odom_frame_id`. I am really appreciated if you could give me more advice.

I am also interesting to know what the `guess_frame_id` really do. The document say it uses as the motion model. Then, do we still prefer to do ekf_localization after I set the `guess_frame_id=odom`? 

Thanks.

matlabbe, 2021-10-01:
If you removed all /tf from the bag, you can try without guess_frame_id as it would not exist anymore. If you have 2 map published, your efk_localization node may be not publishing the odom one, but may still publishing the map one. If you set odom_frame_id, rtabmap won't subscribe to an odom topic, you can remove it to subscribe to odom topic.