introlab/rtabmap_ros #647: error: ‘_1’ was not declared in this scope

windelbouwman, 2021-09-21:
The usage of `boost::bind` with the placeholders `_1`, `_2`, etc can result in the compilation error `error: ‘_1’ was not declared in this scope`

For example, one usage of `boost::bind` is here:

https://github.com/introlab/rtabmap_ros/blob/ros2/src/CoreWrapper.cpp#L3432

Is there a reason to have both `boost::bind` and `std::bind`? A solution would be to replace `_1` by `boost::placeholders::_1`.

matlabbe, 2021-09-21:
The reason is that I forgot to change them when I ported the ros1 code to ros2: https://github.com/introlab/rtabmap_ros/commit/e6ea46f5f083f52a22d99404968efde7e785464f

I just committed the fix. If you find other `boost` stuff, it means they may be not migrated to `std` yet.

closed 2021-09-21 by matlabbe

windelbouwman, 2021-09-21:
Thanks! This was not the last occurence, if you do a find-in-all-files for `boost::bind`, you'll find the other occurences (approximately 26 locations).

windelbouwman, 2021-09-21:
Addressed in #648

commit a67df777 (2021-09-21): fixed #647