introlab/rtabmap_ros #650: How to verify ArUco Markers are actually being detected?

wrmichaelmak, 2021-09-27:
Hi Matthieu, 

Thanks for your prompt replies to my previous issues. Just to give some background of my setup, I have successfully integrated ORB-SLAM3 with RTABMAP and now I am trying to add ArUco marker detection. I have set:
```
<param name="RGBD/MarkerDetection" value="true"/>
<param name="Marker/Dictionary" value="10"/>
<param name="Marker/Length" value="0.059"/>
```
in my launch file (in addition with ORB SLAM 3 parameters) and also open up `rtabmapviz`. However, I am not able to verify visually while mapping if ArUco markers are being detected by rtabmap and registered as landmarks.

wrmichaelmak, 2021-09-27:
I have also noticed that when I open the map that I mapped using ROS on rtabmap standalone application, the database seems to have not saved the parameters that I have set for rtabmap via my launch file. This is the warning I got from it:

```
[ WARN] (2021-09-27 14:31:27.735) MainWindow.cpp:5252::openDatabase() Parameter "Marker/Dictionary": database="0" Preferences="10"
[ WARN] (2021-09-27 14:31:27.735) MainWindow.cpp:5252::openDatabase() Parameter "Marker/Length": database="0" Preferences="0.059"
[ WARN] (2021-09-27 14:31:27.735) MainWindow.cpp:5252::openDatabase() Parameter "Marker/MaxDepthError": database="0.01" Preferences="0.2"
[ WARN] (2021-09-27 14:31:27.735) MainWindow.cpp:5252::openDatabase() Parameter "Odom/Strategy": database="0" Preferences="5"
[ WARN] (2021-09-27 14:31:27.736) MainWindow.cpp:5252::openDatabase() Parameter "OdomORBSLAM/VocPath": database="" Preferences="/home/nuc-develop/build/ORB_SLAM3/Vocabulary/ORBvoc.txt"
[ WARN] (2021-09-27 14:31:27.736) MainWindow.cpp:5252::openDatabase() Parameter "RGBD/CreateOccupancyGrid": database="true" Preferences="false"
[ WARN] (2021-09-27 14:31:27.736) MainWindow.cpp:5252::openDatabase() Parameter "RGBD/MarkerDetection": database="false" Preferences="true"
```

matlabbe, 2021-09-28:
It depends where you put the parameters. Those param tags should be inside rtabmap node tag.

In rtabmapviz, to when a tag is detected, look at this [post](http://official-rtab-map-forum.206.s1.nabble.com/Lidar-SLAM-with-loop-closure-tp7846p8115.html) and the referring links.

wrmichaelmak, 2021-09-28:
[rtabmap_launch.txt](https://github.com/introlab/rtabmap_ros/files/7247703/rtabmap_launch.txt)

This is my launch file. I will try it with the param tags inside the rtabmap node tag

matlabbe, 2021-09-30:
Hi, try this in your launch file:
```
	<group ns="rtabmap/rtabmap">
		<param name="Odom/Strategy" 		value="5"/>
		<param name="g2o/Solver" 			value="3"/>
		<param name="Optimizer/Strategy" 	value="2"/>

		<param name="RGBD/MarkerDetection" 	value="true"/>
		<param name="Marker/Dictionary" 	value="10"/>
		<param name="Marker/Length" 		value="0.059"/>

		<param name="Grid/CellSize" 		value="0.09"/>
		<param name="Grid/ClusterSize" 		value="0.12"/>
		<param name="Grid/FlatObstacleDetected" value="false"/>
	</group>
```
It is because the parameters are private, and the node is in rtabmap namespace, so we should put them in `/rtabmap/rtabmap/...`.

wrmichaelmak, 2021-10-01:
Thanks a lot @matlabbe once again, I am able to verify that ArUco markers are now being detected with rtabmapviz! However, I am still seeing that the database did not save my `Odom/Strategy` and `OdomORBSLAM/VocPath` values when I load my map through rtabmap standalone application. Although the database did save my marker detection setting.

My new launch file format as advised by you:
[motion_rover_slam.txt](https://github.com/introlab/rtabmap_ros/files/7266016/motion_rover_slam.txt)
s.

matlabbe, 2021-10-01:
By default rtabmap node doesn't use or save odometry parameters. They are ignored here:
https://github.com/introlab/rtabmap_ros/blob/1ecce1587845ce8383de09ff8640e07ec1127974/src/CoreWrapper.cpp#L306-L317

Is it because you want to know what odometry parameters were used when recording a database?

I could make rtabmap to save them (for post information purpose) by removing the lines above if it would be useful for you.

wrmichaelmak, 2021-10-01:
Hi Matthieu, 

Oh no, it is ok. I do not need to know the odometry parameters used to record the database. I was wondering if it was a problem at my side. 

I will be closing the issue now. I was wondering if possible can I submit a PR in the future to add more information to the existing docs on using ArUco markers and building ORB-SLAM3 support with rtabmap

Thanks a lot!

closed 2021-10-01 by wrmichaelmak

matlabbe, 2021-10-01:
Yes it is possible!

matlabbe, 2025-06-20:
If you are using `rtabmap.launch.py`, the most convenient way to add rtabmap's library parameters is to pass them in `rtabmap_args`:
```bash
ros2 launch rtabmap_launch rtabmap.launch.py rtabmap_args:="-d \
  --Odom/Strategy 5 \
  --g2o/Solver 3 \
  --Optimizer/Strategy 2 \
  --RGBD/MarkerDetection true \
  --Marker/Dictionary 10 \
  --Marker/Length 0.059 \
  --Grid/CellSize 0.09 \
  --Grid/ClusterSize 0.12 \
  --Grid/FlatObstacleDetected false"
```