introlab/rtabmap_ros #651: ROS2: ZED2 camera with RGB + Depth crashes rtabmap on startup

peredwardsson, 2021-09-27:
Jetson AGX Xavier
ROS2 Foxy via Docker based on [Dusty's Dockerfiles](https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ros.foxy)
Zed2 via PCI-E USB

To reproduce:
- Set `subscribe_rgb` and `subscribe_depth` true. Point to appropriate topics. Observe the following crash:
```
# ros2 launch hugo rtabmap.launch.py | grep rtabmap
[INFO] [rtabmap-17]: process started with pid [4829]
[rtabmap-17] 1632743874.265642726 [rtabmap] [INFO] rtabmap: subscribe_depth = true
[rtabmap-17] 1632743874.266235879 [rtabmap] [INFO] rtabmap: subscribe_rgb = true
[rtabmap-17] 1632743874.266321388 [rtabmap] [INFO] rtabmap: subscribe_stereo = false
[rtabmap-17] 1632743874.266370799 [rtabmap] [INFO] rtabmap: subscribe_rgbd = false (rgbd_cameras=1)
[rtabmap-17] 1632743874.266420850 [rtabmap] [INFO] rtabmap: subscribe_odom_info = false
[rtabmap-17] 1632743874.266467156 [rtabmap] [INFO] rtabmap: subscribe_user_data = false
[rtabmap-17] 1632743874.266539865 [rtabmap] [INFO] rtabmap: subscribe_scan = false
[rtabmap-17] 1632743874.266586267 [rtabmap] [INFO] rtabmap: subscribe_scan_cloud = false
[rtabmap-17] 1632743874.266632542 [rtabmap] [INFO] rtabmap: queue_size    = 10
[rtabmap-17] 1632743874.266679872 [rtabmap] [INFO] rtabmap: approx_sync   = false
[rtabmap-17] 1632743874.273861142 [rtabmap] [INFO] rtabmap(maps): map_filter_radius          = 0.000000
[rtabmap-17] 1632743874.273999134 [rtabmap] [INFO] rtabmap(maps): map_filter_angle           = 30.000000
[rtabmap-17] 1632743874.274065282 [rtabmap] [INFO] rtabmap(maps): map_cleanup                = true
[rtabmap-17] 1632743874.274126149 [rtabmap] [INFO] rtabmap(maps): map_always_update          = false
[rtabmap-17] 1632743874.274176936 [rtabmap] [INFO] rtabmap(maps): map_empty_ray_tracing      = true
[rtabmap-17] 1632743874.274229547 [rtabmap] [INFO] rtabmap(maps): cloud_output_voxelized     = true
[rtabmap-17] 1632743874.274280238 [rtabmap] [INFO] rtabmap(maps): cloud_subtract_filtering   = false
[rtabmap-17] 1632743874.274343249 [rtabmap] [INFO] rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[rtabmap-17] 1632743874.341379963 [rtabmap] [INFO] rtabmap: frame_id      = base_link
[rtabmap-17] 1632743874.341533956 [rtabmap] [INFO] rtabmap: odom_frame_id = odom
[rtabmap-17] 1632743874.341591495 [rtabmap] [INFO] rtabmap: map_frame_id  = map
[rtabmap-17] 1632743874.341641578 [rtabmap] [INFO] rtabmap: use_action_for_goal  = false
[rtabmap-17] 1632743874.341701230 [rtabmap] [INFO] rtabmap: tf_delay      = 0.050000
[rtabmap-17] 1632743874.341762001 [rtabmap] [INFO] rtabmap: tf_tolerance  = 0.100000
[rtabmap-17] 1632743874.341814132 [rtabmap] [INFO] rtabmap: odom_sensor_sync   = false
[rtabmap-17] 1632743874.463858273 [rtabmap] [INFO] Setting RTAB-Map parameter "Grid/RangeMax"="10.0"
[rtabmap-17] 1632743874.467973321 [rtabmap] [INFO] Setting RTAB-Map parameter "Grid/RayTracing"="True"
[rtabmap-17] 1632743874.821971331 [rtabmap] [INFO] Update RTAB-Map parameter "Mem/IncrementalMemory"="false" from database
[rtabmap-17] 1632743875.186967337 [rtabmap] [INFO] RTAB-Map detection rate = 1.000000 Hz
[rtabmap-17] 1632743875.190368522 [rtabmap] [INFO] rtabmap: Using database from "/root/.ros/rtabmap.db" (965 MB).
[rtabmap-17] 1632743889.197649460 [rtabmap] [INFO] rtabmap: 2D occupancy grid map loaded (319x511).
[rtabmap-17] 1632743890.059005332 [rtabmap] [INFO] rtabmap: Database version = "0.20.13".
[rtabmap-17] 1632743890.132928387 [rtabmap] [INFO] Setup callbacks
[rtabmap-17] 1632743890.133119501 [rtabmap] [INFO] Setup depth callback
[rtabmap-17] 1632743890.165767009 [rtabmap] [INFO] 
[rtabmap-17] rtabmap subscribed to (exact sync):
[rtabmap-17]    /sensors/zed_front/rgb/image_rect_color,
[rtabmap-17]    /sensors/zed_front/depth/depth_registered,
[rtabmap-17]    rgb/camera_info
[rtabmap-17] terminate called after throwing an instance of 'UException'
[rtabmap-17]   what():  [FATAL] (2021-09-27 11:58:10.381) Transform.cpp:191::inverse() Condition (invertible) not met! [This transform is not invertible! xyz=[null] rpy=[null] 
[rtabmap-17] [0.000000 0.000000 0.000000 0.000000;
[rtabmap-17]  0.000000 0.000000 0.000000 0.000000;
[rtabmap-17]  0.000000 0.000000 0.000000 0.000000;
[rtabmap-17]  0 0 0 1]]
[ERROR] [rtabmap-17]: process has died [pid 4829, exit code -6, cmd '/opt/ros/foxy/install/lib/rtabmap_ros/rtabmap --ros-args -r __node:=rtabmap --params-file /tmp/launch_params_ovly2iry -r rgbd_image:=/sensors/rtabmap/rgbd -r depth/image:=/sensors/zed_front/depth/depth_registered -r rgb/camera_info:=/sensors/zed_front/rgb/camera_info -r rgb/image:=/sensors/zed_front/rgb/image_rect_color -r left/image_rect:=/sensors/zed_front/left/image_rect_color -r left/camera_info:=/sensors/zed_front/left/camera_info -r right/image_rect:=/sensors/zed_front/right/image_rect_color -r right/camera_info:=/sensors/zed_front/right/camera_info'].
```
(sidenote: rtabmap (and all nodelets I've tested so far) reports subscribing to the incorrect camera info topic. It actually subscribes to the correct one, but does not report the correct one)
- Crash also occurs when using the `rgbd_sync` nodelet to parse the data into an RGBD message. The nodelet works fine, but rtabmap crashes.
 
Echoing one such RGBD message (with no-arr) gives the following result:
```
---
header:
  stamp:
    sec: 1632743708
    nanosec: 332855704
  frame_id: ''
rgb_camera_info:
  header:
    stamp:
      sec: 1632743708
      nanosec: 332855704
    frame_id: ''
  height: 360
  width: 640
  distortion_model: plumb_bob
  d: '<sequence type: double, length: 5>'
  k: '<array type: double[9]>'
  r: '<array type: double[9]>'
  p: '<array type: double[12]>'
  binning_x: 0
  binning_y: 0
  roi:
    x_offset: 0
    y_offset: 0
    height: 0
    width: 0
    do_rectify: false
depth_camera_info:
  header:
    stamp:
      sec: 1632743708
      nanosec: 332855704
    frame_id: ''
  height: 360
  width: 640
  distortion_model: plumb_bob
  d: '<sequence type: double, length: 5>'
  k: '<array type: double[9]>'
  r: '<array type: double[9]>'
  p: '<array type: double[12]>'
  binning_x: 0
  binning_y: 0
  roi:
    x_offset: 0
    y_offset: 0
    height: 0
    width: 0
    do_rectify: false
rgb:
  header:
    stamp:
      sec: 1632743708
      nanosec: 332855704
    frame_id: zed2_front_left_camera_optical_frame
  height: 360
  width: 640
  encoding: bgra8
  is_bigendian: 0
  step: 2560
  data: '<sequence type: uint8, length: 921600>'
depth:
  header:
    stamp:
      sec: 1632743708
      nanosec: 332855704
    frame_id: zed2_front_left_camera_optical_frame
  height: 72
  width: 128
  encoding: 32FC1
  is_bigendian: 0
  step: 512
  data: '<sequence type: uint8, length: 36864>'
rgb_compressed:
  header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: ''
  format: ''
  data: '<sequence type: uint8, length: 0>'
depth_compressed:
  header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: ''
  format: ''
  data: '<sequence type: uint8, length: 0>'
---
```

I am unable to use multiple cameras as input due to this issue.

matlabbe, 2021-10-01:
For some reasons, rtabmap is trying to invert a null transform:
```
 what():  [FATAL] (2021-09-27 11:58:10.381) Transform.cpp:191::inverse() Condition (invertible) not met! [This transform is not invertible! xyz=[null] rpy=[null] 
```
Can you share the launch you used?

About the wrong topic names, it is a not convenient issue when getting topic resolved names from messages_filters::Subscriber objects. The [messages_filters::Subscriber::getTopic()](https://github.com/ros2/message_filters/blob/ee9f798a156b4c7a10ee34595c5e50e22733f814/include/message_filters/subscriber.h#L322-L325) funciton returns the original topic name (the one hard-coded), while [messages_filters::Subscriber.getSubscriber()->get_topic_name()](https://github.com/ros2/rclcpp/blob/893b9b4f8269f4dbee3fc11b821961adc1822860/rclcpp/src/rclcpp/subscription_base.cpp#L100-L104) returns the actual resolved remap name. Also found that [image_transport::getTopic()](https://github.com/ros-perception/image_common/blob/b2d8d5cbaf3edad81f02c572f073368f7ce17615/image_transport/include/image_transport/subscriber.hpp#L92) returns the right resolved topic name, but doesn't have `get_topic_name()`... so I had to do [some c++11 magic](https://stackoverflow.com/questions/257288/templated-check-for-the-existence-of-a-class-member-function) to continue to use my preprocessor directives independently of the type of the Subscriber. This is fixed in this [commit](https://github.com/introlab/rtabmap_ros/commit/ccf152d8777af067569e89e5d0e5c0c985b6409b).

cheers,
Mathieu

matlabbe, 2021-10-05:
I [merged](https://github.com/introlab/rtabmap_ros/commit/af1c7edeefb8aaaa99fc509891cb652c78ea9195) master branch to ros2 branch for possible fixes for the fatal error you had. I tested with a ZED2 camera like the following and I was not able to reproduce the error.

Note: On ROS Foxy, make sure to [rebuild image_common](https://github.com/stereolabs/zed-ros2-wrapper#image-transport-and-topic-subscriptions), otherwise images won't be published by zed wrapper

## RTAB-Map with ZED odometry
For RTAB-Map for mapping and ZED for odometry, you should disable map->odom frame published by zed node by setting this [publish_map_tf](https://github.com/stereolabs/zed-ros2-wrapper/blob/b512dce6ad4565f4770273995b147122e735ca0f/zed_wrapper/config/common.yaml#L60) parameter to false in zed's `common.yaml`. Launch (based on those ros1 [RGB-D](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping) and [stereo](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping) examples with ZED odometry):

```bash
ros2 launch zed_wrapper zed2.launch.py

# RGB-D mode with ZED odometry:
ros2 launch rtabmap_ros rtabmap.launch.py \
    rtabmap_args:="--delete_db_on_start" \
     rgb_topic:=/zed2/zed_node/rgb/image_rect_color \
    depth_topic:=/zed2/zed_node/depth/depth_registered \
    camera_info_topic:=/zed2/zed_node/depth/camera_info \
    odom_topic:=/zed2/zed_node/odom \
    visual_odometry:=false \
    frame_id:=base_link \
    approx_sync:=false \
    rgbd_sync:=true \
    approx_rgbd_sync:=false \
    imu_topic:=/zed2/zed_node/imu/data

# Stereo mode with ZED odometry:
ros2 launch rtabmap_ros rtabmap.launch.py \
    rtabmap_args:="--delete_db_on_start --Vis/CorFlowMaxLevel 5 --Stereo/MaxDisparity 200" \
    stereo_namespace:=/zed2/zed_node \
    right_image_topic:=/zed2/zed_node/right/image_rect_color \
    odom_topic:=/zed2/zed_node/odom \
    visual_odometry:=false \
    stereo:=true \
    frame_id:=base_link \
    imu_topic:=/zed2/zed_node/imu/data
 ```
Adjust `zed2` namespace depending on the zed launch file used.

## With RTAB-Map odometry
To use odometry from RTAB-Map, you should disable positional tracking of zed node by setting this [parameters](https://github.com/stereolabs/zed-ros2-wrapper/blob/b512dce6ad4565f4770273995b147122e735ca0f/zed_wrapper/config/common.yaml#L58-L60) parameter to false in zed's `common.yaml`. Launch (based on those ros1 [RGB-D](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping) and [stereo](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping) examples with RTAB-Map odometry):
```bash
ros2 launch zed_wrapper zed2.launch.py

#RGB-D mode:
ros2 launch rtabmap_ros rtabmap.launch.py \
    rtabmap_args:="--delete_db_on_start" \
    rgb_topic:=/zed2/zed_node/rgb/image_rect_color \
    depth_topic:=/zed2/zed_node/depth/depth_registered \
    camera_info_topic:=/zed2/zed_node/rgb/camera_info \
    frame_id:=base_link \
    approx_sync:=false \
    imu_topic:=/zed2/zed_node/imu/data \
    wait_imu_to_init:=true

# Stereo mode:
ros2 launch rtabmap_ros rtabmap.launch.py \
   rtabmap_args:="--delete_db_on_start --Vis/CorFlowMaxLevel 5 --Stereo/MaxDisparity 200" \
   stereo_namespace:=/zed2/zed_node \
   right_image_topic:=/zed2/zed_node/right/image_rect_color \
   stereo:=true \
   frame_id:=base_link \
   imu_topic:=/zed2/zed_node/imu/data \
   wait_imu_to_init:=true
```
Remove `wait_imu_to_init` if the zed camera used doesn't have an IMU. Adjust `zed2` namespace depending on the zed launch file used.

peredwardsson, 2021-10-11:
Indeed, it must have. Maybe some part of the Dockerfile was cached even though I told it not to cache. Now, onto the next step. 

RGBD sync nodelets do not receive input from Zed node because the reliability policy does not match. Messages of this type shows up
```
[rgbd_sync-3] 1633947551.470787365 [rgbd_sync] [WARN] New publisher discovered on this topic, offering incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
```
rgbd_sync wants to receive a message of type **reliable**, but the zed node (though adjustable) does not provide such, [nor is it recommended for any sensor data](https://design.ros2.org/articles/qos.html) to advertise reliable data output. Verbose topic info shows 

```
# ros2 topic info /sensors/zed_back/rgb/image_rect_color --verbose
Type: sensor_msgs/msg/Image

Publisher count: 1

Node name: zed_back
Node namespace: /sensors
Topic type: sensor_msgs/msg/Image
Endpoint type: PUBLISHER
GID: 41.44.10.01.77.1a.28.f5.92.93.69.89.00.00.2c.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT
  Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
  Lifespan: 9223372036854775807 nanoseconds
  Deadline: 9223372036854775807 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 9223372036854775807 nanoseconds

Subscription count: 1

Node name: rgbd_sync
Node namespace: /
Topic type: sensor_msgs/msg/Image
Endpoint type: SUBSCRIPTION
GID: 92.e1.10.01.a7.7f.87.b6.dd.d2.e3.37.00.00.17.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
  Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
  Lifespan: 9223372036854775807 nanoseconds
  Deadline: 9223372036854775807 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 9223372036854775807 nanoseconds
```

Do you recommend that I change the zed node publisher settings, or does it make more sense for rtabmap to use the sensor data qos for sensor input?