introlab/rtabmap_ros #654: proximity detection problem (loop closure for lidar)

robertsenps, 2021-10-06:
Hi @matlabbe ,

I'm currently using rtabmap/icp_odometry with leishen lidar 16 channels with `test_velodyne.launch` as a template. I'm using 3d lidar only without any camera and imu. 

I found two problems:

1. Sometimes rtabmap can't detect loop closure even the robot comes back to the previous pose. Do you know why? Should I change some of the parameters?
2. Sometimes the robot detects a lot of loop closure so the map that is previously correct (align) will be unaligned. Should I change the parameters? how can I limit the number of loop closure? Normally, it should be separated by a distance to search for a loop closing candidate again?

Thank you. Any help from you will be very helpful for me and the community.

Regards,

This is my parameters:
* **rtabmap/icp_odometry**

>**Icp:**
      Strategy:                  1        #Libpointmatcher
      PointToPlane:        true
      PointToPlaneK:       20
      PointToPlaneRadius:  0
      PointToPlaneMinComplexity: 0.0
      RangeMin:            0.3
      RangeMax:            20
      Iterations:          30
      VoxelSize:           0.05
      DownsamplingStep:    1
      Epsilon:             0.001
      MaxTranslation:      2.0
      MaxCorrespondenceDistance: 1.5
      PMOutlierRatio:      0.7
      CorrespondenceRatio: 0.2  
      Force4DoF:          false
**Odom:**     
      ScanKeyFrameThr:       0.8
      Strategy:              0
**OdomF2M:**
      ScanSubtractRadius: 0.05
      ScanMaxSize:        30000     

* **rtabmap/rtabmap**
>
frame_id: "base_link"
subscribe_depth: false
subscribe_rgb: false
subscribe_scan_cloud: true
approx_sync: true
**Rtabmap**:
  DetectionRate: 4
**RGBD**:
  NeighborLinkRefining: false
  AngularUpdate: 5.0
  LinearUpdate: 0.5
  #Proximity Detection
  ProximityBySpace: true
  ProximityMaxGraphDepth: 0
  ProximityPathMaxNeighbors: 1
  LocalRadius: 5.0
  ProximityPathFilteringRadius: 2.0
  ProximityMaxPaths: 2
  ProximityOdomGuess: true
  ProximityAngle: 180
**Kp**:
  MaxFeatures: -1
**Mem**:
  NotLinkedNodesKept: false
  STMSize: 30
**Reg**:
  Strategy: 1
**Optimizer**:
  GravitySigma: 0.3
**Grid**:
  CellSize:                  0.05
  RangeMax:                  30
  GroundIsObstacle:          false
  FromDepth:                 false
  RayTracing:                false
  3D:                        true 
  PreVoxelFiltering:         true
  #PCL CropBox filtering 
  FootprintLength:           0.81
  FootprintWidth:            0.635
  FootprintHeight:           0.7
  #Normal Segmentation 
  NormalK:                   20
  MaxGroundAngle:            45
  MaxGroundHeight:           0.05
  #Clustering  
  ClusterRadius:             1.0
  MinClusterSize:            30
  #Flat obstacle 
  FlatObstacleDtected:       true
  MaxObstacleHeight:         2.0
  #After segmentation filtering 
  NoiseFilteringRadius:       0.0
  NoiseFilteringMinNeighbors: 0
**GridGlobal**:
  #Auxiliary function 
  Eroded:                   true
  FootprintRadius:           0.255
  OccupancyThr:             0.5
**Icp**:
  Strategy:                  0        #Libpointmatcher
  VoxelSize:                0.05
  PointToPlaneK:            20
  PointToPlaneRadius:       0
  PointToPlane:             true
  Iterations:               100
  Epsilon:                  0.001
  MaxTranslation:           2.0
  MaxCorrespondenceDistance: 1.5
  PMOutlierRatio:           0.7
  CorrespondenceRatio:      0.4
  PointToPlaneGroundNormalsUp:  0.8
  RangeMin:          0.3
  RangeMax:          100
```
![image](https://user-images.githubusercontent.com/27759735/136202189-9313ae0a-4a6d-4545-9d25-a348ffdfaa4c.png)

matlabbe, 2021-10-06:
Can you share the database? You can set `RGBD/LocalRadius` smaller (default 10 meters) to avoid trying to loop with too far scans from the current position. `RGBD/ProximityMaxGraphDepth` can also be set to avoid loop closures on too old scans.

Is rtabmap built with libpointmatcher?

robertsenps, 2021-10-07:
Thanks for the suggestions. I'm working on the big map, so is it ok if I want to get loop closures with the old scan? any drawbacks? 

Sure, this is the db file: [rtabmap.db](https://drive.google.com/file/d/1_bsxmmvRCkXsv-Vq1QK7aLMtNSAdw3Yr/view?usp=sharing)

Yes, my rtabmap is built with libpointmatcher. I'm curious, is libpointmatcher give a huge difference rather than PCL ICP?

Could you tell me what can be improved from the db file? Probably the new registered pointcloud? looks like it doesn't fit the map well?

matlabbe, 2021-10-10:
I don't have the odometry statistics in the database, but on rtabmap side, the node used a lot of computation time to process some scans. This would affect icp_odometry, making it lag and that would explain why odometry was good (almost perfect) at the beggining, and more and more the map is bigger, it was drifting a lot more, like it was not able to process the scans faster enough. You can check in the terminal the actual processing time used by icp_odometry (or in rtabmapviz, show Statistics->Odometry panel->plot Estimation Time). It should stay under 100 ms for good odometry. You can increase voxel size of odometry node, not sure what parameters did you use for `test_velodyne.launch` (default resolution is 10 cm). As your robot is 2D, I would set `Reg/Force3DoF=true` for both odometry and rtabmap nodes.

Here is the computation time detail from rtabmap node:
[image]

Do you need the occupancy grid? or just the point cloud is enough? You can disable occupancy grid generation with `RGBD/CreateOccupancyGrid=false`. If you want to keep the occupancy grid at 5 cm cell size, setting `Mem/LaserScanVoxelSize` to 0.05 or even 0.10 will help a lot to reduce the computation time. I would limit also the range of the local occupancy grid, with `Grid/RangeMax=5` instead of 30 meters. The Icp/VoxelSize is also set at 5 cm in your map, I would increase it to 20 cm to make proximity detection faster.

The biggest difference between libpointmatcher ICP and PCL ICP (as implemented in RTAB-Map), is that PCL's ICP tries to minimize the distance between **all** points of the 2 clouds, so points from different point of view will affect the global registration. With libpointmatcher, `Icp/OutlierRatio` (default 85%) can be used, ICP can ignore some wrong correspondences in the final registration, which makes the approach a lot more robust to dynamic objects or simply different parts of the point clouds not really overlapping.

closed 2021-11-24 by robertsenps