introlab/rtabmap_ros #663: Questions related to multi-session mapping

szx0112, 2021-10-19:
Hi @matlabbe , I just have several quick questions related to the multi-session mapping. I really appreciate if you could give me the advice.

1. Based on the documents, if there is a _rtabmap.db_ available, and we do not create new db when starts the _rtabmap_ node. We can build the new map based on the previous map. In rtabmapviz, Edit->, there are options _"download all clouds"_ and _"download graph only"_. Do these options actually affect the multi-session mapping results (e.g., loop closure disabled/enabled), or it is just for visualization purposes?

2.  If we build the new map based on the previous map. Does the new map also update/correct the previous map (if visit a previous location and close some loops that is not available previously)? This also affects the selection of the order of multiple maps. For example, if I have three passes to map an area. The 1st pass maps the 80% of the area. Each of the 2nd and 3rd pass has large area of overlap with the 1st pass, but the inter-overlap between 2nd and 3rd pass is limited. Thus, should we select the mapping process as 1st -> 2nd -> 3rd (2nd and 3rd are interchangeable) to incrementally build the map, Or each of 2nd and 3rd map is build only on 1st map (not incrementally), and then these two maps (1+2 and 1+3) can be fused with ICP or other registration methods. But the second strategy only hold if the 1st map never change.

3. In order to build a new map based on the previous map, I have to copy/paste/rename the existing db (if saved with other names) as 'rtabmap.db' and put it at the right directory. Is it possible to directly load the db in the launch.file (for example in _rtabmap.launch_)? So we can skip to manually edit the database (e.g., rename, copy, paste) each time. 

Thanks.

szx0112, 2021-11-01:
@matlabbe Thanks for the comments. 

I have a follow up question. Is it possible to just output a partial of the map?

Like above mentioned, the original multi-session map is built with 3 passes (1st -> 2nd -> 3rd), I can export the map at the end of each session: 1) map with 1st pass only, 2) map with 1st + 2nd passes combined, and 3) map with all three passes (1st + 2nd + 3rd).

But what if we want to export the result (i.e., map) from the 2nd + 3rd passes only? or even 2nd/3rd map individually?

Noted I always need the 1st pass as the starting point of the mapping because only the 1st pass contains loop closure. Mapping 2nd/3rd maps alone always cause large distortions (no loop closures).

Thanks.

matlabbe, 2021-11-04:
In rtabmap-databaseViewer, you can export/show maps of each individual session. In Graph View, uncheck "Span to all maps" and select a root id from one of the session. Only that session will be assembled.
[image]

If your map is not too huge, another manual way is in rtabmap, show Map Visibility panel, then uncheck individually the nodes you don't want to show/export.
[image]

Other option is if you want to export only nodes in a specific region (independently of the session), you can use the Nodes filtering option when exporting
[image]
:

szx0112, 2021-11-15:
The 1st option seemly works best for my case. Thanks.

closed 2021-11-15 by szx0112

fbliman, 2023-08-22:
> In rtabmap-databaseViewer, you can export/show maps of each individual session. In Graph View, uncheck "Span to all maps" and select a root id from one of the session. Only that session will be [...]

Regarding your first comment, when I look for more loop closures (I have only one seesion with 150 maps id's) it only looks for loop closures in the map id i choose the root (with or without the checkbox) I have some isolated nodes (only joined by imu odometry) that aren't checked against the rest of the map
[image]
[image]
[image]

matlabbe, 2023-08-27:
Do you have multiple sessions? Are nodes from different sessions? The automatic detection for more loop closures only work between sessions that have at least already one loop closure (sessions already linked).

fbliman, 2023-08-28:
It's only one session, but the VO get lost several times, so I have multiple maps id in the same session. The data is not the best (is from undergroud seweage with low light) but from looking at the frames there are multiple candidate for loop closure from different maps, but it seems it only looks inside each indiviudal map.
When I change the seed I get loop clourse only from that map

matlabbe, 2023-08-29:
The "Detect more loop closures" function only works between nodes of same session and between sessions that are linked by at least one loop closure. In general, if I have two disjoint sessions in a database, I would manually add a loop closure between them, then call "Detect more loop closures" to add more loop closures automatically.