introlab/rtabmap_ros #664: RTABMAP version 0.20.14 build from source with Odometry node carsh.

AlysonQ, 2021-10-20:
Hello,
[image]

I was trying to build RTABMAP from source code recently. 
I found that the /rtabmap/rtabmap node is fine to do every function like map_server(load map, just open rtabmap application... and so on) but the /rtabmap/rgbd_odometry and /rtabmap/stereo_odometry will crash everytime I roslaunch.

Here are some information of my setting：
Ubuntu 18.04
ROS Melodic
x86 platform

Build from source steps：
cd rtabmap/build
cmake -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel  -DWITH_G2O=OFF ..
make -j4
make install

would you give me some advice or maybe I should try version between 0.20.9~0.20.14?

Thank you,
Alyson

matlabbe, 2021-10-21:
I see it is building with ORB_SLAM2, it is why g2o is available (g2o built inside ORB_SLAM2 is used). However, you should make sure you are using this [patch](https://gist.github.com/matlabbe/c10403c5d44af85cc3585c0e1c601a60) for ORB_SLAM2, to not have malloc problems with Eigen.

matlabbe, 2021-11-03:
It is crashing in `libimage_transport_plugins.so`. There is something wrong with the ros binaries. I suggest to do a `sudo apt upgrade` to make sure all ros binaries are at the latest version. Then delete your `~/catkin_ws/build` directory and redo a `catkin_make`.

matlabbe, 2023-05-14:
The `malloc.c:2379: sysmalloc: Assertion '(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed. ` is caused by latest realsense2 library when rtabmap is built with it. The workaround to make the ros node not crash on this malloc assertion is to build rtabmap library without realsense2 support:
```
cd rtabmap/build
cmake -DWITH_REALSENSE2=OFF ..
make
```
Related issue: https://github.com/introlab/rtabmap_ros/issues/881

closed 2023-05-14 by matlabbe