introlab/rtabmap_ros #668: Reg/Strategy=1 giving errors

Nosille, 2021-10-29:
Recently switched to the latest noetic-devel branch and started having issues with Reg/Strategy=1.  I have confirmed that Reg/Strategy=0 works without issue.

I get the following message anytime it finds a possible loop closure when using Reg/Strategy=1:

[ WARN] (2021-10-29 16:49:46.003) Rtabmap.cpp:2770::process() Rejected loop closure 61 -> 256: Missing correspondences for registration (61->256). fromWords = 0 fromImageEmpty=1 toWords = 0 toImageEmpty=0
[ INFO] [1635544186.111398816, 1633351931.987364697]: Assembled 1 obstacle and 0 ground clouds (69737 points, 0.005741s)
[ INFO] [1635544186.112797853, 1633351931.989892610]: rtabmap (256): Rate=1.00s, Limit=3.000s, Conversion=0.0138s, RTAB-Map=0.6160s, Maps update=0.0260s pub=0.0071s (local map=214, WM=214)
[ WARN] (2021-10-29 16:49:48.393) Rtabmap.cpp:2770::process() Rejected loop closure 62 -> 257: Missing correspondences for registration (62->257). fromWords = 0 fromImageEmpty=1 toWords = 0 toImageEmpty=0
[ INFO] [1635544188.487804249, 1633351933.175667385]: Assembled 1 obstacle and 0 ground clouds (70267 points, 0.004138s)
[ INFO] [1635544188.488492958, 1633351933.175667385]: rtabmap (257): Rate=1.00s, Limit=3.000s, Conversion=0.0098s, RTAB-Map=0.6121s, Maps update=0.0191s pub=0.0048s (local map=215, WM=215)
[ WARN] (2021-10-29 16:49:50.459) Rtabmap.cpp:2770::process() Rejected loop closure 64 -> 258: Missing correspondences for registration (64->258). fromWords = 0 fromImageEmpty=1 toWords = 0 toImageEmpty=0
[ INFO] [1635544190.565701025, 1633351934.216431607]: Assembled 1 obstacle and 0 ground clouds (70812 points, 0.005342s)
[ INFO] [1635544190.566373315, 1633351934.216431607]: rtabmap (258): Rate=1.00s, Limit=3.000s, Conversion=0.0159s, RTAB-Map=0.6358s, Maps update=0.0252s pub=0.0060s (local map=216, WM=216)
[ WARN] (2021-10-29 16:49:52.411) Rtabmap.cpp:2770::process() Rejected loop closure 65 -> 259: Missing correspondences for registration (65->259). fromWords = 0 fromImageEmpty=1 toWords = 0 toImageEmpty=0

matlabbe, 2021-11-04:
It is like there are no visual features in the corresponding images. Can you share the database or the info?
```
rtabmap-info rtabmap.db
```

Nosille, 2021-11-19:
Sorry for the slow response.  We lost a video card in the computer and have been waiting on a replacement.  However, after the video card replacement, I tried to record a short database file to send you, and I could not repeat the issue.  The day I reported it, it was a consistent issue that only happened when Reg/Strategy = 1.  If I run across it again, I will provide a database.

matlabbe, 2021-11-21:
Alright, re-open the issue if needed.

closed 2021-11-21 by matlabbe

Nosille, 2021-11-30:
OK.  I finally managed to repeat the error.  Warning we are using four 4K cameras and three lidar so the file is large.  

https://drive.google.com/file/d/1KXpwqDsbAXB89B-gQbwiV8DGu6qgMi8p/view?usp=sharing

matlabbe, 2021-12-05:
Thanks a lot for this database, I fixed the warning you saw in this [commit](https://github.com/introlab/rtabmap/commit/3275bcc0da02fed49286406ccbca0c8cb284eae4). However, as you set RGBD/LoopClosureReextractFeatures=true, the loop closure transformation estimation is quite long:
[image]

With optimizing Icp parameters to use less computation time, here the final log (this was done on a laptop):
[image]

For reference, here are my overloaded parameters:
`./rtabmap-reprocess --uwarn --RGBD/CreateOccupancyGrid false --Icp/VoxelSize 0.3 --RGBD/ProximityMaxPaths 1 --Icp/Iterations 10 --RGBD/LoopClosureReextractFeatures false ~/Downloads/rtabmap4x4K.db output.db`