introlab/rtabmap_ros #670: Rtabmapviz chrashes with error 'wait_for_tranform' has invalid type in ROS2 Foxy

ericdepotter, 2021-11-01:
Hi,

I tried launching the rtabmap module on ROS2 Foxy. It looks like the algorithm itself is working correctly, but the visualisation keeps crashing. In the logs I see the following error:

```
[rtabmapviz-3] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
[rtabmapviz-3]   what():  parameter 'wait_for_transform' has invalid type: expected [double] got [string]
```

I launched it without specifying this parameter, so it has it's default value of 0.2.

How can I resolve this issue?

matlabbe, 2021-11-01:
How do you launch rtabmapviz? Tried this:
```
ros2 run rtabmap_ros rtabmapviz --ros-args -p wait_for_transform:=0.5
```
and no error.
If I do
```
ros2 run rtabmap_ros rtabmapviz --ros-args -p wait_for_transform:="0,5"
```
your error shows up but it is normal as it cannot be converted to double.

ericdepotter, 2021-11-01:
I launch rtabmapviz using the launch file with the following command:

`ros2 launch rtabmap_ros rtabmap.launch.py 
    args:="--delete_db_on_start" 
    stereo:=true 
    left_image_topic:=/left/image 
    right_image_topic:=/right/image 
    left_camera_info_topic:=/left/camera_info 
    right_camera_info_topic:=/right/camera_info 
    approx_sync:=true 
    visual_odometry:=true`

I noticed I was using the master branch of the rtabmap repository, so I assume the error may be related to that. However I tried building the foxy-devel branch, but I get the following error:

`--- stderr: rtabmap                               
In file included from /usr/local/include/depthai/device/CalibrationHandler.hpp:5,
                 from /usr/local/include/depthai/depthai.hpp:9,
                 from /home/eric/ROS/dev_ws/src/rtabmap/corelib/src/../include/rtabmap/core/camera/CameraDepthAI.h:40,
                 from /home/eric/ROS/dev_ws/src/rtabmap/corelib/src/camera/CameraDepthAI.cpp:28:
/usr/local/include/depthai-shared/common/CameraBoardSocket.hpp:3:10: fatal error: nlohmann/json.hpp: No such file or directory
    3 | #include <nlohmann/json.hpp>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
`

I assume this is an issue in the rtabmap repository.

matlabbe, 2021-11-04:
I am using `master` branch of rtabmap and `ros2` branch of rtabmap_ros, no errors:
```
ros2 launch rtabmap_ros rtabmap.launch.py args:="--delete_db_on_start" stereo:=true left_image_topic:=/left/image right_image_topic:=/right/image left_camera_info_topic:=/left/camera_info right_camera_info_topic:=/right/camera_info approx_sync:=true visual_odometry:=true
[INFO] [launch]: All log files can be found below /home/mathieu/.ros/log/2021-11-04-17-21-06-732211-mathieu-XPS-15-7590-121299
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [stereo_odometry-1]: process started with pid [121302]
[INFO] [rtabmap-2]: process started with pid [121304]
[INFO] [rtabmapviz-3]: process started with pid [121306]
[rtabmap-2] [INFO] [1636060867.602206792] [rtabmap.rtabmap]: rtabmap(maps): map_filter_radius          = 0.000000
[rtabmap-2] [INFO] [1636060867.602826118] [rtabmap.rtabmap]: rtabmap(maps): map_filter_angle           = 30.000000
[rtabmap-2] [INFO] [1636060867.603071880] [rtabmap.rtabmap]: rtabmap(maps): map_cleanup                = true
[rtabmap-2] [INFO] [1636060867.603289083] [rtabmap.rtabmap]: rtabmap(maps): map_always_update          = false
[rtabmap-2] [INFO] [1636060867.603500070] [rtabmap.rtabmap]: rtabmap(maps): map_empty_ray_tracing      = true
[rtabmap-2] [INFO] [1636060867.603708273] [rtabmap.rtabmap]: rtabmap(maps): cloud_output_voxelized     = true
[rtabmap-2] [INFO] [1636060867.603918341] [rtabmap.rtabmap]: rtabmap(maps): cloud_subtract_filtering   = false
[rtabmap-2] [INFO] [1636060867.604124658] [rtabmap.rtabmap]: rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[rtabmap-2] [INFO] [1636060867.604363109] [rtabmap.rtabmap]: rtabmap(maps): octomap_tree_depth         = 16
[stereo_odometry-1] [INFO] [1636060867.617419376] [rtabmap.stereo_odometry]: Odometry: frame_id               = base_link
[stereo_odometry-1] [INFO] [1636060867.618003340] [rtabmap.stereo_odometry]: Odometry: odom_frame_id          = odom
[stereo_odometry-1] [INFO] [1636060867.618246953] [rtabmap.stereo_odometry]: Odometry: publish_tf             = true
[stereo_odometry-1] [INFO] [1636060867.618461008] [rtabmap.stereo_odometry]: Odometry: wait_for_transform     = 0.200000
[stereo_odometry-1] [INFO] [1636060867.618700296] [rtabmap.stereo_odometry]: Odometry: initial_pose           = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000
[stereo_odometry-1] [INFO] [1636060867.618915488] [rtabmap.stereo_odometry]: Odometry: ground_truth_frame_id  = 
[stereo_odometry-1] [INFO] [1636060867.619126690] [rtabmap.stereo_odometry]: Odometry: ground_truth_base_frame_id = 
[stereo_odometry-1] [INFO] [1636060867.619336321] [rtabmap.stereo_odometry]: Odometry: config_path            = 
[stereo_odometry-1] [INFO] [1636060867.619542964] [rtabmap.stereo_odometry]: Odometry: publish_null_when_lost = true
[stereo_odometry-1] [INFO] [1636060867.619748322] [rtabmap.stereo_odometry]: Odometry: guess_frame_id         = 
[stereo_odometry-1] [INFO] [1636060867.619952176] [rtabmap.stereo_odometry]: Odometry: guess_min_translation  = 0.000000
[stereo_odometry-1] [INFO] [1636060867.620159220] [rtabmap.stereo_odometry]: Odometry: guess_min_rotation     = 0.000000
[stereo_odometry-1] [INFO] [1636060867.620376325] [rtabmap.stereo_odometry]: Odometry: guess_min_time         = 0.000000
[stereo_odometry-1] [INFO] [1636060867.620583653] [rtabmap.stereo_odometry]: Odometry: expected_update_rate   = 0.000000 Hz
[stereo_odometry-1] [INFO] [1636060867.620789284] [rtabmap.stereo_odometry]: Odometry: max_update_rate        = 0.000000 Hz
[stereo_odometry-1] [INFO] [1636060867.620999582] [rtabmap.stereo_odometry]: Odometry: wait_imu_to_init       = false
[stereo_odometry-1] [INFO] [1636060867.621246836] [rtabmap.stereo_odometry]: Odometry: stereoParams_=1 visParams_=1 icpParams_=0
[rtabmap-2] [INFO] [1636060867.634734412] [rtabmap.rtabmap]: rtabmap: frame_id      = base_link
[rtabmap-2] [INFO] [1636060867.634760175] [rtabmap.rtabmap]: rtabmap: map_frame_id  = map
[rtabmap-2] [INFO] [1636060867.634766410] [rtabmap.rtabmap]: rtabmap: use_action_for_goal  = false
[rtabmap-2] [INFO] [1636060867.634771417] [rtabmap.rtabmap]: rtabmap: tf_delay      = 0.050000
[rtabmap-2] [INFO] [1636060867.634781963] [rtabmap.rtabmap]: rtabmap: tf_tolerance  = 0.100000
[rtabmap-2] [INFO] [1636060867.634791317] [rtabmap.rtabmap]: rtabmap: odom_sensor_sync   = false
[rtabmap-2] [INFO] [1636060867.634798072] [rtabmap.rtabmap]: rtabmap: stereo_to_depth = false
[rtabmap-2] [INFO] [1636060867.634802738] [rtabmap.rtabmap]: rtabmap: gen_scan  = false
[rtabmap-2] [INFO] [1636060867.634806984] [rtabmap.rtabmap]: rtabmap: gen_depth  = false
[stereo_odometry-1] [INFO] [1636060867.635838579] [rtabmap.stereo_odometry]: StereoOdometry: approx_sync = true
[stereo_odometry-1] [INFO] [1636060867.635862243] [rtabmap.stereo_odometry]: StereoOdometry: queue_size  = 10
[stereo_odometry-1] [INFO] [1636060867.635869708] [rtabmap.stereo_odometry]: StereoOdometry: subscribe_rgbd = false
[stereo_odometry-1] [INFO] [1636060867.635874720] [rtabmap.stereo_odometry]: StereoOdometry: keep_color     = false
[stereo_odometry-1] [INFO] [1636060867.651695816] [rtabmap.stereo_odometry]: 
[stereo_odometry-1] stereo_odometry subscribed to (approx sync):
[stereo_odometry-1]    /left/image,
[stereo_odometry-1]    /right/image,
[stereo_odometry-1]    /left/camera_info,
[stereo_odometry-1]    /right/camera_info
[... cut ...]
[rtabmapviz-3] [INFO] [1636060868.956715859] [rtabmap.rtabmapviz]: Setup stereo callback
[rtabmapviz-3] [INFO] [1636060868.968282513] [rtabmap.rtabmapviz]: 
[rtabmapviz-3] rtabmapviz subscribed to (approx sync):
[rtabmapviz-3]    /rtabmap/odom \
[rtabmapviz-3]    /left/image \
[rtabmapviz-3]    /right/image \
[rtabmapviz-3]    /left/camera_info \
[rtabmapviz-3]    /right/camera_info \
[rtabmapviz-3]    /rtabmap/odom_info
[rtabmapviz-3] [INFO] [1636060868.968425882] [rtabmap.rtabmapviz]: rtabmapviz started.
```

Nykri, 2021-11-17:
Same problem here. I tried :
`ros2 run rtabmap_ros rtabmapviz --ros-args -p wait_for_transform:=0.5`
and the result is :
```
[INFO] [1637161167.833227235] [rtabmapviz]: rtabmapviz: Using configuration from "/home/mathieu/.ros/rtabmapGUI.ini"
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[WARN] [1637161168.800946868] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[INFO] [1637161168.804760712] [rtabmapviz]: Reading parameters from the ROS server...
[ERROR] [1637161173.808801371] [rtabmapviz]: Can't call rtabmap parameters service, is the node running?
[INFO] [1637161173.808964672] [rtabmapviz]: Parameters read = 0
terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
  what():  parameter 'wait_for_transform' has invalid type: expected [double] got [string]
```
I can see Rtabmapviz trying to start but then it shutdowns. Maybe the problem is related to the ROS server but I can't figure where this is.

matlabbe, 2021-11-21:
Hi, 
Which version of rtabmap_ros are you using? Maybe it is because of your computer locale, which may interpret `0.5` as string instead of double. For example on french systems, you may have to use `0,5` for double instead. On my system, I can get your error only if I change `0.5` by `0,5`, which will be interpreted as string:
```
ros2 run rtabmap_ros rtabmapviz --ros-args -p wait_for_transform:=0,5
[INFO] [1637535070.441746284] [rtabmapviz]: rtabmapviz: Using configuration from "/home/mathieu/.ros/rtabmapGUI.ini"
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[WARN] [1637535071.416773619] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[INFO] [1637535071.417978505] [rtabmapviz]: Reading parameters from the ROS server...
[ERROR] [1637535076.419542090] [rtabmapviz]: Can't call rtabmap parameters service, is the node running?
[INFO] [1637535076.419657646] [rtabmapviz]: Parameters read = 0
terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
  what():  parameter 'wait_for_transform' has invalid type: expected [double] got [string]
```

With `0.5`:
```
ros2 run rtabmap_ros rtabmapviz --ros-args -p wait_for_transform:=0.5
[INFO] [1637535227.912371262] [rtabmapviz]: rtabmapviz: Using configuration from "/home/mathieu/.ros/rtabmapGUI.ini"
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[WARN] [1637535228.906308789] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[INFO] [1637535228.907494746] [rtabmapviz]: Reading parameters from the ROS server...
[ERROR] [1637535233.908970275] [rtabmapviz]: Can't call rtabmap parameters service, is the node running?
[INFO] [1637535233.909090483] [rtabmapviz]: Parameters read = 0
[INFO] [1637535234.219667863] [rtabmapviz]: rtabmapviz: subscribe_depth = false
[INFO] [1637535234.219712361] [rtabmapviz]: rtabmapviz: subscribe_rgb = false
[INFO] [1637535234.219720189] [rtabmapviz]: rtabmapviz: subscribe_stereo = false
[INFO] [1637535234.219725754] [rtabmapviz]: rtabmapviz: subscribe_rgbd = false (rgbd_cameras=1)
[INFO] [1637535234.219747564] [rtabmapviz]: rtabmapviz: subscribe_odom_info = false
[INFO] [1637535234.219773476] [rtabmapviz]: rtabmapviz: subscribe_user_data = false
[INFO] [1637535234.219778037] [rtabmapviz]: rtabmapviz: subscribe_scan = false
[INFO] [1637535234.219784011] [rtabmapviz]: rtabmapviz: subscribe_scan_cloud = false
[INFO] [1637535234.219789467] [rtabmapviz]: rtabmapviz: subscribe_scan_descriptor = false
[INFO] [1637535234.219796905] [rtabmapviz]: rtabmapviz: queue_size      = 10
[INFO] [1637535234.219823049] [rtabmapviz]: rtabmapviz: qos_image       = 0
[INFO] [1637535234.219844732] [rtabmapviz]: rtabmapviz: qos_camera_info = 0
[INFO] [1637535234.219850738] [rtabmapviz]: rtabmapviz: qos_scan        = 0
[INFO] [1637535234.219874758] [rtabmapviz]: rtabmapviz: qos_odom        = 0
[INFO] [1637535234.219880233] [rtabmapviz]: rtabmapviz: qos_user_data   = 0
[INFO] [1637535234.219886435] [rtabmapviz]: rtabmapviz: approx_sync     = true
[INFO] [1637535234.219893731] [rtabmapviz]: Setup scan callback
[INFO] [1637535234.221160098] [rtabmapviz]: 
rtabmapviz subscribed to:
   /odom
[INFO] [1637535234.221299891] [rtabmapviz]: rtabmapviz started.
```

ericdepotter, 2021-11-24:
That was indeed the problem. With ```wait_for_transform:=0,5``` it is not crashing anymore. Thank you so much for figuring this out.

closed 2021-11-24 by ericdepotter

ericdepotter, 2021-11-27:
To come back to my previous comment. Using ```wait_for_transform:=0,5```, the ```rtabmapviz``` component did not crash, but the ```rtabmap``` component did.

However knowing what the problem is, I was able to quickly found a solution that works for both components, namely specifying the locale before the launch command like so:
```
LC_NUMERIC="en_US.UTF-8"; ros2 launch rtabmap_ros rtabmap.launch.py \
    args:="--delete_db_on_start" \
    stereo:=true \
    left_image_topic:=/left/image \
    right_image_topic:=/right/image \
    left_camera_info_topic:=/left/camera_info \
    right_camera_info_topic:=/right/camera_info \
    frame_id:=oak-d_frame \
    approx_sync:=true \
    wait_for_transform:=0.5 \
    visual_odometry:=true
```