introlab/rtabmap_ros #672: Ground detection off

kiprock, 2021-11-02:
Hello, I am testing rtab-map with isaac sim using a single rgb+depth camera. I am having issues where the ground plane is not horizontal, even though the camera is pointing forward. I'm having this issue if I use robot odom or visual odom. Any suggestions as to how to debug? The camera is very low to the ground.

[image]

matlabbe, 2021-11-02:
If the whole point cloud is tilted, in general it is because the static TF of the camera is not correctly set (not matching the reality). As this is a simulator, I would assume that TFs are perfect if they are taken from same model. It could be wrong camera calibration intrinsics or wrong depth values. Do you have a database to share?

kiprock, 2021-11-02:
I would have thought the same. I checked the camera TF several times, and even played with the forward rotation.. Perhaps the intrinsics provided by Issac Sim are wrong?

http://kiphaynes.com/LIL/rtabmap11-2.db

kiprock, 2021-11-03:
The camera intrinsics published by isaacsim seem to be causing the issue. When I generated a calibration file in the simulation, the ground issues becomes resolved. Closing.

closed 2021-11-03 by kiprock