introlab/rtabmap_ros #673: symbol lookup error;undefined symbo

hocken-li, 2021-11-03:
When I perform:   "roslaunch rtabmap_ros demo_turtlebot_mapping.launch simulation:=true"

there is error:

/opt/ros/indigo/lib/rtabmap_ros/rtabmap: symbol lookup error: /opt/ros/indigo/lib/x86_64-linux-gnu/librtabmap_core.so.0.17: undefined symbol: _ZN3pcl6search6KdTreeINS_11PointXYZRGBEE16setSortedResultsEb

I install pcl1.7 by 

sudo apt-get install libpcl-all 

I run these in ubuntu14.04, what can I do to solve the problem?

matlabbe, 2021-11-04:
I don't have a machine to test on Indigo anymore (this ubuntu and ros distros are EOL), the only thing I can test is with the docker image ros:indigo-perception:
```
$ docker run -it --rm ros:indigo-perception
# apt update && apt install ros-indigo-rtabmap-ros -y
# source /opt/ros/indigo/setup.bash
# roslaunch rtabmap_ros rtabmap.launch rtabmapviz:=false

... logging to /root/.ros/log/dd5d5e02-3dbb-11ec-bc69-0242ac110002/roslaunch-dfc252353c3b-17813.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dfc252353c3b:41403/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /rtabmap/rgbd_odometry/approx_sync: True
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/frame_id: camera_link
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rgbd_odometry/subscribe_rgbd: False
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 1.0
 * /rtabmap/rtabmap/odom_tf_linear_variance: 1.0
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/scan_normal_k: 0
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_stereo: False
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2

NODES
  /rtabmap/
    rgbd_odometry (rtabmap_ros/rgbd_odometry)
    rtabmap (rtabmap_ros/rtabmap)

auto-starting new master
process[master]: started with pid [17824]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to dd5d5e02-3dbb-11ec-bc69-0242ac110002
process[rosout-1]: started with pid [17837]
started core service [/rosout]
process[rtabmap/rgbd_odometry-2]: started with pid [17854]
libdc1394 error: Failed to initialize libdc1394
[... 35 lines cut ...]
[ INFO] [1636063764.884462920]: RGBDOdometry: rgbd_cameras   = 1
[ INFO] [1636063764.902837724]: 
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info
[ INFO] [1636063765.152359107]: Setting RTAB-Map parameter "Mem/IncrementalMemory"="true"
[ INFO] [1636063765.154283181]: Setting RTAB-Map parameter "Mem/InitWMWithAllNodes"="false"
[ INFO] [1636063768.625539736]: RTAB-Map detection rate = 1.000000 Hz
[ INFO] [1636063768.625781089]: rtabmap: Using database from "/root/.ros/rtabmap.db" (0 MB).
[ INFO] [1636063768.648121272]: rtabmap: Database version = "0.17.6".
[ INFO] [1636063768.687648564]: /rtabmap/rtabmap: queue_size    = 10
[ INFO] [1636063768.687674676]: /rtabmap/rtabmap: rgbd_cameras = 1
[ INFO] [1636063768.687686792]: /rtabmap/rtabmap: approx_sync   = true
[ INFO] [1636063768.687730509]: Setup depth callback
[ INFO] [1636063768.707561846]: 
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom,
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info,
   /rtabmap/odom_info
[ INFO] [1636063768.713419279]: rtabmap 0.17.6 started...
```
It seems okay.

hocken-li, 2021-11-08:
I would like to ask if the error is related to the PCL version，and I would like to ask the way you install the PCL.

matlabbe, 2021-11-08:
PCL is installed at the same time you install ros-indigo-rtabmap-ros (no need to install it explicitly), back on indigo, I think it is the version installed by ros package pcl.