introlab/rtabmap_ros #674: Unable to build ros2 branch

felrock, 2021-11-03:
Hello, I tried to build ros2 branch and I got the following error. 
It seems to not find **kGridFromDepth** in Parameters
```
In file included from /usr/local/include/pcl-1.10/pcl/octree/octree_iterator.h:46:0,
                 from /usr/local/include/pcl-1.10/pcl/octree/octree_base.h:44,
                 from /usr/local/include/pcl-1.10/pcl/octree/octree_pointcloud.h:41,
                 from /usr/local/include/pcl-1.10/pcl/octree/octree_search.h:43,
                 from /usr/local/include/pcl-1.10/pcl/search/octree.h:42,
                 from /usr/local/include/pcl-1.10/pcl/search/pcl_search.h:44,
                 from /usr/local/include/pcl-1.10/pcl/filters/radius_outlier_removal.h:43,
                 from /opt/ros/foxy/src/rtabmap_ros/src/nodelets/point_cloud_assembler.cpp:35:
/usr/local/include/pcl-1.10/pcl/octree/octree_key.h:151:5: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
     };
     ^
/opt/ros/foxy/src/rtabmap_ros/src/CoreWrapper.cpp: In constructor ‘rtabmap_ros::CoreWrapper::CoreWrapper(const rclcpp::NodeOptions&)’:
/opt/ros/foxy/src/rtabmap_ros/src/CoreWrapper.cpp:406:35: error: ‘defaultGridFromDepth’ is not a member of ‘rtabmap::Parameters’
  bool gridFromDepth = Parameters::defaultGridFromDepth();
                                   ^~~~~~~~~~~~~~~~~~~~
/opt/ros/foxy/src/rtabmap_ros/src/CoreWrapper.cpp:407:112: error: ‘kGridFromDepth’ is not a member of ‘rtabmap::Parameters’
  if((this->isSubscribedToScan2d() || this->isSubscribedToScan3d() || genScan_) && parameters_.find(Parameters::kGridFromDepth()) == parameters_.end())
                                                                                                                ^~~~~~~~~~~~~~
In file included from /opt/ros/foxy/install/include/rclcpp/client.hpp:40:0,
                 from /opt/ros/foxy/install/include/rclcpp/callback_group.hpp:23,
                 from /opt/ros/foxy/install/include/rclcpp/any_executable.hpp:20,
                 from /opt/ros/foxy/install/include/rclcpp/memory_strategy.hpp:24,
                 from /opt/ros/foxy/install/include/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/foxy/install/include/rclcpp/executor_options.hpp:20,
                 from /opt/ros/foxy/install/include/rclcpp/executor.hpp:33,
                 from /opt/ros/foxy/install/include/rclcpp/executors/multi_threaded_executor.hpp:26,
                 from /opt/ros/foxy/install/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/install/include/rclcpp/rclcpp.hpp:146,
                 from /opt/ros/foxy/src/rtabmap_ros/include/rtabmap_ros/CoreWrapper.h:32,
                 from /opt/ros/foxy/src/rtabmap_ros/src/CoreWrapper.cpp:28:
/opt/ros/foxy/src/rtabmap_ros/src/CoreWrapper.cpp:414:17: error: ‘kGridFromDepth’ is not a member of ‘rtabmap::Parameters’
     Parameters::kGridFromDepth().c_str(),
                 ^
/opt/ros/foxy/src/rtabmap_ros/src/CoreWrapper.cpp:415:17: error: ‘kGridFromDepth’ is not a member of ‘rtabmap::Parameters’
     Parameters::kGridFromDepth().c_str(),
                 ^
/opt/ros/foxy/src/rtabmap_ros/src/CoreWrapper.cpp:416:17: error: ‘kGridFromDepth’ is not a member of ‘rtabmap::Parameters’
     Parameters::kGridFromDepth().c_str());
                 ^
/opt/ros/foxy/src/rtabmap_ros/src/CoreWrapper.cpp:417:49: error: ‘kGridFromDepth’ is not a member of ‘rtabmap::Parameters’
   parameters_.insert(ParametersPair(Parameters::kGridFromDepth(), "false"));
                                                 ^~~~~~~~~~~~~~
/opt/ros/foxy/src/rtabmap_ros/src/CoreWrapper.cpp:419:45: error: ‘kGridFromDepth’ is not a member of ‘rtabmap::Parameters’
  Parameters::parse(parameters_, Parameters::kGridFromDepth(), gridFromDepth);
                                             ^~~~~~~~~~~~~~
In file included from /opt/ros/foxy/install/include/rclcpp/client.hpp:40:0,
                 from /opt/ros/foxy/install/include/rclcpp/callback_group.hpp:23,
                 from /opt/ros/foxy/install/include/rclcpp/any_executable.hpp:20,
                 from /opt/ros/foxy/install/include/rclcpp/memory_strategy.hpp:24,
                 from /opt/ros/foxy/install/include/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/foxy/install/include/rclcpp/executor_options.hpp:20,
                 from /opt/ros/foxy/install/include/rclcpp/executor.hpp:33,
                 from /opt/ros/foxy/install/include/rclcpp/executors/multi_threaded_executor.hpp:26,
                 from /opt/ros/foxy/install/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/install/include/rclcpp/rclcpp.hpp:146,
                 from /opt/ros/foxy/src/rtabmap_ros/include/rtabmap_ros/CoreWrapper.h:32,
                 from /opt/ros/foxy/src/rtabmap_ros/src/CoreWrapper.cpp:28:
/opt/ros/foxy/src/rtabmap_ros/src/CoreWrapper.cpp:425:17: error: ‘kGridFromDepth’ is not a member of ‘rtabmap::Parameters’
     Parameters::kGridFromDepth().c_str());
                 ^
make[2]: *** [CMakeFiles/rtabmap_sync.dir/src/CoreWrapper.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/rtabmap_sync.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rtabmap_ros [10min 28s, exited with code 2]

```

matlabbe, 2021-11-03:
I forgot to update ros2 branch with upstream rtabmap changes. This should be fixed by the commit above.

closed 2021-11-03 by matlabbe

felrock, 2021-11-08:
Great thanks, when do you plan on the next release for? would be awesome to use it

matlabbe, 2021-11-14:
Current version 0.20.15 is on rosdistro for foxy and galactic. With the new rtabmap_ros binary release on ros2, it will be easier to clone the right compatible versions (following foxy-devel or galactic-devel on rtabmap and rtabmap_ros).

commit ab1288cc (2021-11-03): Merge master->ros2 (fixed #674)