introlab/rtabmap_ros #675: Libpointmatcher error after running at some time

robertsenps, 2021-11-04:
Hi @matlabbe ,

I'm the one from this post who is working on libpointmatcher with 3D lidar. 

First, I would like to say thank you for your reply, so my slam is now faster. But I got another issue while running rtabmap with libpointmatcher. After some time, my odometry estimation which is libpointmatcher dies and produces this error, and libpointmatcher doesn't produce odom anymore. 

[image]

Can you help me to solve this problem?

If you need my parameters, those are not that different from this: https://github.com/introlab/rtabmap_ros/issues/654 

Thank you.

matlabbe, 2021-11-14:
It is currently lost, those errors are saying that the current scan can be registered with latest valid one. I would check the first warning/error message of the log to see exactly what happened to make icp fail. Maybe a too fast rotation? a too large update time? sometimes is may be just that Icp/MaxTranslation is too low (it will be shown in the log as error). From #654 , there was issue about computation time, if the odometry cannot be update for a couple of seconds, ICP will likely fail. 

If anything above doesn't work, setting `Odom/ResetCountdown=1` can automatically reset odometry when lost, but I would rather fix the original error before using this. If you have an external odometry (e.g., wheel odometry), adding `guess_frame_id` to `icp_odometry` node can also help ICP to recover in some cases.

robertsenps, 2021-11-15:
Hi. Thanks for your suggestions. I'll try your solutions. Regarding the first warning node, this is the first error of the libpointmatcher. It said that "limit out of bounds". I'm curious what causes this problem.

[image]

matlabbe, 2021-11-15:
> Maybe a too fast rotation?

It seems what I thought. ICP estimated a 0.88 rad rotation, which is 50 deg. If it is correct, your robot is indeed rotating too fast for the odometry update rate. `Icp/MaxRotation` could be increase to accept such rotation, but I would prefer slow down the robot or better tune odometry to be faster. Oh I see, your odometry FPS is too low, even got almost 1 second update the previous iteration! icp_odometry should update under 100 ms to be reliable, but you have like 300 ms on average with a peak at 979 ms. What is the computer you are using? Based on parameters in this [post](https://github.com/introlab/rtabmap_ros/issues/654), I would increase voxel size of odometry to `0.2`, decrease the number of ICP iterations and reduce maximum size of the local map:

```
Icp/VoxelSize: 0.2
Icp/Iterations: 10
OdomF2M/ScanSubtractRadius: 0.2
OdomF2M/ScanMaxSize: 10000
```

robertsenps, 2021-11-24:
Thank you @matlabbe. I'll try your suggestion. catch with you later with the updates :)

robertsenps, 2022-01-21:
Hi @matlabbe ,

I hope you are well.

First, I would like to say thank you for your support so far. In this comment, I would like to follow up on this issue regarding libpointmatcher error. Last time, your solution works perfectly. But after some time, I encountered this issue again. Basically, I made rtabmap run faster by using your suggestion.However, I have deployed rtabmap to another bot, and I found that after 10-15 second, the libpointmatcher lags and it affected the point_cloud_assembler and rtabmap/rtabmap. 

Here's the video :
[rtabmap_error.zip](https://github.com/introlab/rtabmap_ros/files/7912349/rtabmap_error.zip)

I also attached the db file and some configuration files:
[db_and_config.zip](https://github.com/introlab/rtabmap_ros/files/7912411/db_and_config.zip)

From your perspective, what is the problem with libpointmatcher this time, I'm not sure it is regarding the computation power because It happened at the very beginning and I could recreate everytime.

I need your help and your suggestion :). I hope I can hear from you soon. 

Thanks

kosmonauta144, 2023-12-15:
I also got this error when using Ros humble and rplidar for odometry in corridor and I wonder how could I improve it. After some time I got the same error and increasing accepted transform does not resolve the problem, because even when robot is not moving, it seams that uniformity of the environment is the cause of translation error :/

matlabbe, 2023-12-16:
@kosmonauta144 Which error did you get from the ones stated in that thread?

kosmonauta144, 2023-12-19:
@matlabbe , I got error with 

```
libpointmatcher has failed. limit out of bounds
```

but I discovered that when I removed `config.ini` file which I've been using and put parameters in launch file as ros2 params, now everything is working  . I was playing with parameters in automatically generated `ini` file and I wonder if it is possible, that some arguments influence the value of other and when I change for example number of iterations for ICP or Odometry Strategy or Reg/Strategy some other parameters

matlabbe, 2023-12-23:
How did you include the `config.ini` file? I try to understand the problem, is using the config file giving different results than setting the parameters are rosparam directly? normally, the result should be the same. You can check in the log which parameters were set from the config file or the rosparam.

To change dynamically the parameters while the node is running, I'll suggest to use rtabmap_viz.