introlab/rtabmap_ros #676: Errors: [Not Supporting ROI] when superpoint is used as the feature detector

szx0112, 2021-11-10:
Hi @matlabbe,

I tried to test the performance of Superpoint on my project. I followed your [your comment](https://github.com/introlab/rtabmap/issues/531#issuecomment-615221478) to build rtabmap with libtorch 1.10 and cuda 11.3.

I can launch the rtabmap with superpoint enabled without errors. However, once I start streaming (e..g, RGBD image sequence from a D435i camera with resolution as 848x480 and 60fps), I got the N**ot Supporting ROI** errors as below:

```
[FATAL] (2021-11-10 17:26:29.041) Features2d.cpp:2092::generateKeypointsImpl() Condition (roi.x==0 && roi.y ==0) not met! [Not supporting ROI]
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2021-11-10 17:26:29.041) Features2d.cpp:2092::generateKeypointsImpl() Condition (roi.x==0 && roi.y ==0) not met! [Not supporting ROI]
```

The roi is used as the input in this function `			sub_keypoints = this->generateKeypointsImpl(image, roi, mask)
` which is seemly shared by all feature types.

However, when SIFT, ORB or other hand craft feature is chose, I can run the code smoothly without any error.

Could you give me some advice how to handle this issue? Thanks.

matlabbe, 2021-11-14:
The Region of Interest (ROI) seems to be set. What are the parameters you are using? Make sure those parameters are set to defaults:
```
Param: Kp/RoiRatios = "0.0 0.0 0.0 0.0"                    [Region of interest ratios [left, right, top, bottom].]
Param: Vis/RoiRatios = "0.0 0.0 0.0 0.0"                   [Region of interest ratios [left, right, top, bottom].]
```

szx0112, 2021-11-14:
@matlabbe Thanks for the reply. I manually set Roi parameters as suggested. I still receive the same error message. Below is the full return from the command line:

`
started roslaunch server http://szx19:43199/

SUMMARY
========

PARAMETERS
 * /rgbd_sync/approx_sync: False
 * /rgbd_sync/depth_scale: 1.0
 * /rgbd_sync/queue_size: 10
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /rtabmap/rgbd_odometry/Kp/DetectorStrategy: 11
 * /rtabmap/rgbd_odometry/Odom/Strategy: 0
 * /rtabmap/rgbd_odometry/SuperPoint/Cuda: true
 * /rtabmap/rgbd_odometry/SuperPoint/ModelPath: /home/szx/catkin_...
 * /rtabmap/rgbd_odometry/Vis/BundleAdjustment: 1
 * /rtabmap/rgbd_odometry/Vis/CorGuessWinSize: 20
 * /rtabmap/rgbd_odometry/Vis/CorNNType: 1
 * /rtabmap/rgbd_odometry/Vis/CorType: 0
 * /rtabmap/rgbd_odometry/Vis/DepthMask: True
 * /rtabmap/rgbd_odometry/Vis/EstimationType: 1
 * /rtabmap/rgbd_odometry/Vis/FeatureType: 11
 * /rtabmap/rgbd_odometry/Vis/GridCols: 20
 * /rtabmap/rgbd_odometry/Vis/GridRows: 20
 * /rtabmap/rgbd_odometry/Vis/InlierDistance: 0.01
 * /rtabmap/rgbd_odometry/Vis/MaxDepth: 0.7
 * /rtabmap/rgbd_odometry/Vis/MaxFeatures: 1000
 * /rtabmap/rgbd_odometry/Vis/MinDepth: 0.15
 * /rtabmap/rgbd_odometry/Vis/MinInliers: 20
 * /rtabmap/rgbd_odometry/Vis/RefineIterations: 20
 * /rtabmap/rgbd_odometry/approx_sync: False
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/frame_id: base_link
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rgbd_odometry/publish_tf: True
 * /rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rgbd_odometry/subscribe_rgbd: True
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 2.0
 * /rtabmap/rgbd_odometry/wait_imu_to_init: False
 * /rtabmap/rtabmap/Grid/Sensor: True
 * /rtabmap/rtabmap/KP/MaxFeatures: 500
 * /rtabmap/rtabmap/Kp/DetectorStrategy: 11
 * /rtabmap/rtabmap/Kp/MaxDepth: 0.8
 * /rtabmap/rtabmap/Kp/NNStrategy: 1
 * /rtabmap/rtabmap/Kp/RoiRatios: 0.0 0.0 0.0 0.0
 * /rtabmap/rtabmap/Marker/Dictionary: 16
 * /rtabmap/rtabmap/Marker/Length: 0.055
 * /rtabmap/rtabmap/Odom/Strategy: 0
 * /rtabmap/rtabmap/Optimizer/Robust: false
 * /rtabmap/rtabmap/Optimizer/Strategy: 2
 * /rtabmap/rtabmap/RGBD/AngularUpdate: 0.0
 * /rtabmap/rtabmap/RGBD/LinearUpdate: 0.01
 * /rtabmap/rtabmap/RGBD/LocalRadius: 5
 * /rtabmap/rtabmap/RGBD/LoopClosureReextractFeatures: true
 * /rtabmap/rtabmap/RGBD/MarkerDetection: True
 * /rtabmap/rtabmap/RGBD/NeighborLinkRefining: false
 * /rtabmap/rtabmap/RGBD/OptimizeMaxError: 3
 * /rtabmap/rtabmap/RGBD/ProximityAngle: 25
 * /rtabmap/rtabmap/RGBD/ProximityBySpace: true
 * /rtabmap/rtabmap/Reg/Force3DoF: False
 * /rtabmap/rtabmap/Reg/Strategy: 0
 * /rtabmap/rtabmap/Rtabmap/DetectionRate: 0
[... 740 lines cut ...]
[ INFO] [1636927796.625362521]: rtabmapviz started.
**[FATAL] (2021-11-14 16:15:03.416) Features2d.cpp:2092::generateKeypointsImpl() Condition (roi.x==0 && roi.y ==0) not met! [Not supporting ROI]
terminate called after throwing an instance of 'UException'**
  **what():  [FATAL] (2021-11-14 16:15:03.416) Features2d.cpp:2092::generateKeypointsImpl() Condition (roi.x==0 && roi.y ==0) not met! [Not supporting ROI]**
[rtabmap/rgbd_odometry-2] process has died [pid 25811, exit code -6, cmd /home/szx/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry rgb/image:=/camera5/color/image_raw depth/image:=/camera5/aligned_depth_to_color/image_raw rgb/camera_info:=/camera5/color/camera_info rgbd_image:=/rgbd_image odom:=/front_odom odom_info:=/front_odom_info __name:=rgbd_odometry __log:=/home/szx/.ros/log/7368028a-4578-11ec-9aaa-989096b12255/rtabmap-rgbd_odometry-2.log].
log file: /home/szx/.ros/log/7368028a-4578-11ec-9aaa-989096b12255/rtabmap-rgbd_odometry-2*.log
[ WARN] [1636928597.774591240, 1636862410.129271139]: /rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmapviz subscribed to (approx sync):
   /front_odom \
   /rgbd_image \
   /front_odom_info
[ WARN] [1636928597.774591300, 1636862410.129271139]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap subscribed to (exact sync):
   /rgbd_image \
   /front_odom_info
^C[rtabmap/rtabmapviz-4] killing on exit
[rtabmap/rtabmap-3] killing on exit
[rgbd_sync-1] killing on exit
[ INFO] [1636928689.904613742, 1636862411.050575863]: rtabmapviz: ctrl-c catched! Exiting Qt app...
rtabmap: Saving database/long-term memory... (located at /home/szx/.ros/rtabmap.db)
rtabmap: Saving database/long-term memory...done! (located at /home/szx/.ros/rtabmap.db, 0 MB)
shutting down processing monitor...
... shutting down processing monitor complete
done
`

matlabbe, 2021-11-15:
```
[ INFO] [1636927786.399028386]: Setting odometry parameter "Vis/GridCols"="20"
[ INFO] [1636927786.400009847]: Setting odometry parameter "Vis/GridRows"="20"
```
Set those parameters to 0. I will update the error message to be easier to detect this problem.

szx0112, 2021-11-15:
It works now. Thanks @matlabbe .

I see the Superpoint detected features are color coded as green/blue/red/yellow. Could you point out what are the meanings of these colors?

matlabbe, 2021-11-15:
You can mouse over the label of the loop closure detection to see the color code:
[image]

szx0112, 2021-11-15:
Got it. Thanks for the info.

szx0112, 2021-11-15:
@matlabbe This might also related to SuperPoint setup. I got my rtabmap node crashed at the middle of the mapping as below. I don't see this error when ORB or other feature is used for loop closure. I use GTFF for VO.
```
[ INFO] [1637009483.962312404, 1636862612.929951076]: rtabmap (2230): Rate=0.00s, Limit=0.000s, Conversion=0.0046s, RTAB-Map=0.0607s, Maps update=0.0032s pub=0.0000s (local map=826, WM=826)
[ INFO] [1637009485.772973694, 1636862612.944458826]: Odom: quality=155, std dev=0.003095m|0.066028rad, update time=0.071611s
[ INFO] [1637009485.974480762, 1636862612.946072746]: rtabmap (2231): Rate=0.00s, Limit=0.000s, Conversion=0.0042s, RTAB-Map=0.1929s, Maps update=0.0027s pub=0.0000s (local map=827, WM=827)
[ INFO] [1637009488.690867839, 1636862612.967796757]: Odom: quality=146, std dev=0.003792m|0.074679rad, update time=0.070582s
**terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at**
[rtabmap/rtabmap-3] process has died [pid 7411, exit code -6, cmd /home/szx/catkin_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start rgbd_image:=/rgbd_image odom_info:=/front_odom_info odom:=/front_odom __name:=rtabmap __log:=/home/szx/.ros/log/7368028a-4578-11ec-9aaa-989096b12255/rtabmap-rtabmap-3.log].
log file: /home/szx/.ros/log/7368028a-4578-11ec-9aaa-989096b12255/rtabmap-rtabmap-3*.log
[ INFO] [1637009493.025963220, 1636862613.002436800]: Odom: quality=142, std dev=0.002640m|0.071352rad, update time=0.067335s
[ INFO] [1637009493.099727246, 1636862613.003084130]: Odom: quality=129, std dev=0.002886m|0.073978rad, update time=0.072117s
[ INFO] [1637009494.247543852, 1636862613.012218656]: Odom: quality=166, std dev=0.002820m|0.070952rad, update time=0.061395s
[ INFO] [1637009497.522379295, 1636862613.038414389]: Odom: quality=231, std dev=0.001967m|0.062797rad, update time=0.060832s
```

matlabbe, 2021-11-15:
can you call `rosservice call /rtabmap/rtabmap/log_debug` to show more debug log?

szx0112, 2021-11-16:
Please see a snapshot of the command line returns in debug mode (start from the last Odom message before rtabmap node crash). 

`[ INFO] [1637022259.598089279, 1636862607.811334906]: Odom: quality=112, std dev=0.004583m|0.082900rad, update time=0.075661s
[DEBUG] (2021-11-15 18:24:19.606) Features2d.cpp:751::generateKeypoints() Keypoints extraction time = 0.016018 s, keypoints extracted = 1000 (grid=1x1, mask empty=0)
[DEBUG] (2021-11-15 18:24:19.606) RegistrationVis.cpp:599::computeTransformationImpl() 
[DEBUG] (2021-11-15 18:24:19.608) util2d.cpp:1162::computeRoi() roi ratios = 0.000000, 0.000000, 0.000000, 0.000000
[DEBUG] (2021-11-15 18:24:19.608) util2d.cpp:1163::computeRoi() roi = 0, 0, 640, 480
[DEBUG] (2021-11-15 18:24:19.608) util2d.cpp:1190::computeRoi() roi = 0, 0, 640, 480
[DEBUG] (2021-11-15 18:24:19.618) Features2d.cpp:751::generateKeypoints() Keypoints extraction time = 0.009736 s, keypoints extracted = 1000 (grid=1x1, mask empty=0)
[DEBUG] (2021-11-15 18:24:19.618) RegistrationVis.cpp:649::computeTransformationImpl() kptsFrom=1000 kptsFromSource=0
[DEBUG] (2021-11-15 18:24:19.618) RegistrationVis.cpp:650::computeTransformationImpl() kptsTo=1000 kptsToSource=0
[DEBUG] (2021-11-15 18:24:19.618) RegistrationVis.cpp:672::computeTransformationImpl() cleared orignalWordsFromIds
[DEBUG] (2021-11-15 18:24:19.622) Features2d.cpp:783::generateDescriptors() Descriptors extracted = 846, remaining kpts=846
[DEBUG] (2021-11-15 18:24:19.625) Features2d.cpp:783::generateDescriptors() Descriptors extracted = 797, remaining kpts=797
[DEBUG] (2021-11-15 18:24:19.625) RegistrationVis.cpp:733::computeTransformationImpl() generated kptsFrom3D=846
[DEBUG] (2021-11-15 18:24:19.625) Features2d.cpp:164::filterKeypointsByDepth() 
[DEBUG] (2021-11-15 18:24:19.625) Features2d.cpp:193::filterKeypointsByDepth() Removed 0 invalid 3D points
[DEBUG] (2021-11-15 18:24:19.626) Features2d.cpp:164::filterKeypointsByDepth() 
[DEBUG] (2021-11-15 18:24:19.626) Features2d.cpp:193::filterKeypointsByDepth() Removed 0 invalid 3D points
[DEBUG] (2021-11-15 18:24:19.626) RegistrationVis.cpp:783::computeTransformationImpl() descriptorsFrom=846
[DEBUG] (2021-11-15 18:24:19.626) RegistrationVis.cpp:784::computeTransformationImpl() descriptorsTo=797
[DEBUG] (2021-11-15 18:24:19.626) RegistrationVis.cpp:785::computeTransformationImpl() orignalWordsFromIds=0
[DEBUG] (2021-11-15 18:24:19.626) RegistrationVis.cpp:1165::computeTransformationImpl() 
[DEBUG] (2021-11-15 18:24:19.626) RegistrationVis.cpp:1265::computeTransformationImpl() VWDictionary knn matching
[DEBUG] (2021-11-15 18:24:19.626) VWDictionary.cpp:470::update() incremental=1
[DEBUG] (2021-11-15 18:24:19.626) FlannIndex.cpp:52::release() 
[DEBUG] (2021-11-15 18:24:19.626) FlannIndex.cpp:80::release() 
[DEBUG] (2021-11-15 18:24:19.626) VWDictionary.cpp:681::update() Dictionary updated! (size=0 added=0 removed=0)
[DEBUG] (2021-11-15 18:24:19.626) VWDictionary.cpp:689::update() 
[DEBUG] (2021-11-15 18:24:19.626) VWDictionary.cpp:763::addNewWords() id=1 descriptors=846
[DEBUG] (2021-11-15 18:24:19.627) VWDictionary.cpp:1059::addNewWords() naive search and add ref/words time = 0.001109 s
[DEBUG] (2021-11-15 18:24:19.627) VWDictionary.cpp:1061::addNewWords() 846 new words added...
[DEBUG] (2021-11-15 18:24:19.627) VWDictionary.cpp:1063::addNewWords() 0 duplicated words added (from current image = 0)...
[DEBUG] (2021-11-15 18:24:19.627) VWDictionary.cpp:1064::addNewWords() total time 0.001288s
[DEBUG] (2021-11-15 18:24:19.627) VWDictionary.cpp:470::update() incremental=1
[DEBUG] (2021-11-15 18:24:19.627) VWDictionary.cpp:491::update() Incremental FLANN: Removing 0 words...
[DEBUG] (2021-11-15 18:24:19.627) VWDictionary.cpp:500::update() Incremental FLANN: Removing 0 words... done!
[DEBUG] (2021-11-15 18:24:19.627) VWDictionary.cpp:504::update() Incremental FLANN: Inserting 846 words...
[DEBUG] (2021-11-15 18:24:19.627) VWDictionary.cpp:531::update() Building FLANN index...
[DEBUG] (2021-11-15 18:24:19.627) FlannIndex.cpp:208::buildKDTreeIndex() 
[DEBUG] (2021-11-15 18:24:19.627) FlannIndex.cpp:52::release() 
[DEBUG] (2021-11-15 18:24:19.627) FlannIndex.cpp:80::release() 
[DEBUG] (2021-11-15 18:24:19.627) FlannIndex.cpp:260::buildKDTreeIndex() 
[DEBUG] (2021-11-15 18:24:19.627) VWDictionary.cpp:549::update() Building FLANN index... done!
[DEBUG] (2021-11-15 18:24:19.627) FlannIndex.cpp:405::addPoints() Rebuilding FLANN index: 1 -> 3
[DEBUG] (2021-11-15 18:24:19.627) FlannIndex.cpp:405::addPoints() Rebuilding FLANN index: 3 -> 7
[DEBUG] (2021-11-15 18:24:19.628) FlannIndex.cpp:405::addPoints() Rebuilding FLANN index: 7 -> 15
[DEBUG] (2021-11-15 18:24:19.628) FlannIndex.cpp:405::addPoints() Rebuilding FLANN index: 15 -> 31
[DEBUG] (2021-11-15 18:24:19.628) FlannIndex.cpp:405::addPoints() Rebuilding FLANN index: 31 -> 63
[DEBUG] (2021-11-15 18:24:19.629) FlannIndex.cpp:405::addPoints() Rebuilding FLANN index: 63 -> 127
[DEBUG] (2021-11-15 18:24:19.630) FlannIndex.cpp:405::addPoints() Rebuilding FLANN index: 127 -> 255
[DEBUG] (2021-11-15 18:24:19.633) FlannIndex.cpp:405::addPoints() Rebuilding FLANN index: 255 -> 511
[DEBUG] (2021-11-15 18:24:19.639) VWDictionary.cpp:564::update() Incremental FLANN: Inserting 846 words... done!
[DEBUG] (2021-11-15 18:24:19.639) VWDictionary.cpp:681::update() Dictionary updated! (size=0 added=846 removed=0)
[DEBUG] (2021-11-15 18:24:19.639) VWDictionary.cpp:689::update() 
[... cut ...]
[ INFO] [1637022265.840372121, 1636862607.873764117]: Odom: quality=124, std dev=0.004562m|0.072132rad, update time=0.078361s
[ INFO] [1637022268.575516049, 1636862607.901054276]: Odom: quality=129, std dev=0.003533m|0.073256rad, update time=0.075268s
[ INFO] [1637022268.941270478, 1636862607.904792966]: Odom: quality=134, std dev=0.003664m|0.075361rad, update time=0.078438s
[ INFO] [1637022273.258745624, 1636862607.947954083]: Odom: quality=146, std dev=0.002415m|0.072512rad, update time=0.081833s

matlabbe, 2021-11-16:
You could try with Optimizer/Strategy=1 (g2o). It seems crashing inside GTSAM for some reasons (after 12 iterations). Which GTSAM version are you using? To debug more, you would have to launch the node inside gdb and show the debug trace (see [Section A of this issue](https://github.com/introlab/rtabmap_ros/issues/28)), we may see where inside GTSAM the map is called with out of range.

szx0112, 2021-11-17:
Thanks for the suggestion. After swith to g2o, every works fine now. For the gtsam, I installed it through PPA. The version is 4.1 and my OS is Ubuntu 18.04. 

Below is the screen shot of the gdb return when gtsam selected as the optimizer.

[image]

matlabbe, 2021-11-17:
Thank you, but I would also take the backtrace of gdb: type `bt` after the crash right you were in the screenshot.

szx0112, 2021-11-17:
No problem. Attached is the backtrace return. Please let me know if you need other info.

[gbt.txt](https://github.com/introlab/rtabmap_ros/files/7556259/gbt.txt)

matlabbe, 2021-11-21:
I am using [GTSAM 4.0.3-1ubuntu1](https://launchpad.net/~borglab/+archive/ubuntu/gtsam-release-4.0):
```
sudo add-apt-repository ppa:borglab/gtsam-release-4.0
```
maybe this is an issue with GTSAM 4.1. Are you using the nightly build from https://launchpad.net/~borglab, which seems to be 4.1?

szx0112, 2021-11-22:
Yes, it seems the problem is related to the gtsam 4.1. I see no more errors after revert back to 4.0. 

However, I found the gtsam shows much worse result than g2o. Please see the results of mapping a loop using gtsam (top) and g2o below (bottom). 
[image]

[image]

Both use superpoint for loop closure. It seems gtsam detect wrong loops then mess the entire model. I have tried to tuning some paramters such as increase RGBD/OptimizeMaxError to 10, but it does not help much.

I am ok to stick with g2o because it works as expected. Just curious why this happen. Here are the a copy of returns printed on the command line when wrong loops are detected/rejected by gtsam:

`[ INFO] [1637611558.816105256, 1636862644.811008352]: rtabmap (3815): Rate=0.00s, Limit=0.000s, Conversion=0.0058s, RTAB-Map=1.3650s, Maps update=0.0085s pub=0.0000s (local map=2365, WM=2365)
[ WARN] (2021-11-22 14:06:00.382) Rtabmap.cpp:3180::process() Rejecting all added loop closures (2, first is 3816 <-> 3804) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 16.195126 (edge 3815->3816, type=0, abs error=0.512135 m, stddev=0.031623). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2021-11-22 14:06:00.382) Rtabmap.cpp:3214::process() Loop closure 3816->3804 rejected!
[ WARN] (2021-11-22 14:06:00.382) Rtabmap.cpp:3214::process() Loop closure 3816->3798 rejected!
[ INFO] [1637611560.399813620, 1636862644.826850411]: rtabmap (3816): Rate=0.00s, Limit=0.000s, Conversion=0.0026s, RTAB-Map=1.5712s, Maps update=0.0078s pub=0.0000s (local map=2366, WM=2366)
[ INFO] [1637611560.923651001, 1636862644.832097016]: Odom: quality=390, std dev=0.001342m|0.047716rad, update time=0.052916s
[ WARN] (2021-11-22 14:06:02.385) Rtabmap.cpp:3180::process() Rejecting all added loop closures (2, first is 3817 <-> 3805) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 9.486082 (edge 3798->3800, type=0, abs error=0.299976 m, stddev=0.031623). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2021-11-22 14:06:02.385) Rtabmap.cpp:3214::process() Loop closure 3817->3805 rejected!
[ WARN] (2021-11-22 14:06:02.385) Rtabmap.cpp:3214::process() Loop closure 3817->3798 rejected!
[ INFO] [1637611562.403141445, 1636862644.846928761]: rtabmap (3817): Rate=0.00s, Limit=0.000s, Conversion=0.0019s, RTAB-Map=1.4694s, Maps update=0.0073s pub=0.0000s (local map=2366, WM=2366)
[ INFO] [1637611566.996049509, 1636862644.892793945]: Odom: quality=374, std dev=0.001327m|0.047716rad, update time=0.067247s
`

matlabbe, 2021-11-23:
If you can share the database, I could take a look why there is so much difference between both.

szx0112, 2021-11-23:
@matlabbe  Please try [this](https://uofnelincoln-my.sharepoint.com/:u:/g/personal/zshang2_unl_edu/EVxgyArYdRxPl-KNzM0b3BQB7FNxegGGThWR2RFkGdKaiQ?e=5ytKHT)

You might need to play it at the reduced speed (-r 0.01) or the VO can easily get lost.

matlabbe, 2021-11-25:
The link expired, sorry I could not download it earlier

szx0112, 2021-11-28:
No problem, use this [link](https://uofnelincoln-my.sharepoint.com/:u:/g/personal/zshang2_unl_edu/EVxgyArYdRxPl-KNzM0b3BQB7FNxegGGThWR2RFkGdKaiQ?e=o8sos3)

szx0112, 2021-12-02:
Hi @matlabbe , did you have a chance to look at the bag? I just noticed I adjusted some parameters in VO and RTABMAP nodes, not quite sure if it can be reproduced with the default settings. Pls let me know, I can send you my launch files if needed.

matlabbe, 2021-12-07:
Hi, I tried the bag with this:
```bash
$ roslaunch rtabmap_ros rtabmap.launch \
   args:="-d \
      --Rtabmap/DetectionRate 2 \
      --GFTT/QualityLevel 0.0001 \
      --GFTT/MinDistance 7" \
   rgb_topic:=/camera5/color/image_raw  \
   camera_info_topic:=/camera5/color/camera_info \
   depth_topic:=/camera5/aligned_depth_to_color/image_raw \
   use_sim_time:=true \
   approx_sync:=false \
   frame_id:=camera5_link

$ rosbag play --clock --pause -r 0.2 loop2_camera5.bag
```
Indeed, we have to reduce speed of the bag for the fast motions in textureless parts of the pipe. There are some drifts, mostly along Z axis:
Top and side views of odometry only:
[image]
[image]

With loop closure:
[image] 
[image]

I couldn't reproduce the GTSAM problem though. I tried with gtsam4.0 and gtsam develop versions from PPA and the optimization is the same:
[image]

I see yellow links on your graph, did you enable RGBD/ProximityByTime? If you can share the database directly, it would be easier to reproduce exactly your problem.

szx0112, 2021-12-07:
@matlabbe Thanks for testing the bag. I will try to rebuild trabmap and see if the problem happens again. I don't have RGBD/ProximityByTime setup, but indeed have RGBD/ProximityBySpace set to true.  Maybe it relates to other parameters selected in my [launch file](url). 

My results are obtain from the exact same bag as I shared. The overall database is just multiple cameras mounted on a rig. I splited the database per camera, tried to register them one by one using same ideal as multi-session map, I start with registering two cameras like this (https://github.com/introlab/rtabmap_ros/issues/645#issue-1001375272), but find it is very difficult to register them together due to the sparse features and insufficient overlapping between the maps.

matlabbe, 2021-12-08:
The launch file link is not working.

szx0112, 2021-12-08:
@matlabbe  sorry I paste the wrong link, please try [this](https://uofnelincoln-my.sharepoint.com/:u:/g/personal/zshang2_unl_edu/ERPGnoQ5iltFkfrSTGfStmMBEtrVu5zefNSJfkgRq3oyeg)

matlabbe, 2021-12-08:
At first view I don't see why there are as may yellow links. If you can share your database directly, it will be easier to look at the problem.

szx0112, 2021-12-11:
@matlabbe please see the [dataset](https://uofnelincoln-my.sharepoint.com/:u:/g/personal/zshang2_unl_edu/EdIUAejaHklCrxC44otDt8MBP4a48N8B2ZIQs7mRalqvCg?e=ggafMN). It contain multiple cams so it is quite large.

matlabbe, 2021-12-14:
It looks like the link is not public. However, I would rather have the database you are generating than the raw bags, because I may ended up like my previous [post](https://github.com/introlab/rtabmap_ros/issues/676#issuecomment-987466084) without reproducing the wrong links you have...

szx0112, 2021-12-14:
That is weird, the link should work. I haven't got a chance to reinstall rtabmap on my desktop yet. Will test the results later.

Here are the separate bags [cam4](https://uofnelincoln-my.sharepoint.com/:u:/g/personal/zshang2_unl_edu/ET5wNFMGBBZMmtnarOmRDCoBpMonY58iIJhOEM9yRf--FQ?e=UOzgNU), [cam1](https://uofnelincoln-my.sharepoint.com/:u:/g/personal/zshang2_unl_edu/Ef2zm1Cpgi9OozyNgGAHuxcBUPlyBuh8PAjXqf2otO40Ug?e=kuff2Q), [cam2](https://uofnelincoln-my.sharepoint.com/:u:/g/personal/zshang2_unl_edu/EU10faqxtIFOpdBZAcncl68BrJioXRTRjB0WUo9z3p3IOQ?e=Fyd3Jg)

Could you give a try for the multi-session mapping when you are available? I had difficulty to fully register them together (in sequence: 5->4->1->2). Thanks.

closed 2021-12-30 by szx0112

fbliman, 2023-08-14:
@szx0112 could you please share your launch files for this project, thanks!!