introlab/rtabmap_ros #677: catkin_make error: field ‘param_k_’ has incomplete type ‘flann::SearchParams’::flann::SearchParams param_k_;

43998213, 2021-11-15:
[ 73%] Built target rtabmap_ros_generate_messages
[ 73%] Building CXX object rtabmap_ros-0.20.14-melodic/CMakeFiles/rtabmap_camera.dir/src/CameraNode.cpp.o
[ 73%] Building CXX object rtabmap_ros-0.20.14-melodic/CMakeFiles/rtabmap_ros.dir/src/OdometryROS.cpp.o
[ 73%] Building CXX object rtabmap_ros-0.20.14-melodic/CMakeFiles/rtabmap_ros.dir/src/MsgConversion.cpp.o
[ 74%] Building CXX object rtabmap_ros-0.20.14-melodic/CMakeFiles/rtabmap_ros.dir/src/MapsManager.cpp.o
In file included from /usr/include/pcl-1.8/pcl/search/kdtree.h:44:0,
                 from /home/zjh/planner_ws01/src/rtabmap_ros-0.20.14-melodic/src/MapsManager.cpp:42:
/usr/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:230:29: error: field ‘param_k_’ has incomplete type ‘flann::SearchParams’
       ::flann::SearchParams param_k_;
                             ^~~~~~~~
In file included from /usr/include/pcl-1.8/pcl/search/kdtree.h:44:0,
                 from /home/zjh/planner_ws01/src/rtabmap_ros-0.20.14-melodic/src/MapsManager.cpp:42:
/usr/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:52:10: note: forward declaration of ‘struct flann::SearchParams’
   struct SearchParams;
          ^~~~~~~~~~~~
In file included from /usr/include/pcl-1.8/pcl/search/kdtree.h:44:0,
                 from /home/zjh/planner_ws01/src/rtabmap_ros-0.20.14-melodic/src/MapsManager.cpp:42:
/usr/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:233:29: error: field ‘param_radius_’ has incomplete type ‘flann::SearchParams’
       ::flann::SearchParams param_radius_;
                             ^~~~~~~~~~~~~
In file included from /usr/include/pcl-1.8/pcl/search/kdtree.h:44:0,
                 from /home/zjh/planner_ws01/src/rtabmap_ros-0.20.14-melodic/src/MapsManager.cpp:42:
/usr/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:52:10: note: forward declaration of ‘struct flann::SearchParams’
   struct SearchParams;
          ^~~~~~~~~~~~
/home/zjh/planner_ws01/src/rtabmap_ros-0.20.14-melodic/src/MapsManager.cpp: In member function ‘void MapsManager::publishMaps(const std::map<int, rtabmap::Transform>&, const ros::Time&, const string&)’:
/home/zjh/planner_ws01/src/rtabmap_ros-0.20.14-melodic/src/MapsManager.cpp:823:36: error: ‘kGridFromDepth’ is not a member of ‘rtabmap::Parameters’
    if(parameters_.find(Parameters::kGridFromDepth()) != parameters_.end() &&
                                    ^~~~~~~~~~~~~~
/home/zjh/planner_ws01/src/rtabmap_ros-0.20.14-melodic/src/MapsManager.cpp:824:42: error: ‘kGridFromDepth’ is not a member of ‘rtabmap::Parameters’
     uStr2Bool(parameters_.at(Parameters::kGridFromDepth())))
                                          ^~~~~~~~~~~~~~
In file included from /opt/ros/melodic/include/ros/assert.h:35:0,
                 from /opt/ros/melodic/include/ros/common.h:36,
                 from /opt/ros/melodic/include/ros/publisher.h:32,
                 from /home/zjh/planner_ws01/src/rtabmap_ros-0.20.14-melodic/include/rtabmap_ros/MapsManager.h:37,
                 from /home/zjh/planner_ws01/src/rtabmap_ros-0.20.14-melodic/src/MapsManager.cpp:27:
/home/zjh/planner_ws01/src/rtabmap_ros-0.20.14-melodic/src/MapsManager.cpp:830:19: error: ‘kGridFromDepth’ is not a member of ‘rtabmap::Parameters’
       Parameters::kGridFromDepth().c_str());
                   ^
/opt/ros/melodic/include/ros/console.h:348:165: note: in definition of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                     ^~~~~~~~~~~
/opt/ros/melodic/include/ros/console.h:387:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION’
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/ros/console.h:572:35: note: in expansion of macro ‘ROS_LOG_COND’
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^~~~~~~~~~~~
/opt/ros/melodic/include/rosconsole/macros_generated.h:162:23: note: in expansion of macro ‘ROS_LOG’
 #define ROS_WARN(...) ROS_LOG(::ros::console::levels::Warn, ROSCONSOLE_DEFAULT_NAME, __VA_ARGS__)
                       ^~~~~~~
/home/zjh/planner_ws01/src/rtabmap_ros-0.20.14-melodic/src/MapsManager.cpp:826:5: note: in expansion of macro ‘ROS_WARN’
     ROS_WARN("/scan_map topic is deprecated! Subscribe to /cloud_map topic "
     ^
/home/zjh/planner_ws01/src/rtabmap_ros-0.20.14-melodic/src/MapsManager.cpp:1334:50: error: ‘kGridFromDepth’ is not a member of ‘rtabmap::Parameters’
       Parameters::kGrid3D().c_str(), Parameters::kGridFromDepth().c_str());
                                                  ^
/opt/ros/melodic/include/ros/console.h:348:165: note: in definition of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                     ^~~~~~~~~~~
/opt/ros/melodic/include/ros/console.h:387:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION’
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/ros/console.h:572:35: note: in expansion of macro ‘ROS_LOG_COND’
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^~~~~~~~~~~~
/opt/ros/melodic/include/rosconsole/macros_generated.h:162:23: note: in expansion of macro ‘ROS_LOG’
 #define ROS_WARN(...) ROS_LOG(::ros::console::levels::Warn, ROSCONSOLE_DEFAULT_NAME, __VA_ARGS__)
                       ^~~~~~~
[... 57 lines cut ...]
                 from /home/zjh/planner_ws01/src/rtabmap_ros-0.20.14-melodic/src/MsgConversion.cpp:46:
/usr/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:52:10: note: forward declaration of ‘struct flann::SearchParams’
   struct SearchParams;
          ^~~~~~~~~~~~
rtabmap_ros-0.20.14-melodic/CMakeFiles/rtabmap_ros.dir/build.make:86: recipe for target 'rtabmap_ros-0.20.14-melodic/CMakeFiles/rtabmap_ros.dir/src/MapsManager.cpp.o' failed
make[2]: *** [rtabmap_ros-0.20.14-melodic/CMakeFiles/rtabmap_ros.dir/src/MapsManager.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
rtabmap_ros-0.20.14-melodic/CMakeFiles/rtabmap_ros.dir/build.make:62: recipe for target 'rtabmap_ros-0.20.14-melodic/CMakeFiles/rtabmap_ros.dir/src/MsgConversion.cpp.o' failed
make[2]: *** [rtabmap_ros-0.20.14-melodic/CMakeFiles/rtabmap_ros.dir/src/MsgConversion.cpp.o] Error 1
CMakeFiles/Makefile2:4945: recipe for target 'rtabmap_ros-0.20.14-melodic/CMakeFiles/rtabmap_ros.dir/all' failed
make[1]: *** [rtabmap_ros-0.20.14-melodic/CMakeFiles/rtabmap_ros.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
/usr/bin/ld: warning: libboost_regex.so.1.65.1, needed by /opt/ros/melodic/lib/libcv_bridge.so, may conflict with libboost_regex.so.1.61.0
/usr/bin/ld: warning: libopencv_photo.so.4.1, needed by /usr/local/lib/librtabmap_core.so, may conflict with libopencv_photo.so.3.2
/usr/bin/ld: warning: libopencv_stitching.so.4.1, needed by /usr/local/lib/librtabmap_core.so, may conflict with libopencv_stitching.so.3.2
/usr/bin/ld: warning: libopencv_video.so.4.1, needed by /usr/local/lib/librtabmap_core.so, may conflict with libopencv_video.so.3.2
/usr/bin/ld: warning: libboost_system.so.1.61.0, needed by /usr/local/lib/librtabmap_core.so, may conflict with libboost_system.so.1.65.1
/usr/bin/ld: warning: libopencv_calib3d.so.4.1, needed by /usr/local/lib/librtabmap_core.so, may conflict with libopencv_calib3d.so.3.2
/usr/bin/ld: warning: libopencv_features2d.so.4.1, needed by /usr/local/lib/librtabmap_core.so, may conflict with libopencv_features2d.so.3.2
/usr/bin/ld: warning: libopencv_videoio.so.4.1, needed by /usr/local/lib/librtabmap_core.so, may conflict with libopencv_videoio.so.3.2
/usr/bin/ld: warning: libopencv_core.so.4.1, needed by /usr/local/lib/librtabmap_core.so, may conflict with libopencv_core.so.3.2
[ 75%] Built target rtabmap_camera
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

43998213, 2021-11-15:
version: 0.20.14-melodic

matlabbe, 2021-11-15:
If you are using 0.20.14-melodic, A-uninstall rtabmap built from source or B-rebuild-install rtabmap version 0.20.14-melodic. Always use same branch/tag from rtabmap and rtabmap_ros repos to be compatible.

43998213, 2021-11-17:
> If you are using rtabmap_ros 0.20.14-melodic, rebuild-install rtabmap version 0.20.14-melodic. Always use same branch/tag from rtabmap and rtabmap_ros repos to be compatible.
> 
> ```shell
> cd rtabmap/build
> sudo make uninstall
> git checkout 0.20.14-melodic
> make -j6
> sudo make install
> ```
> 
> If you are using rtabmap master branch, then use rtabmap_ros master branch too.

Thanks a million

closed 2021-11-20 by matlabbe