introlab/rtabmap_ros #678: keep ground information

LioRage, 2021-11-16:
I wanna to use rtabmap to SLAM, but there is a filter in PCL and ground information has been filted. Are there some param to set for keep those information on rviz?

matlabbe, 2021-11-16:
Note sure exactly you are referring. Which topic is filtered? Which topic are you visualizing in RVIZ? Maybe a screenshot could help to understand.

LioRage, 2021-11-17:
Sorry for my confusing expession. I use the Kinect+Odometry+2Dlaser to SLAM with sets in Ros wiki. Obviously, I did not get point cloud for ground/floor/surface in /rtabmap/mapdata(line、sign and others). It seem like those information had been filtered. Then I search for the question, someone told me to look at the PCL filter, Progressive Morphological Filter or something else must work for clean those information. So I try to beg for help―is there any param to close this filter and let them register on RVIZ?

从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送

matlabbe, 2021-11-20:
You may visualize `cloud_map`, `cloud_obstacle` and `cloud_ground` topics from rtabmap node in RVIZ. Note that with laser scan data, `cloud_ground` will be empty.

LioRage, 2021-11-23:
Dear labbe
I found the source of my problem,  when I follow the way to visualize those topics, nothing can be found but allow me to doubt the msgs. So I check out the depth msgs from camera―ground can not get back ant depth msg, because of the structured light from camera has poor detection effect on the ground, and all the ground-depth are “zero”. Then rtabmap don’t registe ground imformation on RVIZ(I guess).
In a word, Cheap cameras caused this farce.
At the end of latter, send my best wish to all of researcher and maintainer, especially you. When I get significant progress, please allow me write your name on the Acknowledgements.
从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送

closed 2021-11-23 by matlabbe