introlab/rtabmap_ros #679: Versions used for each dependecy

felrock, 2021-11-17:
Hello, is there any info available for which version of the dependencies 0.20.15 uses for example. Would be really useful to know, thanks

matlabbe, 2021-11-20:
You can get version of major dependencies like OpenCV or PCL:
```
$ rtabmap --version
RTAB-Map:              0.20.16
PCL:                    1.10.0
With VTK:                7.1.1
OpenCV:                  4.2.0
```
you could also use `ldd` tool to show all libraries with their version used on your system.

felrock, 2021-11-25:
I'm building rtabmap from source, so I can't run ldd for example. But isn't there like a log, or defined version for each release?

matlabbe, 2021-11-25:
rtabmap supports a wide range of version of dependencies. Across machines and OS, the default VTK, PCL, OpenCV, Qt versions will vary. The master branch supports all versions from all os at the same time, so it is easier to say to everyone to use the master branch on every systems. For example if you are on Ubuntu 20.04, if you are using all system dependencies, PCL should be 1.10.0, VTK 7.1.1, OpenCV 4.2.0 and Qt 5.12.8. On Ubuntu 18.04, it will be different (PCL 1.8.1, VTK 6.3.0, OpenCV 3.2.0, Qt 5.9.5). For all other optional dependencies, it depends on which versions are installed, which is out of my control (some open source dependencies don't even have a version).

You should be able to use ldd even when built from source:
```
ldd ~/rtabmap/build/bin/rtabmap
```
Similar to "rtabmap --version", you can also launch rtabmap, then click on About menu option to see all dependencies.

If you are building rtabmap from source, then when you do cmake, there is a log status with all dependencies found.

closed 2023-04-11 by felrock