introlab/rtabmap_ros #681: An error occurred in the build in the case of ubuntu18.04 eloquent

miku54, 2021-11-19:
[image]
When I download A to the workspace to compile, there are the following problems, is there any way to solve it? My hardware is jetson nano, and the development environment is ubuntu18.04 eloquent。

CMake Error at CMakeLists.txt:58 (find_package):
  Could not find a configuration file for package "RTABMap" that is
  compatible with requested version "0.20.15".

  The following configuration files were considered but not accepted:

    /opt/ros/melodic/lib/rtabmap-0.20/RTABMapConfig.cmake, version: 0.20.7

matlabbe, 2021-11-21:
Hi, you should remove ros1 paths (don't call `source /opt/ros/melodic/setup.bash`) if you are using ros2. You will have to clone rtabmap repo inside your ros2 workspace. However, it is been a while since I tested eloquent, I have written "ROS2 Foxy minimum required" [here](https://github.com/introlab/rtabmap_ros/tree/ros2#rtabmap_ros-), but don't remember if it is a compilation issue or missing dependencies or some demos like turtlebot3 not working. Well, as Jetpack is still on 18.04, I could check to put more effort on supporting eloquent, if it is not a missing dependencies issue.

madiyarsh, 2022-02-18:
Hi, you can change the version requirement in CMakeLists