introlab/rtabmap_ros #682: Warning message from RTABMAP with melodic

sunglyoungKim, 2021-11-19:
Hello,

I am running the most recent version of RTABMAP + RTABMAP_ROS.

When I run turtle bot mapping with hokuyo 2D lidar on gazebo I am having the following warning message.  

Republishing data of requested node(s) 1 62 from "/rtabmap/republish_node_data" input topic (max_nodes_republished=2)

This error comes from rtabmap_ros CoreWrapper.CPP. Please let me know how I can avoid this warning or is it just worning so I can ignore it.

Thank you

matlabbe, 2021-11-21:
If the warning appears repeatedly with the same node Ids, there is maybe a problem. Only one time, it is fine. This feature is used by rtabmap_ros/MapCloud plugin in rviz, so only for visualization (you can ignore it).

sunglyoungKim, 2021-11-22:
Yes, it appears repeatedly with the same node Ids. 
[ WARN] [1637613417.983715149, 237.798000000]: Republishing data of requested node(s) 1 from "/rtabmap/republish_node_data" input topic (max_nodes_republished=2)
[ WARN] [1637613419.027147367, 238.819000000]: Republishing data of requested node(s) 1 from "/rtabmap/republish_node_data" input topic (max_nodes_republished=2)
[ WARN] [1637613420.014170661, 239.802000000]: Republishing data of requested node(s) 1 from "/rtabmap/republish_node_data" input topic (max_nodes_republished=2)
[ WARN] [1637613421.031935385, 240.816000000]: Republishing data of requested node(s) 1 from "/rtabmap/republish_node_data" input topic (max_nodes_republished=2)
[ WARN] [1637613422.015961740, 241.799000000]: Republishing data of requested node(s) 1 from "/rtabmap/republish_node_data" input topic (max_nodes_republished=2)
[ WARN] [1637613423.040042105, 242.821000000]: Republishing data of requested node(s) 1 from "/rtabmap/republish_node_data" input topic (max_nodes_republished=2)
[ WARN] [1637613424.034127983, 243.814000000]: Republishing data of requested node(s) 1 from "/rtabmap/republish_node_data" input topic (max_nodes_republished=2)
[ WARN] [1637613425.146311930, 244.917000000]: Republishing data of requested node(s) 1 from "/rtabmap/republish_node_data" input topic (max_nodes_republished=2)
[ WARN] [1637613426.167924983, 245.936000000]: Republishing data of requested node(s) 1 from "/rtabmap/republish_node_data" input topic (max_nodes_republished=2)

And yes if I turned off MapCloud on RVIZ the warning message is stoped.
Thank you so much!

matlabbe, 2021-11-25:
I still find it strange that it never clears the node 1 data. Can you show how rtabmap is started?

SYLJSTK, 2021-12-13:
I have the same problem. Have you solved it ?

sunglyoungKim, 2021-12-13:
Hello, 

I have updated the most recent RTABMAP and the Warn message publishes only at the beginning. 

Thank you for your hard work. @matlabbe

I am closing this issue.

Thank you

closed 2021-12-13 by sunglyoungKim