introlab/rtabmap_ros #683: rtab create a transform between map to testes

gongyue666, 2021-11-24:
Hi all, I test rtab with my robot, and use robot's odometry as a external odometry. here is my launch commands:
roslaunch rtabmap_ros rtabmap.launch    rtabmap_args:="\
      --delete_db_on_start \
      --Icp/VoxelSize 0.05 \
      --Icp/PointToPlaneRadius 0 \
      --Icp/PointToPlaneK 20 \
      --Icp/CorrespondenceRatio 0.2 \
      --Icp/PMOutlierRatio 0.65 \
      --Icp/Epsilon 0.005 \
      --Icp/PointToPlaneMinComplexity 0 \
      --Odom/ScanKeyFrameThr 0.7 \
      --OdomF2M/ScanMaxSize 15000 \
      --Optimizer/GravitySigma 0.3 \
      --RGBD/ProximityPathMaxNeighbors 1 \
      --Reg/Strategy 1"     icp_odometry:=false  odom_topic:=/odom frame_id:=/base_link   scan_cloud_topic:=/cloud     subscribe_scan_cloud:=true     depth_topic:=/camera/aligned_depth_to_color/image_raw     rgb_topic:=/camera/color/image_raw     camera_info_topic:=/camera/color/camera_info     approx_sync:=true     wait_imu_to_init:=true     imu_topic:=/rtabmap/imu visual_odometry:=false approx_rgbd_sync:=false

and the result logs are like this:
[ WARN] [1637724083.432914145]: Could not get transform from testes to /base_link after 0.200000 seconds (for stamp=1637724082.880244)! Error="Could not find a connection between 'testes' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.201644 timeout was 0.2.".

why rtab make a tf between map and testes？

matlabbe, 2021-11-25:
It is not making a TF between `testes` and `base_link`, it queries it on TF tree. Those frames would those in your `/odom` topic. Your odometry node should publish `testes` -> `odom` with same stamp than the one used in your topic.

gongyue666, 2021-11-26:
Thx, my robot's driver have a default odometry topic named "testes", I have modified.

closed 2021-11-26 by gongyue666