introlab/rtabmap_ros #685: RTABMAP pointcloud map outputs on a different coordinate system than gazeb

LuccaMorosiniGioia, 2021-11-30:
RTABMAP seems to be working fine, it's receiving data from the camera topics and odom topic, but the pointcloud map is being displayed on the wrong axis in relation to gazebo. What should be the front of the robot points to the Z axis.

I'm using gazebo11 and ros2 galactic with the turtlebot2i robot.

Here's some screenshots:

https://imgur.com/a/N6Eax9a

https://imgur.com/a/BVnyX1t

I also receive this warning at rtabmap terminal

Warning: Invalid frame ID "base_link" passed to canTransform argument source_frame - frame does not exist

But i can't seem to find where this base_link is coming from, i'm using the base_footprint for the robot frame.

Here's the config for the camera:

```
<gazebo reference="camera_link">
    <sensor name="camera" type="depth">
        <always_on>true</always_on>
        <update_rate>30.0</update_rate>
        <camera>
            <!-- <horizontal_fov>1.0472</horizontal_fov> -->
            <horizontal_fov>1.11701</horizontal_fov>
            <aspect_ratio>1.333333333333333</aspect_ratio>  
            <image>
                <format>B8G8R8</format>
                <width>640</width>
                <height>480</height>
            </image>
            <clip>
                <near>0.4</near>
                <far>8.0</far>
            </clip>
        </camera>
        <plugin name="camera_controller" filename="libgazebo_ros_camera.so">
            <ros>
                <!--<namespace></namespace>-->

                <!-- topics need to be prefixed with camera_name for remapping -->
                <argument>camera/image_raw:=camera/rgb/image_raw</argument>
                <argument>camera/depth/image_raw:=camera/depth/image_raw</argument>
                <argument>camera/camera_info:=camera/cam_info_raw</argument>
                <argument>camera/depth/camera_info:=camera/depth/cam_info_raw</argument>
                <argument>camera/points:=camera/points_raw</argument>
            </ros>

            <!-- omit camera_name to default to sensor name -->
            <!--<camera_name>astra</camera_name>-->

            <frame_name>base_footprint</frame_name>

            <min_depth>0.1</min_depth>
            <max_depth>500</max_depth>

            <baseline>0.1</baseline>
            <distortion_k1>-8.23811762e-01</distortion_k1>
            <distortion_k2>5.27469947e+00</distortion_k2>
            <distortion_k3>-6.06324274e-03</distortion_k3>
            <distortion_t1>-3.25047090e-02</distortion_t1>
            <distortion_t2>-1.35477771e+01</distortion_t2>
            <pointCloudCutoff>0.4</pointCloudCutoff>
        </plugin>
    </sensor>
</gazebo>
```

And the rtabmap launch config:
[... 71 lines cut ...]
    DeclareLaunchArgument('odom_args',                  default_value='',      description='More arguments for odometry (overwrite same parameters in rtabmap_args).'),
    DeclareLaunchArgument('odom_sensor_sync',           default_value='false', description=''),
    DeclareLaunchArgument('odom_guess_frame_id',        default_value='',      description=''),
    DeclareLaunchArgument('odom_guess_min_translation', default_value='0.0',   description=''),
    DeclareLaunchArgument('odom_guess_min_rotation',    default_value='0.0',   description=''),

    # imu
    DeclareLaunchArgument('imu_topic',        default_value='/imu/data', description='Used with VIO approaches and for SLAM graph optimization (gravity constraints).'),
    DeclareLaunchArgument('wait_imu_to_init', default_value='false',     description=''),

    # User Data
    DeclareLaunchArgument('subscribe_user_data',   default_value='false',            description='User data synchronized subscription.'),
    DeclareLaunchArgument('user_data_topic',       default_value='/user_data',       description=''),
    DeclareLaunchArgument('user_data_async_topic', default_value='/user_data_async', description='User data async subscription (rate should be lower than map update rate).'),

    #GPS
    DeclareLaunchArgument('gps_topic',  default_value='/gps/fix', description='GPS async subscription. This is used for SLAM graph optimization and loop closure candidates selection.'),

    # Tag/Landmark
    DeclareLaunchArgument('tag_topic',            default_value='/tag_detections', description='AprilTag topic async subscription. This is used for SLAM graph optimization and loop closure detection. Landmark poses are also published accordingly to current optimized map.'),
    DeclareLaunchArgument('tag_linear_variance',  default_value='0.0001',          description=''),
    DeclareLaunchArgument('tag_angular_variance', default_value='9999.0',            description='>=9999 means rotation is ignored in optimization, when rotation estimation of the tag is not reliable or not computed.'),
    OpaqueFunction(function=launch_setup)
])
```

matlabbe, 2021-12-05:
There is a missing optical rotation for the camera like this [turtlebot3 example](https://github.com/ROBOTIS-GIT/turtlebot3/blob/6268cec0beca93f298d1c2d54b13d4d63f241700/turtlebot3_description/urdf/turtlebot3_waffle.urdf#L241). I don't think the frame of the images should be base_footprint. Is the odometry topic publishing on "base_footprint" or "base_link"?

LuccaMorosiniGioia, 2021-12-05:
The odometry is being published on "odom"

matlabbe, 2021-12-06:
Didn't that used TF for odom. Can you show the log output of launch file?

LuccaMorosiniGioia, 2021-12-06:
I'm using the odometry output from the robot, it was being published on odom so i kept it. Is the log saved on a specific file or do you mean the output on the console?

matlabbe, 2021-12-06:
console output. For the odometry topic, there are 2 frames in there, one should be odom, but the other one could be the base_link frame you see.

LuccaMorosiniGioia, 2021-12-06:
Indeed there is a warning concerning the base_link. I can't send you the whole console output right now, but i have the warning message: [rviz2-3] Warning: Invalid frame ID "base_link" passed to canTransform argument source_frame - frame does not exist.
I can send the whole console output later

LuccaMorosiniGioia, 2021-12-11:
@matlabbe sorry it took so long to respond, but here's the console output:

```
[INFO] [launch]: All log files can be found below /home/luccagioia/.ros/log/2021-12-11-20-49-25-555478-LuccaGioia-12873
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [icp_odometry-1]: process started with pid [12875]
[INFO] [rtabmap-2]: process started with pid [12877]
[INFO] [rtabmapviz-3]: process started with pid [12879]
[INFO] [rviz2-4]: process started with pid [12881]
[INFO] [point_cloud_xyzrgb-5]: process started with pid [12883]
[rviz2-4] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-luccagioia'
[rviz2-4] 1639255766.754217 [42]      rviz2: using network interface eno1 (udp/192.168.2.167) selected arbitrarily from: eno1, docker0
[rviz2-4] [INFO] [1639255767.080819256] [rviz2]: Stereo is NOT SUPPORTED
[icp_odometry-1] 1639255767.085883 [42] icp_odomet: using network interface eno1 (udp/192.168.2.167) selected arbitrarily from: eno1, docker0
[rviz2-4] [INFO] [1639255767.086687845] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[point_cloud_xyzrgb-5] 1639255767.108536 [42] point_clou: using network interface eno1 (udp/192.168.2.167) selected arbitrarily from: eno1, docker0
[rviz2-4] [INFO] [1639255767.160791120] [rviz2]: Stereo is NOT SUPPORTED
[point_cloud_xyzrgb-5] [INFO] [1639255767.164827354] [point_cloud_xyzrgb]: Approximate time sync = true
[icp_odometry-1] [INFO] [1639255767.173222720] [rtabmap.icp_odometry]: Odometry: frame_id               = base_footprint
[icp_odometry-1] [INFO] [1639255767.173392762] [rtabmap.icp_odometry]: Odometry: odom_frame_id          = /odom
[icp_odometry-1] [INFO] [1639255767.173411961] [rtabmap.icp_odometry]: Odometry: publish_tf             = true
[icp_odometry-1] [INFO] [1639255767.173425570] [rtabmap.icp_odometry]: Odometry: wait_for_transform     = 0.200000
[icp_odometry-1] [INFO] [1639255767.173459861] [rtabmap.icp_odometry]: Odometry: initial_pose           = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000
[icp_odometry-1] [INFO] [1639255767.173472197] [rtabmap.icp_odometry]: Odometry: ground_truth_frame_id  = 
[icp_odometry-1] [INFO] [1639255767.173482755] [rtabmap.icp_odometry]: Odometry: ground_truth_base_frame_id = 
[icp_odometry-1] [INFO] [1639255767.173492814] [rtabmap.icp_odometry]: Odometry: config_path            = 
[icp_odometry-1] [INFO] [1639255767.173503006] [rtabmap.icp_odometry]: Odometry: publish_null_when_lost = true
[icp_odometry-1] [INFO] [1639255767.173513097] [rtabmap.icp_odometry]: Odometry: guess_frame_id         = 
[icp_odometry-1] [INFO] [1639255767.173523383] [rtabmap.icp_odometry]: Odometry: guess_min_translation  = 0.000000
[icp_odometry-1] [INFO] [1639255767.173536278] [rtabmap.icp_odometry]: Odometry: guess_min_rotation     = 0.000000
[icp_odometry-1] [INFO] [1639255767.173547941] [rtabmap.icp_odometry]: Odometry: guess_min_time         = 0.000000
[icp_odometry-1] [INFO] [1639255767.173558658] [rtabmap.icp_odometry]: Odometry: expected_update_rate   = 0.000000 Hz
[icp_odometry-1] [INFO] [1639255767.173569639] [rtabmap.icp_odometry]: Odometry: max_update_rate        = 0.000000 Hz
[icp_odometry-1] [INFO] [1639255767.173580339] [rtabmap.icp_odometry]: Odometry: wait_imu_to_init       = false
[icp_odometry-1] [INFO] [1639255767.173611570] [rtabmap.icp_odometry]: Odometry: stereoParams_=0 visParams_=0 icpParams_=1
[icp_odometry-1] [WARN] [1639255767.200796996] [rtabmap.icp_odometry]: IcpOdometry: Transferring value 0.05 of "Icp/VoxelSize" to ros parameter "scan_voxel_size" for convenience. "Icp/VoxelSize" is set to 0.
[icp_odometry-1] [WARN] [1639255767.200850504] [rtabmap.icp_odometry]: IcpOdometry: Transferring value 5 of "Icp/PointToPlaneK" to ros parameter "scan_normal_k" for convenience.
[icp_odometry-1] [WARN] [1639255767.200871177] [rtabmap.icp_odometry]: IcpOdometry: Transferring value 1.0 of "Icp/PointToPlaneRadius" to ros parameter "scan_normal_radius" for convenience.
[icp_odometry-1] [INFO] [1639255767.212977343] [rtabmap.icp_odometry]: IcpOdometry: scan_cloud_max_points  = 0
[icp_odometry-1] [INFO] [1639255767.213126936] [rtabmap.icp_odometry]: IcpOdometry: scan_downsampling_step = 1
[icp_odometry-1] [INFO] [1639255767.213188804] [rtabmap.icp_odometry]: IcpOdometry: scan_range_min         = 0.000000 m
[icp_odometry-1] [INFO] [1639255767.213229300] [rtabmap.icp_odometry]: IcpOdometry: scan_range_max         = 0.000000 m
[icp_odometry-1] [INFO] [1639255767.213243571] [rtabmap.icp_odometry]: IcpOdometry: scan_voxel_size        = 0.050000 m
[icp_odometry-1] [INFO] [1639255767.213261118] [rtabmap.icp_odometry]: IcpOdometry: scan_normal_k          = 5
[icp_odometry-1] [INFO] [1639255767.213272455] [rtabmap.icp_odometry]: IcpOdometry: scan_normal_radius     = 1.000000 m
[icp_odometry-1] [INFO] [1639255767.213283867] [rtabmap.icp_odometry]: IcpOdometry: scan_normal_ground_up  = 0.000000
[rtabmapviz-3] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-luccagioia'
[rtabmapviz-3] 1639255767.236427 [42] rtabmapviz: using network interface eno1 (udp/192.168.2.167) selected arbitrarily from: eno1, docker0
[rtabmapviz-3] [INFO] [1639255767.268424697] [rtabmap.rtabmapviz]: rtabmapviz: Using configuration from "/home/luccagioia/.ros/rtabmapGUI.ini"
[rtabmap-2] 1639255767.271596 [42]    rtabmap: using network interface eno1 (udp/192.168.2.167) selected arbitrarily from: eno1, docker0
[... cut ...]
[rviz2-4] Warning: Invalid frame ID "base_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-4]          at line 156 in /home/luccagioia/Documentos/IC-Turtlebot2i-JetsonNano/src/ros2/geometry2/tf2/src/buffer_core.cpp
[rviz2-4] Warning: Invalid frame ID "base_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-4]          at line 156 in /home/luccagioia/Documentos/IC-Turtlebot2i-JetsonNano/src/ros2/geometry2/tf2/src/buffer_core.cpp
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
```

matlabbe, 2021-12-14:
The warning looks coming from rviz, not rtabmap. You would have to debug your tf tree, or find which topic you added to rviz that is requesting `base_link` frame.

For rtabmap nodes, they are all set with `base_footprint`:
```
[icp_odometry-1] [INFO] [1639255767.173222720] [rtabmap.icp_odometry]: Odometry: frame_id               = base_footprint
[rtabmap-2] [INFO] [1639255767.373218462] [rtabmap.rtabmap]: rtabmap: frame_id      = base_footprint
```

matlabbe, 2021-12-14:
If there is really an optical rotation in TF, make sure the frame_id in the image topics are in the optical frame, not the camera frame. In simulation, it is often the problem, the frame_id in the image topic is not the one of the optical rotation.

LuccaMorosiniGioia, 2022-01-07:
Forgot to close this issue. sorry for that. 

The problem was solved. I wasn't publishing all the tf's so rtab-map wasn't able to properly understand the information from the camera. Had to create a node robot_state_publisher to publish the urdf's tf's. After that the problem was solved.

Really appreciated the help,
Lucca

closed 2022-01-07 by LuccaMorosiniGioia

Dragonrock, 2023-04-21:
I have been using Matlab's Simulink with co-simulation in Unreal, to simulate a rgbd camera. I had to create all the the ROS2 msgs by hand and have the same issue, where the point cloud is always displayed on the z-axis while the robot is moving and rotating in the virtual environment.

Can you provide some details, on the layout and data of your tf's, and the node you used to publish them?

Would really appreciate it,
Thanks