introlab/rtabmap_ros #686: Rtabmap not generate MapData to rviz

ragane, 2021-12-02:
Hello,
I have problem with generating map in rviz because i just can spark PointCloud2 to generate points of map. I don't even know why voxel_cloud is not generated. 
I use Kinect v1 1414 connected to RPi 4B on Ubuntu 20.04 with ROS noetic version. Rviz is sparked on PC on ubuntu 20.04 too.
Thanks in advance for any advices.

matlabbe, 2021-12-06:
Which topic are you looking at exactly? MapData should be visualized with rtabmap_ros/MapCloud rviz plugin

howtoloveyou, 2021-12-30:
have you resolved this problem? i can't get topic "/mapData" in rviz(in ros2).
and when i want to use " ros2 topic echo /mapData" to get the information of "/mapData", ros2 report an error : module rtabmap_ros not found.
can you help me? thank you.

ragane, 2022-01-02:
Thank You Matlabbe for fast answer. Im so sorry that I havent answered you so long time but I was very busy and it was christmas time. 
Yes I resolved my problem, Im newbie in rtbamap and my problem was that I tried to spark rviz on rpi console by SSH and topics like MapData/ MapGraph werent seen. Solution was to spark rtabmap on rpi without rviz and simulatenously use rtabmap on PC and everything is visible.

closed 2022-01-19 by ragane