introlab/rtabmap_ros #687: wrong localization

smileyenot983, 2021-12-02:
Robot localizes itself with it's last coordinates before rtabmap was closed. 

What i want to do is the following:
1. launch rtabmap for creating the map with param localization=False, close rtabmap
2. change the robot's position while rtabmap is off. 
3. launch rtabmap for navigation with param localization = True. And here the problem occurs. Robot localizes itself at the same place where it was at the moment when i finished creating the map at step 1. 

Here is visualization of step 3 at rviz and gazebo:

In gazebo robot stays at the initial position(but that could be some other position, which is not the same as final position at step 1):

[image]

In rviz(rectangle) it shows like the robot is at completely different position(position at which mapping was finished at step 1): 

[image]

smileyenot983, 2021-12-02:
For some reason such behavior is only observed when parameter "Mem/IncrementalMemory" is set to false. In localization mode it is set to false by default. When i change it to true localization is done correctly.
Is that ok?

matlabbe, 2021-12-02:
> In rviz(rectangle) it shows like the robot is at completely different position(position at which mapping was finished at step 1):

Yes, by default the robot thinks it is restarting from the last location where it shut down. See this parameter to change this behavior (if you want the robot to restart from origin of the map instead):

- Param: RGBD/StartAtOrigin = "false"                        [If true, rtabmap will assume the robot is starting from origin of the map. If false, rtabmap will assume the robot is restarting from the last saved localization pose from previous session (the place where it shut down previously). Used only in localization mode (Mem/IncrementalMemory=false).]

smileyenot983, 2021-12-03:
> Yes, by default the robot thinks it is restarting from the last location where it shut down. See this parameter to change this behavior (if you want the robot to restart from origin of the map [...]

Thanks for elaboration. Seems like the problem is solved.

closed 2021-12-03 by smileyenot983