introlab/rtabmap_ros #689: Multi session mapping for 3D LiDAR

robertsenputras, 2021-12-06:
Hi @matlabbe ,

I have searched for instructions regarding how to do multi-session mapping with only 3D lidar, and I seem didn't find the answer. For 3D LiDAR, is it possible to do multi-session mapping and stitch the db file together ? If so, could you share with me how to do it? I think a lot of people also need to know about this in the future. hehe. 

Thank you.

matlabbe, 2021-12-10:
### Add manually a link to merge two sessions
For the first question, yes it is possible to create a multi-session database with only lidar. You have to manually link the session afterwards in rtabmap-databaseViewer (https://github.com/introlab/rtabmap/issues/497). Here is an example using demo robot mapping bag played 2 times to create two disjoint sessions (visual loop closure detection disabled to simulate only lidar config):
```bash
$ roslaunch rtabmap_ros demo_robot_mapping.launch rtabmap_args:="-d --Kp/MaxFeatures -1"
$ rosbag play --clock demo_mapping.bag 
# Play till 110 seconds and kill all, then restart from 130 seconds 
# in the bag without clearing the previous session
$ roslaunch rtabmap_ros demo_robot_mapping.launch rtabmap_args:="--Kp/MaxFeatures -1"
$ rosbag play --clock -s 130 demo_mapping.bag
# Play till 85 seconds then kill all
```

Now you should have two disjoint maps where visually no loop closures can be found because the robot traversed the area in opposite directions. However, the lidar should see the same thing at some locations. Open the database in rtabmap-databaseViewer:
```bash
$ rtabmap-databaseViewer ~/.ros/rtabmap.db
```
In the Graph View, it will show one of the two maps, based on the root id parameter. You can switch session by setting a root id from another session. You can find the first node of each session by moving the main sliders until an image shows a label with "map0" or "map1". Here an example of switching the root id in Graph View:
[image]
[image]

To merge the session manually, select two nodes that should be seeing the same structures. Here we choose node 1 and 187 and click on Add button under Constraints View:
[image]
Press Yes for the first dialog. It will then show this one:
[image]
It tried to match the laser scans using Identity transform between them, but it is likely to fail as in reality the nodes have a rotation of about 80 deg between them. Press yes to enter manually the transform and set 80 deg for the yaw (uncheck Radian checkbox to set degrees):
[image]
Press Ok, you should then see in Constraints View the actual constraint between the node and in Graph View, you should see the two maps merged together.
[image]
 Press Refine to better estimate the transform.
[image]
Our new constraint looks pretty accurate, but the two paths don't superpose perfectly, we need to add more constraints. Here an example to better align the bottom right corner (nodes 45 and 148):
[image]
When we hit Add, we would get again this error:
[image]
For this case, as the guess transform should be already quite close (as shown in the Constraints View), we will increase Icp/MaxTranslation to 0.3 to accept that constraint (open Core Parameters->Icp panel):
[image]
Click Add again, the constraint should be accepted and would appear in red in Graph View:
[image]

We could continue manually like that to add more and more constraints to better optimize the whole map, but we can also use Detect More Loop Closures tool to automatically add them. Open Gui Parameters View, go under Detect More Loop Closures panel and set those parameters (find loop closures in any orientation, force only loop closures between the sessions):
[image]
Click on Edit->"Detect more loop closures...", then the final results will look like this:
[image]

### Generate local grids offline
For the grid map generation, you can set RGBD/CreateOccupancyGrid to false to avoid creating the local grids online. To recreate them afterwards with ray tracing, there is an option in rtabmap-databaseViewer (menu Edit->Regenerate all local grids...) to regenerate the local grids based on the parameters set in Core Parameters panel. 
[image]

You can also use rtabmap-reprocess tool like this:
```bash
$ rtabmap-reprocess --RGBD/CreateOccupancyGrid true --Grid/RayTracing true input.db output.db
```

robertsenputras, 2021-12-10:
Thank you for your detailed explanation about the map stitching for multi-session mapping. I really appreciate your answer, It's very clear and detailed. It gives us, especially me a very deep understanding about your package. Glad to find your powerful package. Hope the others also find this explanation in the future. 

Thanks a lot.

closed 2021-12-10 by robertsenputras

matlabbe, 2021-12-10:
You're welcome!