introlab/rtabmap_ros #691: Table is not visible in the gridmap during mapping

benoit-robotics, 2021-12-10:
I am using only a rgbd camera to map an environment. In the GIF, you can see the issue that I am experiencing : the table ( in white ) doesn't appear in the gridmap, nonetheless it's clearly visible in the rtablab accumulated point cloud. The other obstacles, like the yellow chair, appear in the gridmap.
[image]

Here are the parameters that I use :

> 
>           <param name="RGBD/NeighborLinkRefining" type="string" value="true"/>
>           <param name="RGBD/ProximityBySpace"     type="string" value="true"/>
>           <param name="RGBD/AngularUpdate"        type="string" value="0.01"/>
>           <param name="RGBD/LinearUpdate"         type="string" value="0.01"/>
>           <param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/>
>           <param name="Grid/FromDepth"            type="string" value="true"/> 
>           <param name="Grid/RayTracing"            type="string" value="true"/>
>           <param name="Reg/Force3DoF"             type="string" value="true"/>
>           <param name="Reg/Strategy"              type="string" value="1"/> 

Is there a way to tweak the parameters of rtabmap to include the table in the gridmap ?

matlabbe, 2021-12-10:
You may try this parameter:
```xml
<param name="Grid/MaxGroundHeight"            type="string" value="0.3"/> 
```
All flat surfaces over 30 cm will be considered obstacles.

benoit-robotics, 2021-12-10:
Thank you. Is it 30 cm over the base frame ? If it's that, I am sure that the table is higher than that, but I will play with this parameter.

matlabbe, 2021-12-10:
Yes it is over the base frame (`frame_id` parameter of rtabmap node) of the robot. I guessed that the table was easily over 30 cm.

benoit-robotics, 2021-12-10:
This parameter doesn't seem to have any effect on my problem. Do you have any other input to give me ?

matlabbe, 2021-12-10:
Can you share that database?

benoit-robotics, 2021-12-13:
You can download the database here : https://drive.google.com/file/d/1g72_uYbzIOyOgh0THqQU6-1XQIjXE6Ce/view?usp=sharing . Cheers

matlabbe, 2021-12-14:
Hi,

here are some options you have (starting from your current parameters):

1) Your current solution:
[image]

2) With `Grid/MaxGroundHeight=0.3`:
[image]

3) With `Grid/MaxGroundHeight=0.3` and using `octomap_grid` topic:
[image]

4) With `Grid/MaxGroundHeight=0.3` and `Grid/3D=False`:
[image]

5) With `Grid/MaxGroundHeight=0.3` and `Grid/RayTracing=False`:
[image]

You can play with Grid parameters with rtabmap-databaseViewer, then do Edit->Regenerate local grids... to update the map. To show OctoMap result, make sure to check OctoMap checkbox under Gui Parameters -> Grid/Cloud Generation panel. "Show empty space" checkbox can be check or unchecked to better see obstacles or ray tracing. Check Grid checkbox under 3D View to show local grids, while pressing '2' key to show green/red segmentation.
[image]

benoit-robotics, 2021-12-14:
Thank you a lot for your analysis, I can now fix my problem.

closed 2021-12-14 by benoit-robotics

snakehaihai, 2022-11-06:
HI
Im also not able to use the MaxGroundHeight parameter to detect the table seems whatever I change, It does not affect that's being show in gridmap. I`m still only able to see the table leg and not the projected table-top occupancy  

and also i have an additional issue in viewer

[image]

can you help to address it? @matlabbe

matlabbe, 2022-11-26:
Try `Edit->Regenerate local grids...`. I see Grid/3D parameter is set, it should recreate the grid in 3D. If there is nothing shown, maybe there is an issue with the sensor used.