introlab/rtabmap_ros #692: The Map Cloud is not displayed in the rviz

SYLJSTK, 2021-12-13:
[image]
 hello， I use astra pro and ubuntu18,When I run rviz ,The following warning message appears: ,And it always appears,

[ WARN] [1639221548.736497604]: Republishing data of requested node(s) 1 3 from "/rtabmap/republish_node_data" input topic (max_nodes_republished=2) 

[ WARN] [1639224113.431464640]: Republishing data of requested node(s) 12 3 from "/rtabmap/republish_node_data" input topic (max_nodes_republished=2)

Here are my launch:
<launch>

  <include file="$(find turn_on_wheeltec_robot)/launch/mapping.launch" />

  <include file="$(find astra_camera)/launch/astrapro.launch" />

  <include file="$(find turn_on_wheeltec_robot)/launch/include/rtabmap_mapping.launch" />
</launch>

<launch> 
  <group ns="rtabmap"> 
 
     
     
 
    <arg name="database_path" default="~/.ros/test.db"/> 
 
    <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="standalone rtabmap_ros/rgbd_sync" output="screen"> 
      <remap from="rgb/image"       to="/camera/rgb/image_raw"/> 
      <remap from="depth/image"     to="/camera/depth/image"/> 
      <remap from="rgb/camera_info" to="/camera/rgb/camera_info"/> 
      <remap from="rgbd_image"      to="rgbd_image"/>  
      <remap from="rtabmap/get_map_data" to="get_map_data"/> 
       
       
      <param name="approx_sync"       value="true"/>  
    </node> 
 
    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start"> 
          <param name="frame_id" type="string" value="base_footprint"/> 
          <param name="database_path"       type="string" value="$(arg database_path)"/> 
          <param name="subscribe_depth" type="bool" value="false"/> 
          <param name="subscribe_rgb"   type="bool" value="false"/> 
          <param name="subscribe_rgbd"  type="bool" value="true"/> 
          <param name="subscribe_scan"  type="bool" value="true"/> 
 
          <remap from="odom" to="/odom"/> 
          <remap from="scan" to="/scan"/> 
          <remap from="rgbd_image" to="rgbd_image"/> 
 
          <param name="queue_size" type="int" value="10"/> 
 
           
          <param name="RGBD/NeighborLinkRefining" type="string" value="true"/> 
          <param name="RGBD/ProximityBySpace"     type="string" value="true"/> 
          <param name="RGBD/AngularUpdate"        type="string" value="0.01"/> 
          <param name="RGBD/LinearUpdate"         type="string" value="0.01"/> 
          <param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/> 
          <param name="Grid/FromDepth"            type="string" value="false"/>  
 
	 <param name="Vis/EstimationType"            type="string" value="0"/>  
 
           
          <param name="Reg/Force3DoF"             type="string" value="true"/> 
          <param name="Reg/Strategy"              type="string" value="1"/>  
           
           
          <param name="Icp/VoxelSize"                 type="string" value="0.05"/> 
          <param name="Icp/MaxCorrespondenceDistance" type="string" value="0.1"/> 
    </node> 
  </group> 
</launch>

SYLJSTK, 2021-12-13:
miss the start
<launch>
  <include file="$(find turn_on_wheeltec_robot)/launch/mapping.launch" />
  <include file="$(find astra_camera)/launch/astrapro.launch" />
  <include file="$(find turn_on_wheeltec_robot)/launch/include/rtabmap_mapping.launch" />
</launch>

<launch> 
  <group ns="rtabmap">

SYLJSTK, 2021-12-13:
[image]
[image]
[image]

SYLJSTK, 2021-12-15:
I tried and found that the camera calibration file was missing. I've solved the problem. Thank you

closed 2021-12-15 by SYLJSTK

wht88888, 2024-05-06:
> I tried and found that the camera calibration file was missing. I've solved the problem. Thank you

Hello,I 've got the same problem,could u plz tell how to solve this,thanks!