introlab/rtabmap_ros #693: melodic-devel: Not up to date with upstream RTABMAP

jmachuca77, 2021-12-14:
I was trying to use the melodic-devel branch and its no longer compiling due to rtabmap upstream changes, using master fixed it is there a timeline for a new release of the melodic-devel branch?
See this issue for similar issue on the ROS2 branch #674, fixed with commit ab1288cccc846bb60c142c82265151d0ecd1d0cf

matlabbe, 2021-12-14:
You should be using same branch on both projects. If melodic-devel branch of rtabmap doesn't work with melodic-devel branch of rtabmap_ros, then there is a bug. Otherwise, if you are using rtabmap master branch, you should be using rtabmap_ros branch. I update <ROS_DISTRO>_devel branches only when we want a new binary release on ros build farm.

jmachuca77, 2021-12-21:
Ah, ok, thanks.

closed 2021-12-21 by jmachuca77