introlab/rtabmap_ros #694: Localization_pose not published in localization mode in Ouster example

yadawate, 2021-12-18:
I am using ouster os-1 32 gen-2 lidar only for 3D mapping and localisation using rtabmap. I am using launch files given with rtabmap_ros ("test_ouster_gen2.launch") for mapping as it is for mapping. 
For localization I am using the file by removing args="-d" from line 111 and adding Mem/IncrementalMemory ="false" after that. The problem is that, I am getting localization_pose when in mapping mode but not in localization mode. I am using Ouster lidar only. Can someone help me pointing out where I am going wrong?

hibshibshibs, 2022-02-25:
Any luck with this??

matlabbe, 2022-03-06:
`localization_pose` is published only when a localization occurs in localization mode. If it is not published, it means it cannot localize on the map. With lidar-only, rtabmap cannot do global localization (for which a camera is required). However, it assumes by default that the robot/sensor is restarting from the last known location where rtabmap got shut down. So make sure to keep the lidar at the same position when you kill `test_ouster_gen2.launch` then relaunch it in localization mode. If you want to restart the lidar from the origin of the map (where the map started), set `RGBD/StartAtOrigin=true`.

closed 2022-03-07 by yadawate