introlab/rtabmap_ros #696: Not able to navigate after mapping an enviorement with rtab-map on ros2

LuccaMorosiniGioia, 2022-01-04:
After mapping an enviroment using rtabmap for ros2 i switched to localization mode and tried to make the turtlebot2i autonomous navigate, but was inable to properly send the commands to the robot using the goal_pose rviz2 interface. Rtab-map does not seem to notice that a goal_pose was published by rviz so nothing happens. The goal_pose topic published by rviz appears to be fine but all the rtab-map related topics show nothing.

Here's the rtab-map launch file: 
```
#
# To avoid log buffering:
# "stdbuf -o L ros2 launch rtabmap_ros rtabmap.launch.py ..."
#

import os

from launch import LaunchDescription, Substitution, LaunchContext
from launch.actions import DeclareLaunchArgument, SetEnvironmentVariable, LogInfo, OpaqueFunction
from launch.substitutions import LaunchConfiguration, ThisLaunchFileDir, PythonExpression
from launch.conditions import IfCondition, UnlessCondition
from launch_ros.actions import Node
from launch_ros.actions import SetParameter
from typing import Text
from ament_index_python.packages import get_package_share_directory

#Based on https://answers.ros.org/question/363763/ros2-how-best-to-conditionally-include-a-prefix-in-a-launchpy-file/
class ConditionalText(Substitution):
    def __init__(self, text_if, text_else, condition):
        self.text_if = text_if
        self.text_else = text_else
        self.condition = condition

    def perform(self, context: 'LaunchContext') -> Text:
        if self.condition == True or self.condition == 'true' or self.condition == 'True':
            return self.text_if
        else:
            return self.text_else
            
class ConditionalBool(Substitution):
    def __init__(self, text_if, text_else, condition):
        self.text_if = text_if
        self.text_else = text_else
        self.condition = condition

    def perform(self, context: 'LaunchContext') -> bool:
        if self.condition:
            return self.text_if
        else:
            return self.text_else
            
def launch_setup(context, *args, **kwargs):      
               
    return [
        DeclareLaunchArgument('depth', default_value=ConditionalText('false', 'true', IfCondition(PythonExpression(["'", LaunchConfiguration('stereo'), "' == 'true'"]))._predicate_func(context)), description=''),
        DeclareLaunchArgument('subscribe_rgb', default_value=LaunchConfiguration('depth'), description=''),
        DeclareLaunchArgument('args',  default_value=LaunchConfiguration('rtabmap_args'), description='Can be used to pass RTAB-Map\'s parameters or other flags like --udebug and --delete_db_on_start/-d'),
        
        #These arguments should not be modified directly, see referred topics without "_relay" suffix above
        DeclareLaunchArgument('rgb_topic_relay',      default_value=ConditionalText(''.join([LaunchConfiguration('rgb_topic').perform(context), "_relay"]), ''.join(LaunchConfiguration('rgb_topic').perform(context)), LaunchConfiguration('compressed').perform(context)), description='Should not be modified manually!'),
        DeclareLaunchArgument('depth_topic_relay',      default_value=ConditionalText(''.join([LaunchConfiguration('depth_topic').perform(context), "_relay"]), ''.join(LaunchConfiguration('depth_topic').perform(context)), LaunchConfiguration('compressed').perform(context)), description='Should not be modified manually!'),
        DeclareLaunchArgument('left_image_topic_relay',      default_value=ConditionalText(''.join([LaunchConfiguration('left_image_topic').perform(context), "_relay"]), ''.join(LaunchConfiguration('left_image_topic').perform(context)), LaunchConfiguration('compressed').perform(context)), description='Should not be modified manually!'),
        DeclareLaunchArgument('right_image_topic_relay',      default_value=ConditionalText(''.join([LaunchConfiguration('right_image_topic').perform(context), "_relay"]), ''.join(LaunchConfiguration('right_image_topic').perform(context)), LaunchConfiguration('compressed').perform(context)), description='Should not be modified manually!'),
        DeclareLaunchArgument('rgbd_topic_relay',      default_value=ConditionalText(''.join(LaunchConfiguration('rgbd_topic').perform(context)), ''.join([LaunchConfiguration('rgbd_topic').perform(context), "_relay"]), LaunchConfiguration('rgbd_sync').perform(context)), description='Should not be modified manually!'),
    
        SetParameter(name='use_sim_time', value=LaunchConfiguration('use_sim_time')),
        # 'use_sim_time' will be set on all nodes following the line above
    
        # Relays RGB-Depth
        Node(
            package='image_transport', executable='republish', name='republish_rgb',
            condition=IfCondition(PythonExpression(["'", LaunchConfiguration('stereo'), "' != 'true' and ('", LaunchConfiguration('subscribe_rgbd'), "' != 'true' or '", LaunchConfiguration('rgbd_sync'),"'=='true') and '", LaunchConfiguration('compressed'), "' == 'true'"])),
            remappings=[
                (['in/', LaunchConfiguration('rgb_image_transport')], [LaunchConfiguration('rgb_topic'), '/', LaunchConfiguration('rgb_image_transport')]),
                ('out', LaunchConfiguration('rgb_topic_relay'))], 
            arguments=[LaunchConfiguration('rgb_image_transport'), 'raw'],
[... 351 lines cut ...]
        DeclareLaunchArgument('odom_guess_min_translation', default_value='0.0',   description=''),
        DeclareLaunchArgument('odom_guess_min_rotation',    default_value='0.0',   description=''),
        
        # imu
        DeclareLaunchArgument('imu_topic',        default_value='/imu/data', description='Used with VIO approaches and for SLAM graph optimization (gravity constraints).'),
        DeclareLaunchArgument('wait_imu_to_init', default_value='false',     description=''),
        
        # User Data
        DeclareLaunchArgument('subscribe_user_data',   default_value='false',            description='User data synchronized subscription.'),
        DeclareLaunchArgument('user_data_topic',       default_value='/user_data',       description=''),
        DeclareLaunchArgument('user_data_async_topic', default_value='/user_data_async', description='User data async subscription (rate should be lower than map update rate).'),
        
        #GPS
        DeclareLaunchArgument('gps_topic',  default_value='/gps/fix', description='GPS async subscription. This is used for SLAM graph optimization and loop closure candidates selection.'),

        # Tag/Landmark
        DeclareLaunchArgument('tag_topic',            default_value='/tag_detections', description='AprilTag topic async subscription. This is used for SLAM graph optimization and loop closure detection. Landmark poses are also published accordingly to current optimized map.'),
        DeclareLaunchArgument('tag_linear_variance',  default_value='0.0001',          description=''),
        DeclareLaunchArgument('tag_angular_variance', default_value='9999.0',            description='>=9999 means rotation is ignored in optimization, when rotation estimation of the tag is not reliable or not computed.'),
        OpaqueFunction(function=launch_setup)
    ])

```
 
I was not sure what else might be needed so anything that might help i can provide later.

matlabbe, 2022-01-07:
I don't have examples of rtabmap + nav2 integration yet. However, rtabmap can provide the occuapancy grid and tf map->odom, so if the robot is not moving, it is a nav2 problem, or the map from rtabmap is not correctly connected to nav2.

LuccaMorosiniGioia, 2022-01-07:
According to nav2 documentation i need tf between map->odom, odom->base_link and base_link->base_laser(the last one i'm assuming is the base_link->base_footprint in my case, since i'm not using only the sensor, but also the robot). 
Since all these tf's seem to be fine i think the problem is linking the map generated from rtab-map to nav2, and from what i can tell nav2 expects the map as .yaml and rtab-map outputs in the form rtabmap.db how can i get the map from rtab-map in a different format? Or can i directly feed nav2 from the rtab-map topics?

matlabbe, 2022-01-07:
nav2 seems using /map topic for the global costmap (like move_base in ros1). You have to remap `/rtabmap/grid_map` to `/map` topic so that rtabmap and nav2 are connected.

LuccaMorosiniGioia, 2022-01-07:
Oh i see, thanks a lot for the help! 
One more thing if you may, Rtab-map also provides topics for plannning such as local_path and global_path and i didn't understand if  these are the ones which will determine the path of the robot or if this is a function of the nav2 pkg. If not can i link this path to nav2?
Again, i really appreciate your help! Thanks!

matlabbe, 2022-01-07:
Hi, the local_path and global_path are not yet integrated with nav2, those are topics ported from ros1. They are however debugging outputs when using RTAB-Map's internal planning on top of move_base (in case of nav2, they will be on top of nav2). They are the cyan and purple paths in figure 11 of this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/8/87/LabbeAURO2017.pdf).

I didn't look yet in details how we can communicate with nav2, but if there is an ActionLib's like goal input to nav2, I will connect rtabmap to it so that next goals planned by rtabmap will be sent to nav2 directly.

I assume that nav2, like move_base, would update its global costmap when receiving a new map on /map. If so, you can use nav2 with rtabmap during both mapping/SLAM phase and localization-only phase.

cheers,
Mathieu

LuccaMorosiniGioia, 2022-01-07:
Everythings is clearer to me now. Thanks a lot!

Sincerely, 
Lucca

closed 2022-01-07 by LuccaMorosiniGioia

matlabbe, 2022-01-16:
Hi,

I [updated](https://github.com/introlab/rtabmap_ros/commit/bc45e99ba5f866efe5ac46c23efd2f12c4a79c82#diff-1f3d076512e446864104a5fe77a25213d69e89cbf81d300e985c22cd5dca9c01) rtabmap_ros for integration with nav2. I updated turtebot3 [launch file examples](https://github.com/introlab/rtabmap_ros/tree/ros2/launch/ros2). Here is an example of usage:
```bash
$ ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
$ ros2 launch rtabmap_ros turtlebot3_scan.launch.py
$ ros2 launch nav2_bringup navigation_launch.py use_sim_time:=True
$ ros2 launch nav2_bringup rviz_launch.py
```

There was a problem with QoS setting of the published map topics, causing the topics not seen by global_costmap and rviz. The topic should be Reliable and Transient (latched), instead of Reliable and Volatile.

LuccaMorosiniGioia, 2022-01-22:
Hi,

So glad to hear that. I'm sure it will help me a lot. thanks a lot. So what does the fact that the map topics are now Transient matter to how i should use it with nav2? I can still link them and information will flow ensuring that i can map and navigate at the same time, right? The map is just updated differently

matlabbe, 2022-01-25:
[Transient](https://docs.ros.org/en/rolling/Concepts/About-Quality-of-Service-Settings.html) in ros2 is like "[latched](http://wiki.ros.org/roscpp/Overview/Publishers%20and%20Subscribers)" topic in ros1. It keeps the topic alive when there are no subscribers. It should not change how the map is updated.