introlab/rtabmap_ros #697: Catkin_make error while compiling rtabmap_ros in ubuntu 20.04

Rafi-UST, 2022-01-05:
Hi Team, 

I am getting some errors while compiling rtabmap_ros pacakage in my catkin_worksapace. 

Specifications of my PC being : 

OS : Ubuntu 20.04
RAM : 4 GB 

The error portion while compiling  is attached below : 

/home/rafism/catkin_ws/src/src/rtabmap_ros/src/MsgConversion.cpp:1558:7: error: ‘rtabmap_ros::OdomInfo’ {aka ‘struct rtabmap_ros::OdomInfo_<std::allocator<void> >’} has no member named ‘localBundlePoses’; did you mean ‘localBundleTime’?
 1558 |   msg.localBundlePoses.push_back(pose);
      |       ^~~~~~~~~~~~~~~~
      |       localBundleTime
make[2]: *** [src/rtabmap_ros/CMakeFiles/rtabmap_ros.dir/build.make:63: src/rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MsgConversion.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:16825: src/rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j2 -l2" failed

Please feel free to ask if any other particulars needed for clarification.

matlabbe, 2022-01-05:
Hi, 

localBundlePoses exists: https://github.com/introlab/rtabmap_ros/blob/0503f1a1f6edb1e269499f8e1f18187a1b2dd62d/msg/OdomInfo.msg#L39

Force re-run cmake: 
```
/home/rafism/catkin_ws/build
cmake .
```

Rafi-UST, 2022-01-06:
Hi, 

I am facing a different Cmake error now :

CMake Error at rtabmap_ros/CMakeLists.txt:35 (find_package):
  Could not find a configuration file for package "RTABMap" that is
  compatible with requested version "0.20.16".

  The following configuration files were considered but not accepted:

    /opt/ros/noetic/lib/x86_64-linux-gnu/rtabmap-0.20/RTABMapConfig.cmake, version: 0.20.14

-- Configuring incomplete, errors occurred!
See also "/home/rafism/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/rafism/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

matlabbe, 2022-01-07:
You should reinstall rtabmap 0.20.14-noetic too. You should use only compatible branches together (rtabmap 0.20.14 with rtabmap_ros 0.20.14, or rtabmap master with rtabmap_ros master).

Rafi-UST, 2022-01-07:
Hi @matlabbe , 

Yes I have did that and also I removed CmakeCache.txt from catkin_ws/build . I think so in some earlier issues you have mentioned that rtabmap_ros will try to link with ros-noetic-rtabmap instead of the one built locally in our catkin_ws/devel. The compilation error is resolved now after removing the same.

Thanks for all the support @matlabbe .

closed 2022-01-07 by matlabbe