introlab/rtabmap_ros #699: Rtabmapviz not giving any visual after launching a custom mapping.launch file

Rafi-UST, 2022-01-10:
Hi Team,

I am facing an issue while launching the mapping.launch file, attaching the error while executing rosluanch my_robot mapping.launch :

[ WARN] [1641793652.963000151]: /rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).

/rtabmap/rtabmapviz subscribed to (approx sync):

/rtabmap/odom \

/camera/rgb/image_raw \

/camera/depth/image_raw \

/camera/rgb/camera_info \

/scan

Although the topics are getting subscribed:

robond@udacity:~/catkin_ws$ rostopic hz /camera/rgb/image_raw /camera/depth/image_raw /camera/rgb/camera_info /scan

subscribed to [/camera/rgb/image_raw]

subscribed to [/camera/depth/image_raw]

subscribed to [/camera/rgb/camera_info]

subscribed to [/scan]

WARNING: may be using simulated time

topic rate min_delta max_delta std_dev window

==========================================================================

/camera/rgb/image_raw 10.48 0.026 0.541 0.149 11

/camera/depth/image_raw 9.667 0.033 0.545 0.1564 11

/camera/rgb/camera_info 4.454 0.041 0.609 0.2254 10

topic rate min_delta max_delta std_dev window

==========================================================================

/camera/rgb/image_raw 15.22 0.026 0.541 0.09052 30

/camera/depth/image_raw 14.64 0.033 0.545 0.09398 30

/camera/rgb/camera_info 6.071 0.041 0.609 0.1441 28

topic rate min_delta max_delta std_dev window

==========================================================================

/camera/rgb/image_raw 16.6 0.026 0.541 0.07594 43

/camera/depth/image_raw 16.09 0.033 0.545 0.07699 43

/camera/rgb/camera_info 6.555 0.041 0.609 0.1213 42

Please feel free to ask if more information is required.

I have added the path to my launch file here : [https://github.com/Rafi-UST/RTABMAP.git](url)

rafism1997, 2022-01-13:
Hi @matlabbe , 

Can you please look into this issue?

matlabbe, 2022-01-16:
In your rostopic hz output, I don't see the /scan topic.

Rafi-UST, 2022-01-17:
robond@udacity:~/catkin_ws$ rostopic hz /scan 
subscribed to [/scan]
WARNING: may be using simulated time
average rate: 39.863
	min: 0.000s max: 0.054s std dev: 0.01511s window: 36
average rate: 39.630
	min: 0.000s max: 0.061s std dev: 0.01793s window: 91
average rate: 39.807
	min: 0.000s max: 0.061s std dev: 0.01813s window: 125
average rate: 39.738
	min: 0.000s max: 0.061s std dev: 0.01788s window: 159
average rate: 39.870
	min: 0.000s max: 0.061s std dev: 0.01845s window: 197
^Caverage rate: 39.703
	min: 0.000s max: 0.061s std dev: 0.01860s window: 231

So I found out 2 mistakes that I did earlier, one being that I needed to change the frame_id from base_footprint to robot_footprint in the mapping.launch file, second being that in my gazebo file scan topic was also wrong. I changed both the errors but still I am receiving only the 3d map and not the graph view or loop enclosure detection in rtabmap.

rafism1997, 2022-01-17:
`rostopic list `

[image]

matlabbe, 2022-01-17:
If this screenshot contains all topics, those from rtabmap are missing, looks like the node is not running. Is it crashed on startup?