introlab/rtabmap_ros #702: When I input semantic image, mapData doesn't generated

YvonneZhao0305, 2022-01-13:
Hello, I have a problem building a semantic map.
1. When I input the original camera image, MapData is released normally and rtabmapviz displays the map normally.
2. When I input a semantic image (PSPNet  network), MapData will not be published, resulting in rtabmapviz not displaying the map.
3. When I use Centermask to generate semantic images, everything is normal.
I suspect this is related to the message synchronization policy. I changed the synchronization policy in GuiWrapper to sync_policies::ApproximateTime , and increased the queue size, but it still hasn't improved.
How to display semantic map in rtabmapviz normally？
Thanks. @matlabbe

matlabbe, 2022-01-16:
Hi, if 3 works and not 2, what is the frame rate of both approaches? If 2 is slower, this could be indeed a sync problem. Which topics are subscribed? You can set `approx_sync:=true` and `queue_size` parameters directly for rtabmap node. When the map is not showing in rtabmapviz, you should check rtabmap node instead. Are there warnings telling that rtabmap did not received topics for the past 5 seconds?

YvonneZhao0305, 2022-01-17:
> Hi, if 3 works and not 2, what is the frame rate of both approaches? If 2 is slower, this could be indeed a sync problem. Which topics are subscribed? You can set `approx_sync:=true` and [...]

Thanks for your answer. I found that this is really a sync problem. When I pause the dataset, 2 will continue to process some frames. I modified the program for publishing semantic  images, locked the original image messages, and then solved this problem.

closed 2022-01-17 by YvonneZhao0305