introlab/rtabmap_ros #703: change robot position every new start at In localization mode

muratkoc503, 2022-01-13:
@matlabbe , Hi. I work at localization mode in Rtabmap. The robot starts where it left off on the map in rviz. Probably, this reason shutdown. How i start to my initial position at localization mode. Thanks. I work on Gazebo simulation.

matlabbe, 2022-01-13:
You can set `RGBD/StartAtOrigin` parameter to true, so that the robot always thinks it is restarting from first node in the map instead of its latest pose when it shut down (default).

You can also configure the 2D Pose Estimate button of rviz to send the pose on `/rtabmap/initialpose` topic, so that you can manually teleport the robot to its real location. You can also send an arbitrarly initial pose to `/rtabmap/initialpose` directly too by command line.

muratkoc503, 2022-01-14:
thanks. i solve my problem. i use`RGBD/StartAtOrigin` parameter set true. I dont cant use 2D Pose Estimate button. In rviz, 2d button does not working at localization mode. i can't use `/rtabmap/initialpose`. This return Pose received is null! error messsage on terminal. Probably my an other problem cause this.

closed 2022-01-15 by muratkoc503

matlabbe, 2022-01-16:
Either the pose sent by rviz is really null, or there is a bug in the conversion of the message.
https://github.com/introlab/rtabmap_ros/blob/8fff821a871059e06386a64b0f3446a2f3124c65/src/CoreWrapper.cpp#L2573-L2578
You can check what look like the pose sent by rviz with:
```rostopic echo /rtabmap/initialpose```

muratkoc503, 2022-01-16:
`rostopic echo /rtabmap/initialpose` show true:
[image]

matlabbe, 2022-11-19:
The quaternion is invalid, it should be at least `x=o,y=o,y=o,w=1`