introlab/rtabmap_ros #704: local_grid_obstacle topic

muratkoc503, 2022-01-15:
@matlabbe , Hi,
/rtabmap/local_grid_obstacle topic does not publish anything. but /rtabmap/cloud_obstacles publish topic(observed rostopic echo). i dont understandt, wtah is reason. İ want to use /rtabmap/local_grid_obstacle for costmap parameter for local (dynamic) obstacle detection.
Thanks

muratkoc503, 2022-01-15:
this reason probably `queue_size`. i increase this paramater value, obstacle is observed in local map.

matlabbe, 2022-01-16:
In general, for the local costmap, I prefer sending directly a point cloud from the sensor to the costmap as the frame rate would be higher than rtabmap update rate.

muratkoc503, 2022-01-16:
thanks. after, i try it hopefully. now, i use pointcloud_to_scan package. That is fast too.

closed 2022-11-19 by matlabbe