introlab/rtabmap_ros #707: Spawning launch equivalent as a node from C++

Tannz0rz, 2022-01-19:
Hello IntRoLab,

I need to introduce some statefulness with my robot by using a state machine library titled "SMACC" (https://github.com/reelrbtx/SMACC). In order to switch between the "mapping" and "localization" states I need to be able to construct and destruct the instances of RTABMAP nodes from my code.

Suppose I use the following launch settings:
`roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" depth_topic:=/camera/aligned_depth_to_color/image_raw rgb_topic:=/camera/color/image_raw camera_info_topic:=/camera/color/camera_info approx_sync:=false`

How can I equivalently spawn the required nodes from C++ with these settings? I'm certain I could figure it out if I sat down and reviewed all of the various parameters in the `rtabmap.launch`, but there are so many it is a bit confusing. I know I can use `system` to spawn the `roslaunch` process, but this is a less-than-elegant solution to the problem.

Regards,
Tannz0rz

matlabbe, 2022-01-20:
rtabmap and rgbd_odometry can be launched as nodelets, so it would be possible to load them in a nodelet manager and unload them depending on your states. In c++, it looks like this:
https://github.com/introlab/rtabmap_ros/blob/8fff821a871059e06386a64b0f3446a2f3124c65/src/CoreNode.cpp#L91-L94
https://github.com/introlab/rtabmap_ros/blob/8fff821a871059e06386a64b0f3446a2f3124c65/src/RGBDOdometryNode.cpp#L72-L75
It seems you can unload a nodelet with [unload](http://docs.ros.org/en/melodic/api/nodelet/html/classnodelet_1_1Loader.html#af987f73a7bc3b019d378ec1257fd9a6f) function. Parameters would be passed by remap and nargv parameters.

Launching as nodelet from command lines equivalent:
```bash
rosrun nodelet nodelet manager __name:=nodelet_manager

rosrun nodelet nodelet load rtabmap_ros/rtabmap nodelet_manager \
   __name:=rtabmap \
   rgb/image:=/camera/color/image_raw \
   rgb/camera_info:=/camera/color/camera_info \
   depth/image:=/camera/aligned_depth_to_color/image_raw \
   _approx_sync:=false \
   _frame_id:=camera_link \
   --delete_db_on_start

rosrun nodelet nodelet load rtabmap_ros/rgbd_odometry nodelet_manager \
   __name:=rgbd_odometry \
   rgb/image:=/camera/color/image_raw \
   rgb/camera_info:=/camera/color/camera_info \
   depth/image:=/camera/aligned_depth_to_color/image_raw \
   _approx_sync:=false \
   _frame_id:=camera_link
```

Other option is to launch everything and use pause/resume services to enable/disable nodes, or call set_mode_localization and set_mode_mapping to switch between SLAM and localization modes without killing any nodes.

Tannz0rz, 2022-01-21:
This is fantastic, thank you so much for the assistance.

closed 2022-01-21 by Tannz0rz