introlab/rtabmap_ros #708: problem with odom_cache

amyburden, 2022-01-20:
I follow the instructions and when trying catkin_make the rtabmap_ros. It give the error

/catkin_ws/src/rtabmap_ros/src/MsgConversion.cpp:513:23: error: ‘const Info’ {aka ‘const struct rtabmap_ros::Info_<std::allocator<void> >’} has no member named ‘odom_cache’
  513 |  mapGraphFromROS(info.odom_cache, poses, constraints, t);

apt install rtabmap_ros can work normally.

matlabbe, 2022-01-20:
If you build rtabmap_ros from source, remove the binaries `sudo apt remove ros-$ROS_DISTRO-rtabmap`.

Also, you can force catkin to redo cmake so that messages are correctly generated:
```
cd ~/catkin_ws/build
cmake .
cd ~/catkin_ws
catkin_make
```

amyburden, 2022-01-21:
Thanks so much. It has been solved.

closed 2022-01-21 by amyburden