introlab/rtabmap_ros #709: Unable to install RTAB-MAP from build from source

teng339, 2022-01-26:
Hi, I am new to ROS and RTAB-MAP. When i follow the instruction to install RTAB-MAP from build from source.  In the step "3.0 Install RTAB-Map ros-pkg in your src folder of your Catkin workspace" , when I run this command "$catkin_make -j4", There is error occurs as show below. May someone help me to solve it? Thank you.

/home/testlf/abc_ws/src/rtabmap_ros/src/CoreWrapper.cpp: In member function ‘bool rtabmap_ros::CoreWrapper::getNodesInRadiusCallback(rtabmap_ros::GetNodesInRadius::Request&, rtabmap_ros::GetNodesInRadius::Response&)’:
/home/testlf/abc_ws/src/rtabmap_ros/src/CoreWrapper.cpp:4071:66: error: ‘rtabmap_ros::GetNodesInRadius::Request’ {aka ‘struct rtabmap_ros::GetNodesInRadiusRequest_<std::allocator<void> >’} **has no member named ‘k**’
 4071 |   poses = rtabmap_.getNodesInRadius(req.node_id, req.radius, req.k, &dists);
      |                                                                  ^
/home/testlf/abc_ws/src/rtabmap_ros/src/CoreWrapper.cpp:4075:92: error: ‘rtabmap_ros::GetNodesInRadius::Request’ {aka ‘struct rtabmap_ros::GetNodesInRadiusRequest_<std::allocator<void> >’} **has no member named ‘k’**
 4075 |   poses = rtabmap_.getNodesInRadius(Transform(req.x, req.y, req.z, 0,0,0), req.radius, req.k, &dists);
      |                                                                                            ^
/home/testlf/abc_ws/src/rtabmap_ros/src/CoreWrapper.cpp:4081:6: error: ‘rtabmap_ros::GetNodesInRadius::Response’ {aka ‘struct rtabmap_ros::GetNodesInRadiusResponse_<std::allocator<void> >’} **has no member named ‘distsSqr’**
 4081 |  res.distsSqr.resize(poses.size());
      |      ^~~~~~~~
/home/testlf/abc_ws/src/rtabmap_ros/src/CoreWrapper.cpp:4090:7: error: ‘rtabmap_ros::GetNodesInRadius::Response’ {aka ‘struct rtabmap_ros::GetNodesInRadiusResponse_<std::allocator<void> >’} **has no member named ‘distsSqr’**
 4090 |   res.distsSqr[index] = dists.at(iter->first);
      |       ^~~~~~~~
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_sync.dir/build.make:167: rtabmap_ros/CMakeFiles/rtabmap_sync.dir/src/CoreWrapper.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2607: rtabmap_ros/CMakeFiles/rtabmap_sync.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j1" failed

Timple, 2022-01-26:
Did you remove the `apt` binaries?

teng339, 2022-01-26:
It can be installed after I remove the apt binaries. Thank you.

closed 2022-01-26 by teng339