introlab/rtabmap_ros #711: rtabmap received signal SIGABRT only with rviz!

hichamhendy, 2022-01-28:
[image]

I am trying to launch rtabmap with RVIz but I can't. I really can't understand where the issue is!
Here some parameters 

SUMMARY
========

CLEAR PARAMETERS
 * /points_xyzrgb/
 * /rtabmap/icp_odometry/
 * /rtabmap/rtabmap/

PARAMETERS
 * /points_xyzrgb/approx_sync: False
 * /points_xyzrgb/decimation: 4.0
 * /points_xyzrgb/voxel_size: 0.0
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /rtabmap/icp_odometry/config_path: 
 * /rtabmap/icp_odometry/expected_update_rate: 0.0
 * /rtabmap/icp_odometry/frame_id: camera_link
 * /rtabmap/icp_odometry/ground_truth_base_frame_id: 
 * /rtabmap/icp_odometry/ground_truth_frame_id: 
 * /rtabmap/icp_odometry/guess_frame_id: 
 * /rtabmap/icp_odometry/guess_min_rotation: 0.0
 * /rtabmap/icp_odometry/guess_min_translation: 0.0
 * /rtabmap/icp_odometry/max_update_rate: 0.0
 * /rtabmap/icp_odometry/odom_frame_id: odom
 * /rtabmap/icp_odometry/publish_tf: True
 * /rtabmap/icp_odometry/queue_size: 10
 * /rtabmap/icp_odometry/scan_cloud_max_points: 0
 * /rtabmap/icp_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/icp_odometry/wait_imu_to_init: True
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: False
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_frame_id_init: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 0.001
 * /rtabmap/rtabmap/odom_tf_linear_variance: 0.001
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/scan_cloud_max_points: 0
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgb: True
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: True
 * /rtabmap/rtabmap/subscribe_scan_descriptor: False
 * /rtabmap/rtabmap/subscribe_stereo: False
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2

NODES
  /
    points_xyzrgb (nodelet/nodelet)
    rviz (rviz/rviz)
  /rtabmap/
    icp_odometry (rtabmap_ros/icp_odometry)
    rtabmap (rtabmap_ros/rtabmap)

ROS_MASTER_URI=http://localhost:11311

process[rtabmap/icp_odometry-1]: started with pid [10870]
process[rtabmap/rtabmap-2]: started with pid [10871]
process[rviz-3]: started with pid [10872]
process[points_xyzrgb-4]: started with pid [10873]
type is rtabmap_ros/point_cloud_xyzrgb
[ INFO] [1643359376.708105184]: Approximate time sync = false
^C[points_xyzrgb-4] killing on exit
[rviz-3] killing on exit
[rtabmap/icp_odometry-1] killing on exit
[rtabmap/rtabmap-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

matlabbe, 2022-01-28:
Can you type 'bt' in rtabmap gdb terminal?

hichamhendy, 2022-01-31:
I unfortunately managed to solve the problem by using only the binaries in preference to the source.

closed 2022-01-31 by hichamhendy