introlab/rtabmap_ros #712: User Data only for visualization or also used for localization and planning?

PaddyCube, 2022-01-30:
Hello,
I'm a beginner in ROS and implementing rtabmap for outdoor navigation. Beside D435 camera and Lidar, my robot is equipped with sensors which detect signal strength of a perimeter fence.

So I understand that I can use userData to visualize sensor data related to map, which is cool.

But would this data also been used for localization? I want to ensure robot operates only inside the perimeter loop. If robot comes closer to perimeter, signal strength will increase. If robot crosses the border and leave the allowed area, signal strength will be high but with negative sign.

Does rtabmap uses such data, if provided? Or is it just visualization?

BR Patrick

matlabbe, 2022-01-31:
Data included in userData are not used by rtabmap. It is a mechanism for users to save some data (can be anything) in nodes created in the map. Rtabmap will then republish that data with associated nodes, so you know when and where the userData are in the map. The userData is deserialized by user code outside rtabmap. The [example](http://wiki.ros.org/rtabmap_ros/Tutorials/WifiSignalStrengthMappingUserDataUsage) using Wifi strength is doing that. In that example, rtabmap doesn't use Wifi values for localization (it sees it as only a Blob of data).

cheers,
Mathieu

SahanGura, 2022-04-12:
@matlabbe Can we use the given user_data to rtabmap in mapping mode, when we open the database in localization mode? (for visualization)

PaddyCube, 2023-10-21:
Explanation given

closed 2023-10-21 by PaddyCube