introlab/rtabmap_ros #716: D435 outdoor, RGB-D or Stereo?

PaddyCube, 2022-02-15:
Hello,

my robot is equipped with D435 stereo camera, 2D Lidar and wheel odometry. I want to use it for outdoor SLAM.
I was able to run your RGB-D demo with the launch files for D435 you provided for indoor use.

But I'm lost when it comes to outdoor navigation. Should I use RGB-D indoor and Stereo SLAM for outdoor purposes?

BR Patrick

matlabbe, 2022-02-16:
We D435, you can continue using RGB-D mode for outdoor. However with D435, I prefer the stereo IR mode (with ir emitter disabled) for larger field of view, less motion blur and perfect synchronization between the left and right cameras. See http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping#Bring-up_example_with_RealSense_cameras

PaddyCube, 2022-03-02:
Thanks @matlabbe,
I was looking for such a hint.

closed 2022-03-02 by PaddyCube