introlab/rtabmap_ros #720: Feature request: rtabmap odometry output

Timple, 2022-02-16:
rtabmap (in mapping an localization mode) determines the robots position w.r.t. the (generated) pointcloud.
It outputs a `map -> base_link` tf (or `map -> odom` to keep the tf tree-like).

Would it be possible that it also outputs an odometry message?

This could hold more information than the tf (covariances). But it would also allow further fusing of this position estimation with other sensors.

For example high-frequent fusing an imu will increate a high-frequent position estimate (faster than the lidar scan rate for example).

matlabbe, 2022-02-21:
rtabmap's `map->odom` follows REP105: https://www.ros.org/reps/rep-0105.html#map, causing discrete jumps at unknown rate.
`/rtabmap/localization_pose` (which is a [PoseWithCovariance](http://docs.ros.org/en/lunar/api/geometry_msgs/html/msg/PoseWithCovariance.html) msg) is published with graph optimization marginal covariance only when graph is re-optimized (e.g., when a loop closure happens). We don't republish the input odometry with the map->odom correction. Think rtabmap's output as GPS low frame rate output which can be not always published.

I am not sure how we could output odometry topic from rtabmap node, as rtabmap update rate is already slow (1 Hz) in comparison to topics you are looking to do fusion.

Based on REP105 for what you want, I think you should work with odom->base_link instead of map->base_link. Like move_base, you should do your local controller in odom frame, and global planning (done a lower rate) with map frame.

cheers,
Mathieu

Timple, 2022-02-21:
I am familiar with REP105 and how it should function. We have robots fusing 1hz GPS, the Odometry message was meant as message format, not as 'real odometry'.

> `/rtabmap/localization_pose` is published with graph optimization marginal covariance only when graph is re-optimized.

Yes, a PoseWithCovariance would do as well! But I thought of Odometry as I thought rtabmap had an internal velocity model (might not be).

> rtabmap update rate is already slow (1 Hz).

That doesn't matter (just like the gps). But as I understand the `/rtabmap/localization_pose` is not published at this rate?
If that would be the case, we would already be helped!

matlabbe, 2022-02-23:
rtabmap node doesn't have internal velocity model, it just uses its odometry input as the pose. rtabmap_ros's odometry node uses either a constant velocity model, guess from Tf or IMU to better estimate where the next frame will be before doing feature matching or scan matching. 

`/rtabmap/localization_pose` is published only when a loop closure happens or localization happens, so we cannot guarantee it will be published at 1 Hz or even published at all (if robot cannot localize anymore to the map).

Note sure exactly what you want to fuse (or what are you meaning by "real odometry"), but `/rtabmap/localization_pose` is not an odometry at low frame rate, it is localization. The robot can jump 1 meter, or even 1 km if a large loop is closed at some discrete time, so this doesn't follow the odometry definition from REP105. Using `robot_localization`, it is possible to fuse PoseWithCovariance topics with Odometry topics, like the GPS as you may have already tried (`publish_tf` would have to be set to false to let `robot_localization` publishing map->odom). You could then get high frame rate for the final "odometry".

Timple, 2022-03-11:
Yes, `localization_pose` is exactly what we needed!

Thank you!

closed 2022-03-11 by Timple

matlabbe, 2023-02-08:
Update: from this [commit](https://github.com/introlab/rtabmap/commit/dcedd8bb6b9f6e9a8e884a271fcfff6671ee3cbf), `localization_pose` is always published in localization mode and covariance would increase when not re-localized, otherwise covariance would shrink when re-localized.