introlab/rtabmap_ros #721: how to connect map->odom tf ?

kdhRGT, 2022-02-20:
Hello author!

I can run rtapmap by your help.

but now I have a small problem.

I can run rtabmap by my rosbag dataset. but it said some errors.

firstly rtabmap said errors like this

```
could not get transform from base_link to camera_color_optical_frame (stamp = ~~~) ~~~
```

This is my tf free from rosbag

[frames_rosbag.pdf](https://github.com/introlab/rtabmap_ros/files/8103731/frames_rosbag.pdf)

So I found the similar issue on it

https://answers.ros.org/question/224250/rtabmap_ros-with-realsense-odometry-could-not-get-transform-from-base_link-to-camera_color_optical_frame/

I followed a similar solution for my system.

I added some commands on the launch file like this in rtabmap.launch

```
<node pkg="tf" type="static_transform_publisher"  name="base_to_realsense"
      args="0 0 0 0 0 0 /base_link /camera_link 100" /> 
```

so now my current tf tree is this

[frames.pdf](https://github.com/introlab/rtabmap_ros/files/8103732/frames.pdf)

but.. i don't know how can I connect map frame to odom..

i can watch the odometry results in odom frame in rviz..

but map results doest show me anything..

[image]

can you help me how to connect odom frame with map frame?

thank you so much

best regards

matlabbe, 2022-02-21:
If you do "enter" on gdb terminal of rtabmap to continue? If it is crashed, type "bt" in the gdb terminal to show the backtrace if so. Is rtabmap built from source or binaries?

kdhRGT, 2022-02-21:
@matlabbe Thanks for replying to me.

now I am doing a source build~!

I put the "enter" on gdb terminal of rtabmap to continue.

and then it does not show me any errors.

But it shows an error when I play my rosbag file

odometry works very well but the mapping part does not work well.

Can you guess which I did a mistake?

Thank you so much for your help.

matlabbe, 2022-02-23:
Right there you can type "bt" in gdb after the seg fault to see the full backtrace (from which function in rtabmap is happened). 

Is rtabmap built from source or binaries?

kdhRGT, 2022-02-23:
Hello, author~!

My workspace is built from source~!

I also instated a binary program.

Sorry I think I misunderstand what you wanted because I was not friendly with gdb.

**Here, I attached what I had errors from gdb.**

[image]

[image]

[image]

The image resolution is very low so I am not sure that you can see it correctly or not

#0 in cv::Mat::copySize(cv::Mat const&)  () at /usr/lib/x86_64-linux...
#1 in  rtabmap::transform::transform(rtabmap::transform const&)
....

Are all these are errors? I am not sure that that is the reason why I attached a full picture of the error window.

matlabbe, 2022-02-24:
Thank you, it is what I wanted. However, crashing in Transform constructor with cv::Mat may be related to which libraries are loaded on runtime, corrupting memory.

I suggest to uninstall rtabmap built from source and use the binaries one. You can do in rtabmap/build:
```bash
cd rtabmap/build
make
sudo make uninstall
```
This will remove rtabmap from /usr/local path. Remove afterwards rtabmap_ws from your `ROS_PACKAGE_PATH` (just don't source `/rtabmap_ws/devel/setup.bash`). 

for the binaries:
```bash
sudo apt install ros-melodic-rtabmap-ros
```

kdhRGT, 2022-02-24:
@matlabbe

Thank you so much, sir.

I am successful to build the ROS workspace!!!!!!

i solved the problem by this process.

go to rtabmap/build folder

```
cd ~/rtabmap/build
make
sudo make uninstall
````

after that, uninstall the binaries.

```
sudo apt-get remove ros-melodic-rtabmap ros-melodic-rtabmap-ros
```

and install rtabmap again.

```
cd ~/rtabmap/build
sudo make install
```

and catkin_my_workspace

```
cd ~/rtabmap_ros
```

I should remove all folders about devel, build before building the workspace.

and it works!!!

I got some hints from your last answer.

Thank you so much

Best regards

[image]

matlabbe, 2022-02-25:
If the rtabmap binaries were installed before building rtabmap_ros, rtabmap_ros could have been using headers from /usr/local instead of the binaries in /opt/ros, then on runtime the wrong version of rtabmap libraries were loaded causing the crash.

Glad that you've found a solution,
Mathieu

closed 2022-02-25 by matlabbe