introlab/rtabmap_ros #722: I'm using Ubuntu 18.04 ROS Melodic while installing RTAB-Map ros-pkg in your src folder of your Catkin workspace. I am getting this error

adarsha4, 2022-02-21:
#### Running command: "cmake /home/syncmaster/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/syncmaster/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/syncmaster/catkin_ws/install -G Unix Makefiles" in "/home/syncmaster/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/syncmaster/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/syncmaster/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 18 packages in topological order:
-- ~~  - turtlebot3 (metapackage)
-- ~~  - turtlebot3_autorace (metapackage)
-- ~~  - turtlebot3_autorace_control
-- ~~  - turtlebot3_autorace_core
-- ~~  - turtlebot3_msgs
-- ~~  - turtlebot3_navigation
-- ~~  - turtlebot3_simulations (metapackage)
-- ~~  - hls_lfcd_lds_driver
-- ~~  - turtlebot3_autorace_camera
-- ~~  - turtlebot3_autorace_detect
-- ~~  - turtlebot3_bringup
-- ~~  - turtlebot3_example
-- ~~  - turtlebot3_fake
-- ~~  - turtlebot3_gazebo
-- ~~  - turtlebot3_slam
-- ~~  - turtlebot3_teleop
-- ~~  - rtabmap_ros
-- ~~  - turtlebot3_description
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'turtlebot3'
-- ==> add_subdirectory(turtlebot3/turtlebot3)
-- +++ processing catkin metapackage: 'turtlebot3_autorace'
-- ==> add_subdirectory(turtlebot3_autorace/turtlebot3_autorace)
-- +++ processing catkin package: 'turtlebot3_autorace_control'
-- ==> add_subdirectory(turtlebot3_autorace/turtlebot3_autorace_control)
-- +++ processing catkin package: 'turtlebot3_autorace_core'
-- ==> add_subdirectory(turtlebot3_autorace/turtlebot3_autorace_core)
-- +++ processing catkin package: 'turtlebot3_msgs'
-- ==> add_subdirectory(turtlebot3_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- turtlebot3_msgs: 3 messages, 0 services
-- +++ processing catkin package: 'turtlebot3_navigation'
-- ==> add_subdirectory(turtlebot3/turtlebot3_navigation)
-- +++ processing catkin metapackage: 'turtlebot3_simulations'
-- ==> add_subdirectory(turtlebot3_simulations/turtlebot3_simulations)
-- +++ processing catkin package: 'hls_lfcd_lds_driver'
-- ==> add_subdirectory(hls_lfcd_lds_driver)
-- Boost version: 1.65.1
-- +++ processing catkin package: 'turtlebot3_autorace_camera'
-- ==> add_subdirectory(turtlebot3_autorace/turtlebot3_autorace_camera)
-- +++ processing catkin package: 'turtlebot3_autorace_detect'
-- ==> add_subdirectory(turtlebot3_autorace/turtlebot3_autorace_detect)
-- +++ processing catkin package: 'turtlebot3_bringup'
-- ==> add_subdirectory(turtlebot3/turtlebot3_bringup)
-- +++ processing catkin package: 'turtlebot3_example'
-- ==> add_subdirectory(turtlebot3/turtlebot3_example)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Generating .msg files for action turtlebot3_example/Turtlebot3 /home/syncmaster/catkin_ws/src/turtlebot3/turtlebot3_example/action/Turtlebot3.action
-- turtlebot3_example: 7 messages, 0 services
[... 326 lines cut ...]
/home/syncmaster/catkin_ws/src/rtabmap_ros/include/rtabmap_ros/CoreWrapper.h:194:7: note:   no known conversion for argument 4 from ‘std::__cxx11::string {aka std::__cxx11::basic_string<char>}’ to ‘const std::vector<float>&’
/home/syncmaster/catkin_ws/src/rtabmap_ros/src/CoreWrapper.cpp: In member function ‘virtual void rtabmap_ros::CoreWrapper::commonOdomCallback(const OdometryConstPtr&, const UserDataConstPtr&, const OdomInfoConstPtr&)’:
/home/syncmaster/catkin_ws/src/rtabmap_ros/src/CoreWrapper.cpp:1867:27: error: no matching function for call to ‘rtabmap_ros::CoreWrapper::process(ros::Time&, rtabmap::SensorData&, rtabmap::Transform&, std::__cxx11::string&, cv::Mat&, rtabmap::OdometryInfo&, double)’
    timerConversion.ticks());
                           ^
In file included from /home/syncmaster/catkin_ws/src/rtabmap_ros/src/CoreWrapper.cpp:28:0:
/home/syncmaster/catkin_ws/src/rtabmap_ros/include/rtabmap_ros/CoreWrapper.h:194:7: note: candidate: void rtabmap_ros::CoreWrapper::process(const ros::Time&, rtabmap::SensorData&, const rtabmap::Transform&, const std::vector<float>&, const string&, const cv::Mat&, const rtabmap::OdometryInfo&, double)
  void process(
       ^~~~~~~
/home/syncmaster/catkin_ws/src/rtabmap_ros/include/rtabmap_ros/CoreWrapper.h:194:7: note:   no known conversion for argument 4 from ‘std::__cxx11::string {aka std::__cxx11::basic_string<char>}’ to ‘const std::vector<float>&’
/home/syncmaster/catkin_ws/src/rtabmap_ros/src/CoreWrapper.cpp: At global scope:
/home/syncmaster/catkin_ws/src/rtabmap_ros/src/CoreWrapper.cpp:1872:6: error: prototype for ‘void rtabmap_ros::CoreWrapper::process(const ros::Time&, rtabmap::SensorData&, const rtabmap::Transform&, const string&, const cv::Mat&, const rtabmap::OdometryInfo&, double)’ does not match any in class ‘rtabmap_ros::CoreWrapper’
 void CoreWrapper::process(
      ^~~~~~~~~~~
In file included from /home/syncmaster/catkin_ws/src/rtabmap_ros/src/CoreWrapper.cpp:28:0:
/home/syncmaster/catkin_ws/src/rtabmap_ros/include/rtabmap_ros/CoreWrapper.h:194:7: error: candidate is: void rtabmap_ros::CoreWrapper::process(const ros::Time&, rtabmap::SensorData&, const rtabmap::Transform&, const std::vector<float>&, const string&, const cv::Mat&, const rtabmap::OdometryInfo&, double)
  void process(
       ^~~~~~~
rtabmap_ros/CMakeFiles/rtabmap_sync.dir/build.make:254: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_sync.dir/src/CoreWrapper.cpp.o' failed
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_sync.dir/src/CoreWrapper.cpp.o] Error 1
CMakeFiles/Makefile2:8546: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_sync.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_sync.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4" failed

matlabbe, 2022-02-21:
Sorry for this issue, it was a wrong commit yesterday, it should be fixed by https://github.com/introlab/rtabmap_ros/commit/8570fe3179fb7434863706a6b03b59dae440aad0

adarsha4, 2022-03-02:
Thank you so much. Now it is working

closed 2022-03-02 by adarsha4