introlab/rtabmap_ros #725: Parameter has NoneType when launching with ROS 2

ericdepotter, 2022-02-26:
I wanted to test RTAB-Map with ROS2 and a OAK-D stereo camera. The OAK-D part is running correctly, however when I try to launch RTAB-Map, I get an error that one of the parameters is None.

My environment:

- Ros 2 Foxy
- Latest rtabmap on brach master (e7ee02512771992ca447ad559ccc3c1de9627926)
- Latest rtabmap_ros on brach ros2 (bc29e8694f2d978c2de21e1e023722fe41b52c8e)

Execution command (same exception occurs when omitting all parameters):

> ros2 launch rtabmap_ros rtabmap.launch.py \
>     rtabmap_args:="--delete_db_on_start" \
>     subscribe_stereo:=true \
>     left_image_topic:="/left/image_rect" \
>     right_image_topic:="/right/image_rect" \
>     left_camera_info_topic:="/left/camera_info" \
>     right_camera_info_topic:="/right/camera_info" \
>     frame_id:="oak-d_frame" \
>     approx_sync:=true \
>     wait_for_transform:=0.5 \
>     visual_odometry:=true

The full exception: 
> Task exception was never retrieved
> future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=TypeError("Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)")>
> Traceback (most recent call last):
>   File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
>     await self.__process_event(next_event)
>   File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
>     visit_all_entities_and_collect_futures(entity, self.__context))
>   File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
>     futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
>   File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
>     futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
>   File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
>     futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
>   [Previous line repeated 2 more times]
>   File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
>     sub_entities = entity.visit(context)
>   File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
>     return self.execute(context)
>   File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
>     self._perform_substitutions(context)
>   File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
>     evaluated_parameters = evaluate_parameters(context, self.__parameters)
>   File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 160, in evaluate_parameters
>     output_params.append(evaluate_parameter_dict(context, param))
>   File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 95, in evaluate_parameter_dict

How can this error be resolved?

matlabbe, 2022-02-27:
With updating foxy core packages, I could reproduce the problem. This should be fixed in commit above.

matlabbe, 2022-02-27:
Long-story short, ros2 launch doesn't like anymore empty default value like this:
https://github.com/introlab/rtabmap_ros/blob/b81009b9f6f3491c5132ab5547ad73e9475ab345/launch/ros2/rtabmap.launch.py#L391

ericdepotter, 2022-02-28:
Now it is working correctly. Thanks

commit b81009b9 (2022-02-27): fixed #725
closed 2022-02-28 by ericdepotter